raspberrypi/linux #2289: Allow the default user to access the new gpio character device

gohai, 2017-11-26:
I want to switch Processing from using the legacy sysfs interface to the new character device interface (`/dev/gpiochip*`). To make this possible, those devices should be owned by the `gpio` group.

The following line, added to `99-com.rules` achieves this:

    SUBSYSTEM=="gpio", GROUP="gpio", MODE="0660"

Could this be added for upcoming Raspbian releases? Thanks!

pelwell, 2017-11-27:
I'm in favour - an efficient user-space GPIO API has been long overdue.

@XECDesign Can we insert this as line 6 of `/etc/udev/rules.d/99-com.rules`? At some point we should also add the new kernel-supplied tools (`lsgpio`, `gpio-hammer` and `gpio-event-mon`) to the distro, but the current output isn't pretty - all the pins are marked as unused, with no helpful labels(*).

(*) Minor correction - on 4.14 the brcmvirt-gpio driver shows "led0" for line 0, but that's all.

gohai, 2017-11-27:
I remember a discussion with the gpio maintainer Linus Walleij on the
Raspberry Kernel mailing list, where the issue of labels came up. Some, me
included, pushed back against using the physical header pin numbers, as we
seem to have finally settled on GPIO numbers - and reverting this might
only add to confusion. Linus solicited feedback, but I don’t know what
became with the patch ultimately.

My prime motivation for wanting to try out this API is that I *believe* it
will allow us to set and clear pull-ups. Currently board seems to come up
with the pull-ups set to defaults as per datasheet, which seems pretty
random.

pelwell, 2017-11-27:
I want to see BRCM GPIO/"pin" numbering everywhere - it's used exclusively by the firmware and the kernel, and it's guaranteed to make sense on any arbitrary hardware design (how would you use 40-pin header values on a Compute Module?).

If you have any feedback for us in terms of how to make the new GPIO interface more usable, or warnings about problems you have encountered, then please let us know.

XECDesign, 2017-11-27:
Added: https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/bd5a201475b14c7f00add8d8e5f5cd018ad15d8b
It should show up in apt in a week or two.

I'll need to create a package for `lsgpio` and the others, but I'll save that for later.

gohai, 2017-11-27:
> Added: RPi-Distro/raspberrypi-sys-mods@bd5a201

Thank you!

lategoodbye, 2017-11-30:
As a reference here is the lsgpio output of a Raspberry Pi 1 B with upstream 4.14:
```
GPIO chip: gpiochip0, "pinctrl-bcm2835", 54 GPIO lines
        line  0: "SDA0" unused
        line  1: "SCL0" unused
        line  2: "SDA1" unused
        line  3: "SCL1" unused
        line  4: "GPIO_GCLK" unused
        line  5: "CAM_GPIO1" unused [output]
        line  6: "LAN_RUN" unused [output]
        line  7: "SPI_CE1_N" unused
        line  8: "SPI_CE0_N" unused
        line  9: "SPI_MISO" unused
        line 10: "SPI_MOSI" unused
        line 11: "SPI_SCLK" unused
        line 12: "NC" unused
        line 13: "NC" unused
        line 14: "TXD0" unused
        line 15: "RXD0" unused
        line 16: "STATUS_LED_N" "ACT" [kernel output active-low]
        line 17: "GPIO17" unused
        line 18: "GPIO18" unused
        line 19: "NC" unused
        line 20: "NC" unused
        line 21: "GPIO21" unused
        line 22: "GPIO22" unused
        line 23: "GPIO23" unused
        line 24: "GPIO24" unused
        line 25: "GPIO25" unused
        line 26: "NC" unused
        line 27: "CAM_GPIO0" unused [output]
        line 28: "CONFIG0" unused
        line 29: "CONFIG1" unused
        line 30: "CONFIG2" unused
        line 31: "CONFIG3" unused
        line 32: "NC" unused
        line 33: "NC" unused
        line 34: "NC" unused
        line 35: "NC" unused
        line 36: "NC" unused
        line 37: "NC" unused
        line 38: "NC" unused
        line 39: "NC" unused
        line 40: "PWM0_OUT" unused
        line 41: "NC" unused
        line 42: "NC" unused
        line 43: "NC" unused
        line 44: "NC" unused
        line 45: "PWM1_OUT" unused
        line 46: "HDMI_HPD_P" unused
        line 47: "SD_CARD_DET" unused
        line 48: "SD_CLK_R" unused
        line 49: "SD_CMD_R" unused
        line 50: "SD_DATA0_R" unused
        line 51: "SD_DATA1_R" unused
        line 52: "SD_DATA2_R" unused
        line 53: "SD_DATA3_R" unused
```

JamesH65, 2017-12-04:
@XECDesign Can be closed?

XECDesign, 2017-12-04:
@JamesH65 The updated `raspberrypi-sys-mods` package is in the repo, so I would say yes. However, nothing has been done to add `lsgpio` and other binaries or change how they display data.

gohai, 2017-12-04:
Wanted to correct what I posted above: I realized that currently the new API does _not_ allow the setting and clearing of pull-up/down resistors. I confused them with setting open source / open drain, but these things seem to be orthogonal really.

lategoodbye, 2017-12-04:
From my point of view it should be easier to add a pull-up/down IOCTL instead of extending the sysfs.

pelwell, 2017-12-04:
I'm happy for us to be supporting the new interface, whether or not it supports the setting of pulls yet. Let's keep this open until we've added the new utilities.

gohai, 2017-12-05:
@lategoodbye Agree. The new (IOCTL) interface already has an extensible `flags` field, which is used e.g. for indicating whether the line should be an input or output. This should be sufficient for implementing the most basic pull-up/down configuration at least.

JamesH65, 2018-04-20:
Anyone have any progress to report on this one? Would be nice to know what stage we are at.

pelwell, 2018-06-27:
Last we heard we were waiting for `lsgpio` and family to make it into Raspbian.

JamesH65, 2018-06-27:
@XECDesign @spl237 Is this now in your court?

lategoodbye, 2018-06-27:
I think this is related to @gohai intension, starting with 4.18 the pinctrl driver would be [generic](https://github.com/torvalds/linux/commit/0de704955ee44738e88b9cc08546ea10354037ed).

XECDesign, 2018-06-27:
I don't feel a sense of urgency to add the gpio-utils package. Very few distros include them and there are many other binaries we don't include. The original issue is resolved. The new package would be for bonus points.

lategoodbye, 2018-06-27:
Sure there is no urgency, but the sysfs-gpio is obsolete since [2016](https://github.com/raspberrypi/linux/commit/fe95046e960b4b76e73dc1486955d93f47276134).

JamesH65, 2019-07-30:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

XECDesign, 2019-07-30:
Bumping it since it's probably time to package up gpio-utils.

XECDesign, 2019-08-08:
`gpio-utils` added to buster and should show up in the repo within about an hour.

pelwell, 2019-08-08:
Nice.