raspberrypi/linux #2290: Alsaplayer crashes with 24bit FLAC files

billhuey, 2017-11-27:
Alsaplayer crashes with 24bit FLAC files on the latest Raspbian

pelwell, 2017-11-27:
If it's a user space crash then this isn't the right place to report it. If it's a kernel crash we'll need to see a kernel crash log from the output of dmesg.

billhuey, 2017-11-27:
This is a user space crash. Thanks

pelwell, 2017-11-28:
I also get a segmentation fault in libflac_in.so when playing a 24-bit FLAC file with alsaplayer, on a Pi and running Ubuntu on an X86 machine under VirtualBox. This looks like a bug/limitation of alsaplayer, so the best place to report it would be at the Alsaplayer github: https://github.com/alsaplayer/alsaplayer/issues

Note that alsaplayer has it's own FLAC library, while other clients (mplayer, for example) uses the standard libFLAC, which can read 24-bit FLACs.

billhuey, 2017-11-30:
I just filed the report via the link you sent me. Will wait for a response.
Thank you.

bill

JamesH65, 2017-12-04:
Closing as this issue is not kernel related.

closed 2017-12-04 by JamesH65