raspberrypi/linux #2291: OpenGL KMS driver renders SDL window contents slightly crooked

ewmailing, 2017-11-27:
I enabled the GL Full KMS driver on new installation of Raspbian Stretch.
I built SDL 2.0.7 and ran a few simple programs that use the SDL 2D renderer (which is accelerated when available).

I discovered that the contents in the window are slightly crooked...rotated counterclockwise by maybe a degree or two. I am attaching a screenshot.
[image]

The background clear color is cyan. I draw a purple rectangle the size of the window first. You should not see the background, but you can see a little sliver of cyan on the bottom-right corner and the left-bottom corner.

I then draw a red rectangle on top and if you look closely at the edges, you can see the lines are not straight.

Then I draw some text, and if you look closely, you can also see that it is not straight.

I also built SDL with the OpenGL ES driver that works without X11 and takes up the full screen. It does not have this problem. So I believe the problem is with the experimental OpenGL Full KMS driver.

I am also attaching a simple SDL program in case you want to try it.

[main_c.c.zip](https://github.com/raspberrypi/linux/files/1507744/main_c.c.zip)

schnitzeltony, 2017-11-27:
Hmm the rotation issue (it was SDL2 at my case) looks familiar to me. I just tested: it is still there.

https://github.com/anholt/mesa/issues/58

ewmailing, 2017-11-27:
@schnitzeltony That looks like the exact same issue.

popcornmix, 2017-11-29:
I'll close this - the anholt repo is the correct place for this bug report.

closed 2017-11-29 by popcornmix