raspberrypi/linux #2292: Qt5 applications uses software rasterizer for graphics

Talkless, 2017-11-28:
On Raspberry Pi 3 B with Raspbian GNU/Linux 9.1 (stretch), Qt applications, such as opengl/qtquick examples (`/usr/lib/arm-linux-gnueabihf/qt5/examples/opengl/cube/cube` or `/usr/lib/arm-linux-gnueabihf/qt5/examples/quick/animation/animation` or my own built ones, has pretty poor performance.

Looks like Qt applications are using software rasterizer?
```
$ fgrep swrast /proc/$(pgrep cube)/maps
71f07000-7279e000 r-xp 00000000 b3:02 16708      /usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so
7279e000-727ae000 ---p 00897000 b3:02 16708      /usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so
727ae000-727de000 r--p 00897000 b3:02 16708      /usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so
727de000-727e9000 rw-p 008c7000 b3:02 16708      /usr/lib/arm-linux-gnueabihf/dri/swrast_dri.so
```
Meanwhile, `minecraft-pi` links to `libbrcm*.so`:
```
$ fgrep GL /proc/2625/maps
76eaa000-76ed2000 r-xp 00000000 b3:02 217150     /opt/vc/lib/libbrcmEGL.so
76ed2000-76ee2000 ---p 00028000 b3:02 217150     /opt/vc/lib/libbrcmEGL.so
76ee2000-76ee3000 rw-p 00028000 b3:02 217150     /opt/vc/lib/libbrcmEGL.so
76ee3000-76ef7000 r-xp 00000000 b3:02 217151     /opt/vc/lib/libbrcmGLESv2.so
76ef7000-76f07000 ---p 00014000 b3:02 217151     /opt/vc/lib/libbrcmGLESv2.so
76f07000-76f08000 rw-p 00014000 b3:02 217151     /opt/vc/lib/libbrcmGLESv2.so
```
Also, with Qt applications I see this warning:
```
$ /usr/lib/arm-linux-gnueabihf/qt5/examples/quick/animation/animation
libEGL warning: DRI2: failed to authenticate
```

XECDesign, 2017-11-28:
raspbian.org compiles for generic armv6hf platforms, not just the pi. We could compile a special version of qt5 for the raspberrypi.org repo, but that would break acceleration with the experimental accelerated mesa driver, which we will switch to as default at some point in the future.

Talkless, 2017-11-28:
Same issue with VLC by the way.

So I have to rebuild Qt itself? Or something more..?

Can I install that experimental mesa driver?

Does NOOBS provide acceleration by default, or there are other solution?

Thanks!

XECDesign, 2017-11-28:
You could rebuild Qt with the correct options.

In Raspbian, you can enable the experimental driver using raspi-config, under 'advanced options'.

NOOBS is an installer and boot manager, so acceleration doesn't really apply to it. Off the top of my head, I don't know how the version of Qt used by NOOBS is compiled, but it wouldn't make a huge difference anyway in that environment.

pelwell, 2017-11-28:
In case there's any confusion, Raspbian running under NOOBS is the same as a regular Raspbian installation except for the SD/disk partition usage, so using raspi-config to select the experimental VC4 graphics still applies - it's just the boot loader itself which won't make use of it.

Talkless, 2017-11-28:
Thanks @pelwell for clarification.

@XECDesign thanks, I'll try experiental driver first!

Talkless, 2017-11-28:
`raspi-config` indeed helped!

I'll still gonna try to build Qt though, but this is useful solution.

JamesH65, 2017-12-04:
Reading above comments, this can be closed (Its not really a Linux kernel issue anyway)

closed 2017-12-04 by JamesH65

Talkless, 2017-12-04:
>  (Its not really a Linux kernel issue anyway)

Oh sorry, yes, I understood this repo as a RaspberryPi *distribution* bug tracker for some unknown reasons.