raspberrypi/linux #2295: Usb performace over usbserial ftdi since 4.9.x kernel

danyconnect, 2017-12-02:
I encounter some problems with the latency of a serial usb starting with kernel 4.9.
Oscam + ftdi reader, no other hub usb.

Chip is ft232rl

whit kernel 4.4.50 i have 160ms
whit kernel 4.9.x i have  170ms +

Is stable, only problem is latency.
[ftdi.txt](https://github.com/raspberrypi/linux/files/1524054/ftdi.txt)

danyconnect, 2017-12-18:
one of this changes in 4.9.13 breaks ftdi performance
Johan Hovold (8):
      USB: serial: mos7840: fix another NULL-deref at open
      USB: serial: ftdi_sio: fix modem-status error handling
      USB: serial: ftdi_sio: fix extreme low-latency setting
      USB: serial: ftdi_sio: fix line-status over-reporting
      USB: serial: spcp8x5: fix modem-status handling
      USB: serial: opticon: fix CTS retrieval at open
      USB: serial: ark3116: fix register-accessor error handling
      USB: serial: console: fix uninitialised spinlock

https://www.kernel.org/pub/linux/kernel/v4.x/ChangeLog-4.9.13

danyconnect, 2017-12-18:
I tested 4.14.5 and not fixed yet

pelwell, 2017-12-18:
If you are building your own kernel I suggest you try reverting each of these three commits (as ones that seem most relevant):
```
427c3a9 USB: serial: ftdi_sio: fix modem-status error handling
c6dce26 USB: serial: ftdi_sio: fix extreme low-latency setting
a6bb1e1 USB: serial: ftdi_sio: fix line-status over-reporting
```
Use `git revert 427c3a9` to revert the first one, then if that doesn't fix the latency issue revert the others, one at a time. Alternatively, if the commits don't overlap you may be able to use `git reset --hard HEAD^` to remove the first reversion before applying the next. The `reset` is guaranteed to work, but attempting to apply the next commit may fail without first reverting its successor (the one higher up the list, if that's the case, use `git revert --abort` to rewind, then reapply the reversion.

If none of these commits turn out to be responsible for the slow down, cast the net a bit wider.

danyconnect, 2017-12-18:
thanks for your answer, i will try during the holidays and I will post the results.

equino, 2018-01-30:
Some news from @danyconnect ??

danyconnect, 2018-02-01:
I left 4.9.11 because raspberry is at a distance and I do not want to ruin something. I have another raspberry but I do not have a reader to test.

JamesH65, 2018-04-20:
@danyconnect Anything to report on this?

danyconnect, 2018-04-20:
I do not have a reader to test the perfomance at home (the server is at km from me), but I updated to the last kernel 4.14.30-v7 + and the "defect" is still present.

danyconnect, 2018-07-21:
i found a solution in german forum,

**apt update
apt install setserial**

in udev location /etc/udev/rules.d/
create file  **99-usb-serial.rules**   and put this format

**SUBSYSTEMS=="usb", KERNEL=="ttyUSB*", ATTRS{serial}=="xxxxxx", SYMLINK+="easy", RUN+="/bin/setserial /dev/easy low_latency"**

And ECM in oscam is back normal since **kernel 4.9.11** .
Thanks

closed 2018-07-21 by danyconnect