raspberrypi/linux #2299: kvm [1]: Error, CPU 0 not supported!

nkichukov, 2017-12-08:
Hi fellows,

I am trying to use the virtualization extensions on the arm cortex-a53 that comes with the raspberry pi 3b, however, KVM refuses to initialize even though all 4 pCPUs appear to be started in HYP mode.

The error that is reported when the kernel boots is:
[    2.180028] kvm [1]: Error, CPU 0 not supported!

Thus no /dev/kvm device exists.

Seems to be coming from this loop:
        for_each_online_cpu(cpu) {
                smp_call_function_single(cpu, check_kvm_target_cpu, &ret, 1);
                if (ret < 0) {
                        kvm_err("Error, CPU %d not supported!\n", cpu);
                        return -ENODEV;
                }
        }

inside kvm_arch_init in virt/kvm/arm/arm.c .

kernel: 4.14.0-v7+
compiler: gcc-6.4
raspberry pi running in 32bit mode
OS: Gentoo Linux
Kernel configuration and bootlog (dmesg) are attached.

Final goal is to use qemu with hardware acceleration for running guests. I think that --enable-kvm for QEMU will  not work if /dev/kvm is missing(kvm not initialized) thus it will not make use of the hardware virtualization extensions on the CPU.

Attachments:
[dmesg.txt](https://github.com/raspberrypi/linux/files/1542223/dmesg.txt)
[rpi-4.14.0-kernel.txt](https://github.com/raspberrypi/linux/files/1542226/rpi-4.14.0-kernel.txt)

Thank you!

JamesH65, 2018-04-20:
@nkichukov We haven't had the time to look at this, and don't really have any domain experts in house anyway. Have you had any luck with it?

nkichukov, 2018-04-22:
Due to this, I have decided to use LXC as a workaround. I am currently running 4.16.1 kernel but without the KVM support. I may retry it with on of the next kernel releases. 
If you do not have the expertise for this, feel free to close the RFA.
Thank you,
-N

eszabo12, 2022-05-20:
I am also having this issue