raspberrypi/linux #2301: lightdm locales/kbd

moserwi, 2017-12-08:
findings:
lightdm-gtk-greeter-settings is not used by lightdm. even with update-alternatives --config lightdm-greeter
implementing a taskbar in lightdm is not possible, the used variables have been changed with prefix gtk-*

the locales/kbd system/session are not synced

errors:
installed debians lightdm  - taskbar and it uses the lightdm-gtk-greeter.conf
kbd is ok but lang selection without any function
So I assume that you altered the manipulation of locales and keyboard to fit the raspi-config.
BUT: uder these circumstances I'am not able to implement a lang and keyboard chooser at login and
therefore internatioalization is not really possible.

I am sure you are aware of the problem and you arer working on it.
Q: can you give me the timeframe when you solve the lang/kbd selection and the sync problem?

tks
willi

pelwell, 2017-12-08:
This repo is for the RPi Linux kernel, and your issue isn't a kernel problem.

@spl237 This is deep in your territory - where, if anywhere, should willi report this?

spl237, 2017-12-08:
We do not use the lightdm-gtk-greeter any more; we have a cut-down greeter called pi-greeter. It does not offer any of the taskbar or internationalisation features, as they clutter the display and are not fundamentally necessary to enable a user name and password to be entered.

If you wish to restore the original greeter functionality, update /etc/lightdm/lightdm.conf and change the "greeter-session" line to "lightdm-gtk-greeter" instead of "pi-greeter". Any issues with lightdm-gtk-greeter should be reported upstream to the Debian or LightDM maintainers, as we have not changed its code at all.

moserwi, 2017-12-08:
Thank you for info.

brgds willi

----- Message from Simon Long <notifications@github.com> ---------
    Date: Fri, 08 Dec 2017 20:53:14 +0000 (UTC)

<reply@reply.github.com>
      To: raspberrypi/linux <linux@noreply.github.com>
      Cc: moserwi <wilhelm.moser@demos.at>, Author <author@noreply.github.com>

----- End message from Simon Long <notifications@github.com> -----

Links:
------
[1] https://github.com/raspberrypi/linux/issues/2301#issuecomment-350369140
[2]  
  Kontaktdaten:
Wilhelm MOSER

Gentzgasse 65/13
A-1180 WIEN
mobile: +4369919294126
OE1WMR

http://www.moser-willi.at
http://www.net.co.at
http://www.demos.at

HINWEIS: Die in dieser Nachricht enthaltenen Informationen, Aussagen
und/oder Erklärungen etc. dienen ausschliesslich der Information des/der
oben bezeichneten Adressaten. Unter keinen Umständen sind in dieser
Nachricht enthaltene Informationen, Aussagen und/oder Erklärungen etc.
rechtlich verbindlich, noch stellen diese eine in irgendeiner Weise
rechtlich beachtliche Handlung des Absenders der Nachricht dar.

PLEASE NOTE: The information contained in this message, including any
statements and/or declarations etc. made therein, serve as information
intended exclusively for the above recipient(s). Under no circumstances is
any of the information, statements and/or declarations made therein legally
binding, nor does it represent any legal obligation on the sender of this
message.

closed 2017-12-08 by pelwell