raspberrypi/linux #2303: ath9k_htc doesn't work with aarch64 kernel

Zenitur, 2017-12-11:
I just built 64-bit Linux kernel for Raspbian Linux using this manual: http://www.tal.org/tutorials/raspberry-pi3-build-64-bit-kernel

I have hostapd in my runlevels, it starts AP on ath9k_htc, it's working one minute, then stop to work without any errors.

[dmesg on 64-bit](https://pastebin.com/QcieV54V), [dmesg on 32-bit](https://pastebin.com/vL0nGBKa)

pelwell, 2017-12-11:
If you get the same behaviour on a 32-bit kernel then you have a slim chance of some support, but using host AP on a random WiFi device you are almost certainly better off asking in the Forums.

lategoodbye, 2017-12-22:
@Zenitur In case this issue is reproducible with a least Linux 4.14 it should be reported to linux-wireless mailing list and fixed upstream.

https://wireless.wiki.kernel.org/en/users/drivers/ath9k_htc

closed 2018-06-27 by JamesH65