raspberrypi/linux #2307: Kernel Panic on 4.9.x with Atheros AR9271

cyablo, 2017-12-14:
Hi,

I have 2 Atheros AR9271 which I use in monitor mode. I get regular kernel panic's on stretch stock and latest hexxeh kernels. Sadly there are no entrys in syslog, so I had to take a photo of my screen:

[image]

rodizio1, 2017-12-15:
I know these cards work (only up to two, the third only works with nothing else accessing the usb bus, i.e. with ethernet plugged in it doesn't work anymore) with kernels 4.9.28 and 4.9.35 included in Raspian 2017-06-21 and 2017-07-05, maybe that narrows it down a little for you.

Another thing I noticed is, they don't like to be switched into promiscous mode, sometimes they just stop working then (but the system stays running), I'd try not doing that (in monitor mode, all frames are forwarded to userspace anyway, promisc mode is not needed).

cyablo, 2017-12-15:
I already discovered the "2 card's only" problem, its also present on my Linux Mint System, I guess that's a firmware thing :) I don't use promiscous mode and in my case the system totally crashes. It happens completely random, sometimes a few hours, sometimes 1-3 days. The RasPi does nothing other than collecting probe requests, so it doesn't even do some TX, boots straight into a python script.

cyablo, 2018-02-23:
Bug still present in 4.14.20 Kernel :(

[image]

cyablo, 2018-03-20:
Still happening with newest Kernel 4.14. and Firmware (there were some updates lately).

> Mar 19 18:49:07 cybechelon kernel: [80168.171520] ------------[ cut here ]------------
Mar 19 18:49:07 cybechelon kernel: [80168.172741] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:629 ieee80211_rx_napi+0x80c/0x9a4 [mac80211]
Mar 19 18:49:07 cybechelon kernel: [80168.172873] Modules linked in: binfmt_misc arc4 evdev ath9k_htc ath9k_common ath9k_hw ath mac80211 cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq uio fixed ip_tables x_tables ip
Mar 19 18:49:07 cybechelon kernel: [80168.173031] CPU: 0 PID: 0 Comm: swapper Tainted: G        WC      4.14.26+ #1099
Mar 19 18:49:07 cybechelon kernel: [80168.173037] Hardware name: BCM2835
Mar 19 18:49:07 cybechelon kernel: [80168.173152] [<c0016554>] (unwind_backtrace) from [<c0013e8c>] (show_stack+0x20/0x24)
Mar 19 18:49:07 cybechelon kernel: [80168.173188] [<c0013e8c>] (show_stack) from [<c061b0a8>] (dump_stack+0x20/0x28)
Mar 19 18:49:07 cybechelon kernel: [80168.173273] [<c061b0a8>] (dump_stack) from [<c0021fd4>] (__warn+0xe4/0x10c)
Mar 19 18:49:07 cybechelon kernel: [80168.173304] [<c0021fd4>] (__warn) from [<c00220c8>] (warn_slowpath_null+0x30/0x38)
Mar 19 18:49:07 cybechelon kernel: [80168.174466] [<c00220c8>] (warn_slowpath_null) from [<bf2b9c60>] (ieee80211_rx_napi+0x80c/0x9a4 [mac80211])
Mar 19 18:49:07 cybechelon kernel: [80168.175765] [<bf2b9c60>] (ieee80211_rx_napi [mac80211]) from [<bf549a90>] (ath9k_rx_tasklet+0x19c/0x22c [ath9k_htc])
Mar 19 18:49:07 cybechelon kernel: [80168.175982] [<bf549a90>] (ath9k_rx_tasklet [ath9k_htc]) from [<c0026bc4>] (tasklet_action+0x84/0xd8)
Mar 19 18:49:07 cybechelon kernel: [80168.176115] [<c0026bc4>] (tasklet_action) from [<c00095c4>] (__do_softirq+0x124/0x324)
Mar 19 18:49:07 cybechelon kernel: [80168.176143] [<c00095c4>] (__do_softirq) from [<c0026710>] (irq_exit+0xc0/0x120)
Mar 19 18:49:07 cybechelon kernel: [80168.176187] [<c0026710>] (irq_exit) from [<c0061680>] (__handle_domain_irq+0x60/0xb8)
Mar 19 18:49:07 cybechelon kernel: [80168.176269] [<c0061680>] (__handle_domain_irq) from [<c0009418>] (bcm2835_handle_irq+0x28/0x48)
Mar 19 18:49:07 cybechelon kernel: [80168.176312] [<c0009418>] (bcm2835_handle_irq) from [<c0634f1c>] (__irq_svc+0x5c/0x7c)
Mar 19 18:49:07 cybechelon kernel: [80168.176403] Exception stack(0xc091bef8 to 0xc091bf40)
Mar 19 18:49:07 cybechelon kernel: [80168.176416] bee0:                                                       00000000 00000000
Mar 19 18:49:07 cybechelon kernel: [80168.176485] bf00: ffffffff c091d414 c091a000 c091d0a4 c099f4e6 c091d020 c09ac180 c08f6a28
Mar 19 18:49:07 cybechelon kernel: [80168.176504] bf20: cefffae0 c091bf54 c091bf48 c091bf48 c00107dc c00107e0 60000013 ffffffff
Mar 19 18:49:07 cybechelon kernel: [80168.176553] [<c0634f1c>] (__irq_svc) from [<c00107e0>] (arch_cpu_idle+0x30/0x40)
Mar 19 18:49:07 cybechelon kernel: [80168.176652] [<c00107e0>] (arch_cpu_idle) from [<c0634dc4>] (default_idle_call+0x34/0x48)
Mar 19 18:49:07 cybechelon kernel: [80168.176681] [<c0634dc4>] (default_idle_call) from [<c0053bc4>] (do_idle+0x8c/0xec)
Mar 19 18:49:07 cybechelon kernel: [80168.176752] [<c0053bc4>] (do_idle) from [<c0053ea0>] (cpu_startup_entry+0x1c/0x20)
Mar 19 18:49:07 cybechelon kernel: [80168.176775] [<c0053ea0>] (cpu_startup_entry) from [<c062f82c>] (rest_init+0x7c/0x9c)
Mar 19 18:49:07 cybechelon kernel: [80168.176816] [<c062f82c>] (rest_init) from [<c08add48>] (start_kernel+0x358/0x3c8)
Mar 19 18:49:07 cybechelon kernel: [80168.176878] ---[ end trace e9f578a026c6678b ]---
Mar 19 18:49:07 cybechelon kernel: [80168.785874] ------------[ cut here ]------------
Mar 19 18:49:07 cybechelon kernel: [80168.786928] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:629 ieee80211_rx_napi+0x80c/0x9a4 [mac80211]
Mar 19 18:49:07 cybechelon kernel: [80168.787026] Modules linked in: binfmt_misc arc4 evdev ath9k_htc ath9k_common ath9k_hw ath mac80211 cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq uio fixed ip_tables x_tables ip
Mar 19 18:49:07 cybechelon kernel: [80168.787157] CPU: 0 PID: 0 Comm: swapper Tainted: G        WC      4.14.26+ #1099
Mar 19 18:49:07 cybechelon kernel: [80168.787161] Hardware name: BCM2835
Mar 19 18:49:07 cybechelon kernel: [80168.787263] [<c0016554>] (unwind_backtrace) from [<c0013e8c>] (show_stack+0x20/0x24)
Mar 19 18:49:07 cybechelon kernel: [80168.787289] [<c0013e8c>] (show_stack) from [<c061b0a8>] (dump_stack+0x20/0x28)
Mar 19 18:49:07 cybechelon kernel: [80168.787365] [<c061b0a8>] (dump_stack) from [<c0021fd4>] (__warn+0xe4/0x10c)
Mar 19 18:49:07 cybechelon kernel: [80168.787386] [<c0021fd4>] (__warn) from [<c00220c8>] (warn_slowpath_null+0x30/0x38)
Mar 19 18:49:07 cybechelon kernel: [80168.788321] [<c00220c8>] (warn_slowpath_null) from [<bf2b9c60>] (ieee80211_rx_napi+0x80c/0x9a4 [mac80211])
Mar 19 18:49:07 cybechelon kernel: [80168.789469] [<bf2b9c60>] (ieee80211_rx_napi [mac80211]) from [<bf549a90>] (ath9k_rx_tasklet+0x19c/0x22c [ath9k_htc])
Mar 19 18:49:07 cybechelon kernel: [80168.789683] [<bf549a90>] (ath9k_rx_tasklet [ath9k_htc]) from [<c0026bc4>] (tasklet_action+0x84/0xd8)
Mar 19 18:49:07 cybechelon kernel: [80168.789762] [<c0026bc4>] (tasklet_action) from [<c00095c4>] (__do_softirq+0x124/0x324)
Mar 19 18:49:07 cybechelon kernel: [80168.789783] [<c00095c4>] (__do_softirq) from [<c0026710>] (irq_exit+0xc0/0x120)
[... cut ...]
Mar 19 18:49:07 cybechelon kernel: [80168.790146] [<c0634dc4>] (default_idle_call) from [<c0053bc4>] (do_idle+0x8c/0xec)
Mar 19 18:49:07 cybechelon kernel: [80168.790163] [<c0053bc4>] (do_idle) from [<c0053ea0>] (cpu_startup_entry+0x1c/0x20)
Mar 19 18:49:07 cybechelon kernel: [80168.790182] [<c0053ea0>] (cpu_startup_entry) from [<c062f82c>] (rest_init+0x7c/0x9c)
Mar 19 18:49:07 cybechelon kernel: [80168.790259] [<c062f82c>] (rest_init) from [<c08add48>] (start_kernel+0x358/0x3c8)
Mar 19 18:49:07 cybechelon kernel: [80168.790273] ---[ end trace e9f578a026c6678c ]---

cyablo, 2018-04-20:
In fact, it's not only one adapter. I have 3 of them from different brands. They all show the same problem. What's really interesting: If you plug in 2 adapters, the kernel module/driver can crash independently, leaving one still working while the other is not. If I let it run long enough, the other one will crash too, resulting in a complete freeze of the RasPi. Sometimes even with colorful graphic glitches on the screen.

Never really tried using them as a Client for a longer period because the RasPi is hardwired.

cyablo, 2018-04-20:
The Chipset is neither very old nor random. They all use Atheros AR9271 chipset. they are the most common insisde 150 Mbit 2.4 GHz adapters. I use TP-LINK TL-WN722N (only v1 has Atheros, v2 has Realtek), ALFA Network AWUS036NHA and Sophos AP5

macmpi, 2018-04-21:
Have you tried using them through a powered USB hub?
They might draw too much current out of USB port and create such kind of crashes (seen that with few wifi dongles).

cyablo, 2018-04-21:
I'm Always using an powered USB Hub. In this Case a Ugreen 4 Port powered by a Huawei USB charger.

melikyuksel, 2018-05-15:
I'm also having this problem on a Pi Zero.

```
[41660.555778] ------------[ cut here ]------------
[41660.556682] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:629 ieee80211_rx_napi+0x80c/0x9a4 [mac80211]
[41660.556781] Modules linked in: arc4 ath9k_htc ath9k_common ath9k_hw ath asix mac80211 libphy cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[41660.556912] CPU: 0 PID: 0 Comm: swapper Tainted: G         C      4.14.39+ #1112
[41660.556917] Hardware name: BCM2835
[41660.556969] [<c0016514>] (unwind_backtrace) from [<c0013e4c>] (show_stack+0x20/0x24)
[41660.557050] [<c0013e4c>] (show_stack) from [<c0630c04>] (dump_stack+0x20/0x28)
[41660.557077] [<c0630c04>] (dump_stack) from [<c0021f94>] (__warn+0xe4/0x10c)
[41660.557096] [<c0021f94>] (__warn) from [<c0022088>] (warn_slowpath_null+0x30/0x38)
[41660.557964] [<c0022088>] (warn_slowpath_null) from [<bf2cccdc>] (ieee80211_rx_napi+0x80c/0x9a4 [mac80211])
[41660.558932] [<bf2cccdc>] (ieee80211_rx_napi [mac80211]) from [<bf568a90>] (ath9k_rx_tasklet+0x19c/0x22c [ath9k_htc])
[41660.559164] [<bf568a90>] (ath9k_rx_tasklet [ath9k_htc]) from [<c0026b84>] (tasklet_action+0x84/0xd8)
[41660.559251] [<c0026b84>] (tasklet_action) from [<c00095c4>] (__do_softirq+0x124/0x324)
[41660.559270] [<c00095c4>] (__do_softirq) from [<c00266d0>] (irq_exit+0xc0/0x120)
[41660.559299] [<c00266d0>] (irq_exit) from [<c00616b4>] (__handle_domain_irq+0x60/0xb8)
[41660.559316] [<c00616b4>] (__handle_domain_irq) from [<c0009418>] (bcm2835_handle_irq+0x28/0x48)
[41660.559331] [<c0009418>] (bcm2835_handle_irq) from [<c064aa5c>] (__irq_svc+0x5c/0x7c)
[41660.559340] Exception stack(0xc0937ef8 to 0xc0937f40)
[41660.559392] 7ee0:                                                       00000000 00000000
[41660.559406] 7f00: ffffffff c0939414 c0936000 c09390a4 c09c1766 c0939020 c09ce7a0 c0912a28
[41660.559417] 7f20: dbfffa80 c0937f54 c0937f48 c0937f48 c001079c c00107a0 60000013 ffffffff
[41660.559442] [<c064aa5c>] (__irq_svc) from [<c00107a0>] (arch_cpu_idle+0x30/0x40)
[41660.559459] [<c00107a0>] (arch_cpu_idle) from [<c064a90c>] (default_idle_call+0x34/0x48)
[41660.559496] [<c064a90c>] (default_idle_call) from [<c0053bf8>] (do_idle+0x8c/0xec)
[41660.559512] [<c0053bf8>] (do_idle) from [<c0053ed4>] (cpu_startup_entry+0x1c/0x20)
[41660.559533] [<c0053ed4>] (cpu_startup_entry) from [<c0645374>] (rest_init+0x7c/0x9c)
[41660.559562] [<c0645374>] (rest_init) from [<c08c9d48>] (start_kernel+0x358/0x3c8)
[41660.559572] ---[ end trace 4d345c866db60e06 ]---
```

severach, 2018-05-19:
I have this problem on Manjaro. This computer has a wired connection so wifi isn't used. The wifi was turned on so it is constantly browsing available access points for the week it's up. dmesg is spammed with 'link is not ready' messages.

Now I have Cinnamon disable the wifi so no more spamming dmesg.

````
$ uname -a
Linux specialc2 4.14.39-1-MANJARO #1 SMP PREEMPT Wed May 2 19:03:39 UTC 2018 x86_64 GNU/Linux

$ dmesg | grep -i Inspiron
[    0.000000] DMI: Dell Inc. Inspiron One 2020/0YXG0N           , BIOS A08 03/13/2018

$ lspci | grep -ie Atheros
02:00.0 Network controller: Qualcomm Atheros AR9485 Wireless Network Adapter (rev 01)

$ dmesg | grep -i -Fe $'atheros\nwlp' | grep -vFe 'link is not ready'
[   11.717927] ieee80211 phy0: Atheros AR9485 Rev:1 mem=0xffffaf4c01680000, irq=18
[   11.763542] ath9k 0000:02:00.0 wlp2s0: renamed from wlan0

$ dmesg | grep -Fe 'link is not ready' | wc -l
1924

$ dmesg ...
[56434.845183] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
[56749.860791] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
[57064.836518] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
[57065.407605] ------------[ cut here ]------------
[57065.407643] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:629 ieee80211_rx_napi+0x886/0xa00 [mac80211]
[57065.407644] Modules linked in: fuse arc4 ath9k ath9k_common iTCO_wdt ath9k_hw ath mac80211 iTCO_vendor_support cfg80211 r8168(O) dcdbas intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel mousedev uvcvideo kvm irqbypass videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 crct10dif_pclmul videobuf2_core crc32_pclmul snd_hda_codec_realtek snd_hda_codec_generic ghash_clmulni_intel ath3k btusb btrtl evdev btbcm input_leds btintel cryptd i915 intel_cstate intel_rapl_perf videodev pcspkr media led_class mac_hid bluetooth ecdh_generic rfkill dell_smm_hwmon i2c_algo_bit snd_hda_intel drm_kms_helper snd_hda_codec ir_rc6_decoder mei_me i2c_i801 rc_rc6_mce snd_hda_core snd_hwdep snd_pcm snd_timer snd fan mei soundcore ir_lirc_codec lirc_dev thermal drm intel_gtt agpgart syscopyarea sysfillrect
[57065.407703]  sysimgblt fb_sys_fops ite_cir rc_core lpc_ich video shpchp button ie31200_edac sch_fq_codel ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 fscrypto sr_mod cdrom sd_mod hid_generic usbhid hid ata_generic pata_acpi ata_piix crc32c_intel ehci_pci libata ehci_hcd scsi_mod usbcore usb_common
[57065.407733] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O    4.14.39-1-MANJARO #1
[57065.407734] Hardware name: Dell Inc. Inspiron One 2020/0YXG0N           , BIOS A08 03/13/2018
[57065.407736] task: ffffffffab012480 task.stack: ffffffffab000000
[57065.407756] RIP: 0010:ieee80211_rx_napi+0x886/0xa00 [mac80211]
[57065.407758] RSP: 0018:ffff99465f203d50 EFLAGS: 00010246
[57065.407760] RAX: 0000000000010000 RBX: ffff9946581bfd00 RCX: 0000000000000002
[57065.407761] RDX: 0000000000000004 RSI: 0000000000000000 RDI: ffff994650f10780
[57065.407763] RBP: ffff994650f10780 R08: 0000000000000000 R09: 0000000000000000
[57065.407764] R10: 0000000080000000 R11: 0000000000000000 R12: 0000000000000000
[57065.407765] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000004
[57065.407767] FS:  0000000000000000(0000) GS:ffff99465f200000(0000) knlGS:0000000000000000
[57065.407769] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[57065.407770] CR2: 00007fa0deb57930 CR3: 000000010800a001 CR4: 00000000001606f0
[57065.407772] Call Trace:
[57065.407776]  <IRQ>
[57065.407785]  ? ath_tx_process_buffer+0x260/0xcb0 [ath9k]
[57065.407790]  ? kmem_cache_alloc+0x156/0x1a0
[57065.407795]  ? __build_skb+0x25/0xe0
[57065.407798]  ? __netdev_alloc_skb+0xc2/0x110
[57065.407802]  ? ath9k_cmn_rx_skb_postprocess+0x46/0x130 [ath9k_common]
[57065.407808]  ath_rx_tasklet+0x9f2/0xec0 [ath9k]
[57065.407815]  ath9k_tasklet+0x15a/0x220 [ath9k]
[57065.407819]  tasklet_action+0x5f/0x120
[57065.407823]  __do_softirq+0xde/0x2e5
[57065.407826]  irq_exit+0xae/0xb0
[57065.407828]  do_IRQ+0x81/0xd0
[57065.407833]  common_interrupt+0x7d/0x7d
[57065.407835]  </IRQ>
[57065.407839] RIP: 0010:cpuidle_enter_state+0xa2/0x310
[57065.407840] RSP: 0018:ffffffffab003e90 EFLAGS: 00000246 ORIG_RAX: ffffffffffffff8e
[57065.407843] RAX: ffff99465f21f980 RBX: 000033e6935861c5 RCX: 000000000000001f
[57065.407844] RDX: 000033e6935861c5 RSI: fffffffce4a94b84 RDI: 0000000000000000
[57065.407845] RBP: ffff99465f227b28 R08: 00000000000019b1 R09: 000000000000fc18
[57065.407847] R10: ffffffffab003e70 R11: 0000000000000cb1 R12: 0000000000000004
[57065.407848] R13: ffffffffab0af3f8 R14: 0000000000000000 R15: 000033e6908d96f8
[57065.407854]  do_idle+0x179/0x1d0
[57065.407857]  cpu_startup_entry+0x6f/0x80
[57065.407861]  start_kernel+0x4b5/0x4d5
[57065.407866]  secondary_startup_64+0xa5/0xb0
[57065.407868] Code: c6 05 2d 97 08 00 01 e8 34 e4 57 e9 0f 0b e9 59 f8 ff ff 48 89 de 48 89 ef 44 89 04 24 e8 53 25 fe ff 44 8b 04 24 e9 29 fa ff ff <0f> 0b 48 89 df e8 60 40 9f e9 e9 cf fa ff ff 31 d2 b9 20 00 08 
[57065.407912] ---[ end trace 1d0e784c914d9f62 ]---
[57379.828775] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
[57694.830521] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
...
[250413.789660] ------------[ cut here ]------------
(same crash as above)
...
[335778.498921] ------------[ cut here ]------------
...
[604224.096345] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
...end
````

cyablo, 2018-05-20:
I switched from Atheros to Ralink, problem is gone. So it's def. a firmware/driver problem.

lategoodbye, 2018-05-21:
I think you should try Linux 4.16 (using mainline instead of the foundation kernel would be the best) and if the issue still occurs report it to linux-wireless mailing list.

severach, 2018-06-02:
I ran Manjaro on 4.16 for a week with Atheros disabled and everything was fine. I ran it another week with Atheros enabled but not connected. There was no 'cut here' message but the GUI was sluggish, same freezing browsers problem as had been happening with 4.9 and 4.14. Seems that it has been improved enough to eliminate the 'cut here' message but the problem has not been fixed.

JamesH65, 2018-06-27:
It looks like the 4.16 kernel fixes this. I doubt we will backport but if anyone can point to the change that fixes it, we might be able to cherry-pick.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

JamesH65, 2018-08-31:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-08-31 by JamesH65

gourry, 2018-09-02:
Hello, I'm experiencing the same issue with a Raspberry Pi 3 B+ using a very common Atheros USB card ( ALFA AWUS036NHA ).
I did rpi-update of the kernel today but the issue is not fixed.
`uname -a
Linux raspberrypi 4.14.67-v7+ #1139 SMP Wed Aug 29 15:17:05 BST 2018 armv7l GNU/Linux
`

[11274.305757] ------------[ cut here ]------------
[11274.308588] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:629 ieee80211_rx_napi+0x868/0x9dc [mac80211]
[11274.313959] Modules linked in: cmac sha256_generic md4 md5 hmac nls_utf8 cifs ccm arc4 ath9k_htc ath9k_common ath9k_hw ath mac80211 appletalk psnap llc ax25 xt_conntrack iptable_filter ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack brcmfmac brcmutil cfg80211 rfkill i2c_bcm2835 uio_pdrv_genirq fixed uio i2c_dev i2c_bcm2708 snd_bcm2835(C) snd_pcm snd_timer snd ip_tables x_tables ipv6
[11274.329319] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        WC      4.14.67-v7+ #1139
[11274.334187] Hardware name: BCM2835
[11274.336565] [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
[11274.341192] [<8010c240>] (show_stack) from [<80787384>] (dump_stack+0xd4/0x118)
[11274.343554] [<80787384>] (dump_stack) from [<8011da6c>] (__warn+0xf8/0x110)
[11274.345889] [<8011da6c>] (__warn) from [<8011db54>] (warn_slowpath_null+0x30/0x38)
[11274.350610] [<8011db54>] (warn_slowpath_null) from [<7f5f8d10>] (ieee80211_rx_napi+0x868/0x9dc [mac80211])
[11274.355400] [<7f5f8d10>] (ieee80211_rx_napi [mac80211]) from [<7f735894>] (ath9k_rx_tasklet+0x160/0x18c [ath9k_htc])
[11274.360099] [<7f735894>] (ath9k_rx_tasklet [ath9k_htc]) from [<80123d34>] (tasklet_action+0x74/0x10c)
[11274.364714] [<80123d34>] (tasklet_action) from [<80101694>] (__do_softirq+0x18c/0x3d8)
[11274.369398] [<80101694>] (__do_softirq) from [<80123808>] (irq_exit+0x108/0x164)
[11274.374202] [<80123808>] (irq_exit) from [<801756d8>] (__handle_domain_irq+0x70/0xc4)
[11274.379171] [<801756d8>] (__handle_domain_irq) from [<80101504>] (bcm2836_arm_irqchip_handle_irq+0xa8/0xac)
[11274.384327] [<80101504>] (bcm2836_arm_irqchip_handle_irq) from [<807a2f7c>] (__irq_svc+0x5c/0x7c)
[11274.389448] Exception stack(0x80c01ef0 to 0x80c01f38)
[11274.392002] 1ee0:                                     00000000 2d2550dc 3da17000 00000000
[11274.397162] 1f00: 80c00000 80c03dcc 80c03d68 80c882b2 00000001 80b60a30 be5ff9c0 80c01f4c
[11274.402412] 1f20: 80c04174 80c01f40 80108a4c 80108a50 60000013 ffffffff
[11274.405054] [<807a2f7c>] (__irq_svc) from [<80108a50>] (arch_cpu_idle+0x34/0x4c)
[11274.410267] [<80108a50>] (arch_cpu_idle) from [<807a26c4>] (default_idle_call+0x34/0x48)
[11274.415593] [<807a26c4>] (default_idle_call) from [<80161350>] (do_idle+0xd8/0x150)
[11274.420975] [<80161350>] (do_idle) from [<80161664>] (cpu_startup_entry+0x28/0x2c)
[11274.426415] [<80161664>] (cpu_startup_entry) from [<8079c41c>] (rest_init+0xbc/0xc0)
[11274.431960] [<8079c41c>] (rest_init) from [<80b00df8>] (start_kernel+0x3d4/0x3e0)
[11274.437562] ---[ end trace c0f3f54ed3d968d2 ]---