raspberrypi/linux #2308: Installing digiKam package results in audio sync issues in Chromium browser

rsfitz, 2017-12-15:
**Description:**
Installing the currently published digiKam package will result in audio sync issues when using Chromium browser to play YouTube videos.

**Build:**
Raspbian Stretch - November 2017 - Kernel 4.9

**Steps To Reproduce:**
1. Install digiKam (sudo apt-get install digikam)
2. Reboot pi, launch Chromium play YouTube video

XECDesign, 2017-12-19:
digikam drags in 372 dependencies (818MB). Which of those is responsible? Do you see the same behaviour if you install just pulseaudio? If not, you might have to narrow it down and report the issue here: https://github.com/RPi-Distro/repo/issues/ or to the upstream maintainers of the package.

rsfitz, 2017-12-25:
The same symptoms do occur when installing _only_  the pulseaudio package from the repository. When this package is removed audio sync returns to normal. 

I will report this to RPi-Distro/repo/issue if this is more appropriate.

XECDesign, 2017-12-27:
It might be best reported upstream to pulseaudio devs, since they'd be best qualified to address it. We don't modify pulseaudio in any way and we don't ship it in our Raspbian image.

closed 2018-08-31 by JamesH65