raspberrypi/linux #2309: USB bus getting reset

carpa5, 2017-12-19:
Hi I am running an application that needs to have 6 RS232 ports open simultaneously. 
If I disable one port and only use 5, then there are no issues. All problems occur if 6 ports are open.

The hardware is as follows:

Raspberry Pi Details:
  Type: Pi 3, Revision: 02, Memory: 1024MB, Maker: Embest
  * Device tree is enabled.
  *--> Raspberry Pi 3 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.

These ports are via USB FTDI to RS232:

root@agrs:~# lsusb
Bus 001 Device 005: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 001 Device 004: ID 0403:6011 Future Technology Devices International, Ltd FT4232H Quad HS USB-UART/FIFO IC
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@agrs:~# ls /dev/ttyU*
/dev/ttyUSB0  /dev/ttyUSB2  /dev/ttyUSB4  /dev/ttyUSB6
/dev/ttyUSB1  /dev/ttyUSB3  /dev/ttyUSB5  /dev/ttyUSB7

I am running Stretch and I have tested this with the following kernels:
4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
4.9.68-v7+ #1064 SMP Tue Dec 12 15:08:24 GMT 2017 armv7l GNU/Linux (via rpi-update)

On both kernels, after about 6 hours (it is random), we get a USB bus reset (there are no errors in /vas/log/syslog):

Dec 11 20:46:53 localhost dhclient[561]: receive_packet failed on eth0: Network is down
Dec 11 20:46:53 localhost kernel: [19192.911613] usb 1-1.1: USB disconnect, device number 3
Dec 11 20:46:53 localhost kernel: [19192.912150] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Dec 11 20:46:53 localhost kernel: [19192.912336] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Dec 11 20:46:53 localhost systemd[1]: Stopping ifup for eth0...
Dec 11 20:46:53 localhost kernel: [19192.952175] usb 1-1.2: USB disconnect, device number 4
Dec 11 20:46:53 localhost kernel: [19192.952805] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
Dec 11 20:46:53 localhost kernel: [19192.952842] ftdi_sio 1-1.2:1.0: device disconnected
Dec 11 20:46:53 localhost kernel: [19192.953326] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
Dec 11 20:46:53 localhost kernel: [19192.953380] ftdi_sio 1-1.2:1.1: device disconnected
Dec 11 20:46:53 localhost kernel: [19192.960550] ftdi_sio ttyUSB2: FTDI USB Serial Device converter now disconnected from ttyUSB2
Dec 11 20:46:53 localhost kernel: [19192.960717] ftdi_sio 1-1.2:1.2: device disconnected
Dec 11 20:46:53 localhost kernel: [19192.961703] ftdi_sio ttyUSB3: FTDI USB Serial Device converter now disconnected from ttyUSB3
Dec 11 20:46:53 localhost kernel: [19192.961742] ftdi_sio 1-1.2:1.3: device disconnected
Dec 11 20:46:53 localhost kernel: [19192.962360] usb 1-1.3: USB disconnect, device number 5
Dec 11 20:46:53 localhost kernel: [19192.963195] ftdi_sio ttyUSB4: FTDI USB Serial Device converter now disconnected from ttyUSB4
Dec 11 20:46:53 localhost kernel: [19192.963231] ftdi_sio 1-1.3:1.0: device disconnected
Dec 11 20:46:53 localhost kernel: [19192.963941] ftdi_sio ttyUSB5: FTDI USB Serial Device converter now disconnected from ttyUSB5
Dec 11 20:46:53 localhost kernel: [19192.963997] ftdi_sio 1-1.3:1.1: device disconnected
Dec 11 20:46:53 localhost kernel: [19192.964642] ftdi_sio ttyUSB6: FTDI USB Serial Device converter now disconnected from ttyUSB6
Dec 11 20:46:53 localhost kernel: [19192.964675] ftdi_sio 1-1.3:1.2: device disconnected
Dec 11 20:46:53 localhost kernel: [19192.965165] ftdi_sio ttyUSB7: FTDI USB Serial Device converter now disconnected from ttyUSB7
Dec 11 20:46:53 localhost kernel: [19192.965233] ftdi_sio 1-1.3:1.3: device disconnected
Dec 11 20:46:53 localhost kernel: [19192.965751] Indeed it is in host mode hprt0 = 00001101
Dec 11 20:46:53 localhost dhclient[4297]: Killed old client process
Dec 11 20:46:53 localhost ifdown[4279]: Killed old client process

After this the USB/RS232 ports get re enumerated:
root@agrs:~# ls /dev/ttyU*
/dev/ttyUSB1   /dev/ttyUSB11  /dev/ttyUSB13  /dev/ttyUSB8
/dev/ttyUSB10  /dev/ttyUSB12  /dev/ttyUSB7   /dev/ttyUSB9

The only ports that do not change are the ones that were not open (USB1 and USB7).
My /boot/cmdline.txt is:
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait consoleblank=0 ipv6.disable=1

The /boot/config.txt is:
#hdmi_safe=1
#disable_overscan=1
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16
#framebuffer_width=1280
#framebuffer_height=720
#hdmi_force_hotplug=1
#hdmi_group=1
#hdmi_mode=1
#hdmi_drive=2
#config_hdmi_boost=4
#sdtv_mode=2
#arm_freq=800
gpu_mem=16
#w1-gpio
dtoverlay=w1-gpio-pullup,gpiopin=7
dtparam=audio=on
dtoverlay=pi3-disable-bt
max_usb_current=1

So far I have tried the following, but nothing has helped:

- Replaced the RAS PI 3 and the FTDI boards
- Tried using the dwc2 overlay in config.txt:
dtoverlay=dwc2 dr_mode=host
- Added the following to /boot/cmdline.txt
dwc_otg.fiq_fix_enable=0 smsc95xx.turbo_mode=N ......

I am now at a loss. Again If I just lower the number of ports the are open from 6 to 5 there are no problems.

Thank you

Richard Pavlik

P33M, 2017-12-19:
Duplicate of https://github.com/raspberrypi/linux/issues/2050 which documents what happens when you have multiple high-speed endpoints active that don't return data for extended periods of time.

There is no easy workaround for this issue. It is a function of the host hardware.

Ironically, if you switch to full-speed FTDI single-port UART chips (eg FT232R) then you won't have this issue. What baudrate are you using with the UARTs?

carpa5, 2017-12-19:
USB1 is 19200
The rest are: 921600

I tired lowering to 460800 but that did not help.

Each port has a dedicated thread. The high baud rate devices are read in parallel. I also tried reading them sequentially, but that also did not help.

The sample rate is 1 Hz. USB1 is a GPS which uses PPS on DCD for timing.

carpa5, 2017-12-19:
USB1 should be USB0 in above (GPS connection)

JamesH65, 2018-06-27:
Closing as duplicate of #2050

closed 2018-06-27 by JamesH65