raspberrypi/linux #2311: MCP23s17 gpio expander troubles in linux 4.14

dykon, 2017-12-21:
Hello, I am trying to use MCP23s17 with my Raspberry pi 3 linux 4.14. I use a default Overlay.

[https://github.com/raspberrypi/linux/blob/6a2927dd95b32c92b0badfaf0d18efb20c8a50c5/arch/arm/boot/dts/overlays/mcp23s17-overlay.dts](url)

my config.txt:

`dtoverlay=spi1-2cs
dtoverlay=mcp23s17,s17-spi1-0-present=1,s17-spi1-0-int-gpio=24`

I get this massage in
`sudo vcdbg log msg`
`001898.409: brfs: File read: /mfs/sd/overlays/mcp23s17.dtbo
002044.836: Loaded overlay 'mcp23s17'
002044.856: dtparam: s17-spi1-0-present=1
002052.266: dtparam: s17-spi1-0-int-gpio=24
003313.325: Device tree loaded to 0x2eff9800 (size 0x6702)`
so my Overlay is loaded and i can use it, but in /dev I get

`crw-rw---- 1 root gpio    254,   0 Dec 21 11:00 gpiochip0
crw-rw---- 1 root gpio    254,   1 Dec 21 11:00 gpiochip1
crw-rw---- 1 root gpio    254,   2 Dec 21 11:00 gpiochip2
crw-rw---- 1 root gpio    248,   0 Dec 21 11:00 gpiomem`

no gpiochip 3. This problem appeared in Linux 4.14 in linux 4.09 everything was perfect.
What is the solution for this problem?
Thanks a lot.

pelwell, 2017-12-21:
That's odd. Using the same config.txt settings I do get gpiochip3, which was mildly surprising since I don't have an mcp23s17 attached.

Can you run `sudo rpi-update` to make sure you are on the same kernel as me (`Linux raspberrypi 4.9.70-v7+ #1068 SMP Mon Dec 18 22:12:55 GMT 2017 armv7l GNU/Linux`), then retry?

If you still don't get gpiochip3, run these commands and show me your output (I include mine for comparison):
```
pi@raspberrypi:~$ grep -d skip . /sys/class/gpio/gpiochip*/*
/sys/class/gpio/gpiochip0/base:0
/sys/class/gpio/gpiochip0/label:pinctrl-bcm2835
/sys/class/gpio/gpiochip0/ngpio:54
/sys/class/gpio/gpiochip100/base:100
/sys/class/gpio/gpiochip100/label:brcmvirt-gpio
/sys/class/gpio/gpiochip100/ngpio:2
/sys/class/gpio/gpiochip128/base:128
/sys/class/gpio/gpiochip128/label:brcmexp-gpio
/sys/class/gpio/gpiochip128/ngpio:8
/sys/class/gpio/gpiochip496/base:496
/sys/class/gpio/gpiochip496/label:mcp23s17
/sys/class/gpio/gpiochip496/ngpio:16

pi@raspberrypi:~$ lsmod
Module                  Size  Used by
rfcomm                 37723  6
cmac                    3239  1
bnep                   12051  2
hci_uart               20020  1
btbcm                   7916  1 hci_uart
bluetooth             365844  29 hci_uart,bnep,btbcm,rfcomm
gpio_mcp23s08          12526  0
spidev                  7373  0
brcmfmac              292847  0
brcmutil                9863  1 brcmfmac
cfg80211              544609  1 brcmfmac
rfkill                 20851  6 bluetooth,cfg80211
snd_bcm2835            24427  0
spi_bcm2835             7596  0
snd_pcm                98565  1 snd_bcm2835
spi_bcm2835aux          5326  0
snd_timer              23968  1 snd_pcm
snd                    70032  3 snd_timer,snd_bcm2835,snd_pcm
bcm2835_gpiomem         3940  0
uio_pdrv_genirq         3923  0
uio                    10204  1 uio_pdrv_genirq
fixed                   3285  0
i2c_dev                 6913  0
ip_tables              13161  0
x_tables               20578  1 ip_tables
ipv6                  409066  24
```

dykon, 2017-12-21:
Hallo thank you for answer, I am using now 4.14 Kernel

`Linux pi002 4.14.7-v7+ #1067 SMP Mon Dec 18 18:51:12 GMT 2017 armv7l GNU/Linux`

in kernel 4.09 it workes.
 I get.

> /sys/class/gpio/gpiochip0/base:0
/sys/class/gpio/gpiochip0/label:pinctrl-bcm2835
/sys/class/gpio/gpiochip0/ngpio:54
/sys/class/gpio/gpiochip100/base:100
/sys/class/gpio/gpiochip100/label:brcmvirt-gpio
/sys/class/gpio/gpiochip100/ngpio:2
/sys/class/gpio/gpiochip128/base:128
/sys/class/gpio/gpiochip128/label:brcmexp-gpio
/sys/class/gpio/gpiochip128/ngpio:8

and

> Module                  Size  Used by
cmac                   16384  1
bnep                   20480  2
hci_uart               24576  1
bluetooth             368640  23 hci_uart,bnep
ecdh_generic           28672  1 bluetooth
mcp251x                20480  0
can_dev                24576  1 mcp251x
spidev                 16384  0
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              573440  1 brcmfmac
rfkill                 28672  6 bluetooth,cfg80211
snd_bcm2835            32768  0
snd_pcm                98304  1 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    69632  3 snd_timer,snd_bcm2835,snd_pcm
spi_bcm2835aux         16384  0
spi_bcm2835            16384  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
ip_tables              24576  0
x_tables               28672  1 ip_tables
ipv6                  434176  28

it is not appears hear.

pelwell, 2017-12-21:
Are you using another overlay as well, one that would be loading the mcp251x driver?

pelwell, 2017-12-21:
There are a few more things you can try:
```
$ sudo modprobe gpio_mcp23s08
$ lsmod | grep gpio
$ dmesg -l err
```

dykon, 2017-12-22:
I found that mcp23s08 was moved to pinctrl:
https://github.com/raspberrypi/linux/commit/64ac43e6fa283f556f0a8f69ad52d6e7d550307d#diff-47b124fe5924abd771eb72d64edf7d12

I tried to manually compile, but the compilation showed a warning.
```
pi@pi002:~/linux $ make SUBDIRS=drivers/pinctrl modules
  CC [M]  drivers/pinctrl/pinctrl-mcp23s08.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "pinconf_generic_dt_free_map" [drivers/pinctrl/pinctrl-mcp23s08.ko] undefined!
WARNING: "pinconf_generic_dt_node_to_map" [drivers/pinctrl/pinctrl-mcp23s08.ko] undefined!
  CC      drivers/pinctrl/pinctrl-mcp23s08.mod.o
  LD [M]  drivers/pinctrl/pinctrl-mcp23s08.ko
```

This warning was later also shown in dmesg (after reboot)
```
pi@pi002:~/linux $ dmesg | grep s08
[    3.845515] pinctrl_mcp23s08: loading out-of-tree module taints kernel.
[    3.845782] pinctrl_mcp23s08: Unknown symbol pinconf_generic_dt_node_to_map (err 0)
[    3.845938] pinctrl_mcp23s08: Unknown symbol pinconf_generic_dt_free_map (err 0)
[    3.955677] pinctrl_mcp23s08: Unknown symbol pinconf_generic_dt_node_to_map (err 0)
[    3.955760] pinctrl_mcp23s08: Unknown symbol pinconf_generic_dt_free_map (err 0)
[    4.026695] pinctrl_mcp23s08: Unknown symbol pinconf_generic_dt_node_to_map (err 0)
[    4.026802] pinctrl_mcp23s08: Unknown symbol pinconf_generic_dt_free_map (err 0)
```

pelwell, 2017-12-22:
Why are you building the module yourself? It is included as standard in the Raspberry Pi firmware releases.

If you are building a custom kernel then you need to be honest and upfront about it, so that we can quickly get to the bottom of the problem.

pelwell, 2017-12-22:
P.S. 4.14 is currently the base for our BRANCH=next rpi-update releases.

pelwell, 2017-12-22:
...and it appears not to contain the driver. I'm away from the keyboard now, but leave it with me to investigate.

dykon, 2017-12-22:
Sorry for the confusion. I tested the hardware with kernel 4.9 (```4.9.70-v7+```) where it worked fine. Then I upgraded the kernel to 4.14 to test the driver with
```
BRANCH=next rpi-update
```

After upgrading, I opened this issue. I then researched and found that the driver was moved to pinctrl. I then tried to compile the corresponding module (```pinctrl-mcp23s08```) in order to resolve the issue. However, compiling only this module did not work.

pelwell, 2017-12-23:
CONFIG_PINCTRL_MCP23S08 has been added to the standard Raspberry Pi defconfigs, so the driver module will appear in the next firmware release.

dykon, 2018-02-27:
Thanks a lot, now its working es i have awaiting for.

closed 2018-02-27 by dykon