raspberrypi/linux #2312: rotary encoder overlay - add push button?

shawaj, 2017-12-23:
There is a rotary encoder overlay:

```
Name:   rotary-encoder
Info:   Overlay for GPIO connected rotary encoder.
Load:   dtoverlay=rotary-encoder,<param>=<val>
Params: rotary0_pin_a           GPIO connected to rotary encoder channel A
                                (default 4).
        rotary0_pin_b           GPIO connected to rotary encoder channel B
                                (default 17).
```

It would be nice to add a push button input to this for rotary encoders with a built in push button. Would this be feasible?

pelwell, 2017-12-24:
What effect would you want pushing the button to have - a key press? You can see an example of hooking a GPIO up to emulate a key in the gpio-shutdown overlay.

shawaj, 2017-12-25:
The specific thought I had in mind was the navigation of menus and changing of volume in audio and media center setups. But I guess this is context dependent, so perhaps isn't that useful for this generic overlay but should require a be underlay that's more specialised?

pelwell, 2017-12-27:
A generic `gpio-key` overlay would be useful, with parameters to select the label, key code and GPIO. Appending `@0` to the node name, adding a `reg` property and assigning the GPIO number to it will make it possible for multiple keys to be defined by multiple instances of the overlay.

If you want some examples, the gpio-key module is used by the [gpio-shutdown](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/gpio-shutdown-overlay.dts) and [tinylcd35](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/tinylcd35-overlay.dts) overlays.

shawaj, 2017-12-27:
so you mean something like the gpio-shutdown one, but with parameters / overrides to choose other functions / key assignments from the [gpio-keys](https://www.kernel.org/doc/Documentation/devicetree/bindings/input/gpio-keys.txt) bindings?

Might actually add this for easy config of buttons on our PaPiRus / Media Center HAT and other boards. Seems like a nice little way to do it :-)

pelwell, 2018-01-02:
> so you mean something like the gpio-shutdown one, but with parameters / overrides to choose other functions / key assignments from the gpio-keys bindings?

Yes, in use it would look something like `dtoverlay=gpio-key,gpio=24,keycode=59,label="KEY_F1"`.

shawaj, 2018-01-02:
@pelwell for the media-center HAT overlay, I am thinking to use a similar method to set up the joystick / buttons automatically to keys.

Is there a way to do this same method for multiple keys? Do you have an example overlay that does this for multiple keys?

pelwell, 2018-01-02:
Although I haven't tested it, I think we should be able to use the magic `reg` property support to allow multiple keys to be defined with multiple instances of the same overlay. I would prefer this than having a single overlay create multiple keys, because that would become ungainly quite quickly.

Basically, the created node would have to look something like this:
```
    key@4 {
        reg = < 4 >;
        linux,code = < 1 >;
        gpios = <&gpio 4 1>;
        label = "KEY_ESC";
    };
```
Modifying the `reg` parameter will also rename the node to match, so we can use the `gpio` parameter to create a unique "address" for each key.

shawaj, 2018-01-02:
@pelwell sorry if any of the following questions seem a bit stupid, but i am a little bit out of my depth at this point!

With the method you have said above - how would that then work to assign multiple keys to a board? is it some parameters after the dtoverlay call in config?

For reference, our board uses GPIO 13, 17, 22, 26, 27 (corresponding to Up SW1, left SW2, right SW4, down SW5, enter SW3 in image below)

[image]

The joystick uses the same pin mapping as the switches (it is just a 5 way switch)
[image]

shawaj, 2018-01-02:
P.S. i realise this is a bit of topic in regards to the rotary encoder, but talking about the same button mapping technique. Would it be better if I moved this to a new issue?

pelwell, 2018-01-02:
To map your switches to the first 5 function keys you would say:
```
dtoverlay=gpio-key,gpio=13,keycode=59,label="KEY_F1"
dtoverlay=gpio-key,gpio=17,keycode=60,label="KEY_F2"
dtoverlay=gpio-key,gpio=22,keycode=61,label="KEY_F3"
dtoverlay=gpio-key,gpio=26,keycode=62,label="KEY_F4"
dtoverlay=gpio-key,gpio=27,keycode=63,label="KEY_F5"
```

pelwell, 2018-01-02:
You can leave the discussion here or create a Pull Request as soon as you have any kind of work-in-progress.

shawaj, 2018-01-03:
ok, made an attempt here #2329

shawaj, 2018-01-03:
Closing this issue as gpio mappable button is now sorted in gpio-key-overlay.dts so can use this to set a button for a rotary encoder

https://github.com/raspberrypi/linux/pull/2329

closed 2018-01-03 by shawaj