raspberrypi/linux #2314: ER: Two issue -- CPU moniter display err and Panel Setting cause the display resolution problem

starsyiyi, 2017-12-23:
hi ,

Two issue:
  1.CPU usage moniter display garbled in chinese localed.
  2.Panel Setting cause the display resolution problem
     the following is the operation steps:
     a.Right click to select "Panel Setting-Panel Applets-Add spacer between CPU Usage Monitor and Digital Clock"
     b.Remove the just added spacer
     c.Also maybe set up the "Geometry-Position",I'm not sure the issue appeared after this step.

     omg,  that resolution has been changed and terminal's area of activity has also been changed.
     How do i change some configurations to modify this problem? thx

     Append info : dibian version is debian-11.27 , HW  pi3
     BTW, I found the 'create_ap' is unstable in Raspberry ,so I successfully created AP with Hostapd, Why not auto assign IP address to STA? thanks a lot .

starsyiyi, 2017-12-23:
[image]

spl237, 2018-04-23:
1 seems to be the wrong font being used when Chinese locale is set; I'll see what can be done to fix it.

I can't repeat 2 - I suspect it is a VNC issue rather than anything in our code, as the screenshot above shows the desktop running under VNC. We've seen a similar problem before. Suggest a bug should be raised with RealVNC rather than us.

JamesH65, 2018-06-27:
@spl237 Inclined to close this, any further reports?

JamesH65, 2018-06-27:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

starsyiyi, 2018-08-20:
close id ,thx

closed 2018-08-20 by starsyiyi