raspberrypi/linux #2315: vc4 gpu in raspberrypi3-b arm64 doesn't show up

yfang1644, 2017-12-24:
I tried several versions(rpi-4.9.y, rpi-4.13.y and rpi-4.14.y) and always get these dmesg:
[    2.669061] vc4-drm soc:gpu: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
[    2.691932] vc4-drm soc:gpu: master bind failed: -517
The related settings in config.txt are:
  gpu_mem=128
  dispmanx_offline=1

  gpu_mem_256=112
  gpu_mem_512=160

  cma_lwm=16device_tree=bcm2837-rpi-3-b.dtb

  cma_hwm=32
  cma_offline_start=16
  dtoverlay=vc4-kms-v3d
  device_tree=bcm2837-rpi-3-b.dtb
Some other options such as disable_overscan, hdmi_force_hotplug, etc. are tested
but have not good results ,I always get black screen.
Using bcm2710-rpi-3-b.dtb will bring up framebuffer, but
that is not I want. I need gpu works in arm64.

JamesH65, 2017-12-24:
Raspberry Pi do not currently support 64 bit operation on the Pi3. From a purely Linux point of view, its mostly works but there are incompatibilities when using the GPU side of things since the GPU is 32bit, and there are some structures that break when the host is 64bit. 

So you should not be expecting general 64 operation using the GPU acceleration to be fully working in the near future, RPF are not working on it, but there are third parties who are doing some work on it.

yfang1644, 2017-12-28:
thanks @JamesH65 .
I don't think 64bit-CPU with 32bit-GPU is a big problem, since my laptop x86_64 cooperates fine with intel or nVidi GPU.

After trying in several days, here is my solution: when i2c-bcm2835 module is loaded (or compiled into zImage), set CONFIG_CMA_SIZE_MBYTES=128 in .config , vc4 gpu works, glxgears runs at 60+/s.
But I still have a problem: there is no sound output. dmesg shows:
[    0.794146] ALSA device list:
[    0.794280]   #0: vc4-hdmi
but amixer controls shows only 1 device:
numid=1,iface=PCM,name='ELD'

mpv plays only a silent movie.

JamesH65, 2017-12-28:
You cannot really compare your laptop and it's GPU with the Raspberry Pi SoC and it's GPU - they are very different things! The VC4 GPU on the Pi is 32bit, and, to put it rather more simply than it actually is,  requires 32bit data structures passed to it. However, when you use a 64 bit kernel, some structures are automatically converted to contain some 64bit values which breaks the interface. Those are the things that need fixing. Not impossible, just a lot of work, although a lot of it has already been done by community efforts.

UMRnInside, 2018-01-14:
Oh!Why are you using **bcm2837** device tree?
You should use **bcm2710-rpi-3-b.dtb** on downstream kernel!

JamesH65, 2018-04-20:
@yfang1644 I inclined to close this issue, unless you have any other comments.  We are not working on this, and are unlikely to do so in the nr future.

closed 2018-06-27 by JamesH65