raspberrypi/linux #2319: Only one PWM channel with snd_bcm2835

fernando-rodriguez, 2017-12-27:
I'm using a custom buildroot image using the 4.9.59 (6820d0cbec6) release. I have snd_bcm2835 compiled builtin and `dtoverlay=pwm-2chan,pin=12,func=2,pin2=13,func2=4` on config.txt (I also tried `dtoverlay=pwm-2chan,pin=18,func=2,pin2=13,func2=4`). There's no pulseaudio, just ALSA. In both cases only BCM13 (PWM1) is driven and ALSA detects a Mono soundcard.

Here the output of `aplay -L` and `amixer`:
```
# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
# amixer
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 0 [96%] [0.00dB] [on]
```

And `dmesg | grep -i 'snd\|sound\|alsa'`:
```
[    0.228068] Advanced Linux Sound Architecture Driver Initialized.
[    1.009135] ALSA device list:
[    1.009145]   #0: bcm2835 ALSA
```

fernando-rodriguez, 2017-12-27:
The board is a Pi Zero W.

fernando-rodriguez, 2017-12-27:
`dmesg | grep -i 'snd\|sound\|alsa\|pwm'` shows more stuff:
```
[    0.228068] Advanced Linux Sound Architecture Driver Initialized.
[    0.321276] clk: couldn't get clock 0 for /soc/pwm@7e20c000
[    1.009135] ALSA device list:
[    1.009145]   #0: bcm2835 ALSA
```

fernando-rodriguez, 2017-12-27:
Copied the contents of /usr/share/alsa from a Raspbian image and the problem has been solved.

fernando-rodriguez, 2017-12-28:
Actually after copying those files the output of `aplay -L` changed but I still have the same problem. Only the left channel is output to BCM13. I tried all the devices listed (except HDMI) and all of them are the same.

```
# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
```

pelwell, 2017-12-28:
The help information for the overlay explains the usage:
```
$ dtoverlay -h pwm-2chan
Name:   pwm-2chan
Info:   Configures both PWM channels
        Legal pin,function combinations for each channel:
          PWM0: 12,4(Alt0) 18,2(Alt5) 40,4(Alt0)            52,5(Alt1)
          PWM1: 13,4(Alt0) 19,2(Alt5) 41,4(Alt0) 45,4(Alt0) 53,5(Alt1)
        N.B.:
          1) Pin 18 is the only one available on all platforms, and
             it is the one used by the I2S audio interface.
             Pins 12 and 13 might be better choices on an A+, B+ or Pi2.
          2) The onboard analogue audio output uses both PWM channels.
          3) So be careful mixing audio and PWM.
          4) Currently the clock must have been enabled and configured
             by other means.
Load:   dtoverlay=pwm-2chan,<param>=<val>
Params: pin                      Output pin (default 18) - see table
        pin2                     Output pin for other channel (default 19)
        func                     Pin function (default 2 = Alt5) - see above
        func2                    Function for pin2 (default 2 = Alt5)
        clock                    PWM clock frequency (informational)
```
This shows that PWM0 on GPIO12 is function 4 (Alt0), as is PWM1 on 13, so try:
```
dtoverlay=pwm-2chan,pin=12,func=4,pin2=13,func2=4
```

fernando-rodriguez, 2017-12-30:
Thanks. That solved the problem for the most part (amixer still shows a Mono soundcard but speaker-test can drive each channel independently). I didn't look at the dtoverlay help info (didn't knew about it) but can you please explain why `pin=18,func=2` doesn't work as it is listed as a legal combo and google returns several guides for the Pi Zero that use that combo. I don't have any I2S stuff compiled in my kernel. Is it something different in the Pi Zero W? Thank you

pelwell, 2018-01-02:
I tried with `dtoverlay=pwm-2chan,pin=18,func=2,pin2=13,func2=4` and I see the PWM1 function being selected on GPIO 18. Do you have raspi-gpio available? If not, it should be easy to build from https://github.com/RPi-Distro/raspi-gpio, then run `raspi-gpio get 13,18` (or `raspi-gpio get 13; raspi-gpio get 18` if you have an older version).

In the event that you don't see PWM0 on 18, please post the full output of `raspi-gpio get`.

fernando-rodriguez, 2018-01-09:
raspi-gpio shows PWM0 on 18 but there's no output coming from the pin.  Thanks.

JamesH65, 2018-04-20:
@fernando-rodriguez Can this issue be closed?

fernando-rodriguez, 2018-04-21:
Yes. I used `pin=12,func=4`. I never did figured out why `pin=18,func=2` doesn't work. Thanks

closed 2018-04-21 by fernando-rodriguez