raspberrypi/linux #2320: No sound output when vc4-kms-v3d overlayed in raspberrypi3 arm64

yfang1644, 2017-12-28:
Graphics and videos are all fine, but without audio.
dmesg lists an ALSA device    #0: vc4-hdmi

Without vc4-kms-v3d, I can switch audio between hdmi and analog using "amixer cset numid=3.."
But how can I output sound when using vc4-gpu?

JamesH65, 2018-04-20:
@yfang1644 Does using the latest Raspbian and kms driver fix this issue?

yfang1644, 2018-04-27:
I'm not using Raspbian. I compiled alsa-lib and alsa-utils(version1.1.4).

JamesH65, 2018-06-27:
@yfang1644 We don't currently support 64bit kernels, so inclined to close this. Have you made any progress?

yfang1644, 2018-06-28:
Thanks @JamesH65. Pls close this topic as 64bit kernel is not mainly supported. I tried arm64 just for fun.

closed 2018-06-28 by pelwell

vanfanel, 2019-03-11:
@JamesH65 : I am in this exact same situation with a project using Pi and HDMI audio with the VC4 graphics stack. 
"aplay -l" lists the vc4hdmi card, and alsamixer says it has no controls.
I am using the latest raspberry pi kernel sources, built for 64 bits. Is this supported in aarch64 already? It's not 2017 anymore... ;)

JamesH65, 2019-03-11:
It may not be 2017, but we still don't support a 64bit build, and with our current workload, there's not a lot of time for 'other' stuff.

vanfanel, 2019-03-11:
@JamesH65 : I can perfectly understand that, James :)
I already made the same question on the forum, but I'll repeat it here:
Is vc4hdmi supposed to work without the bcm2835 audio modules? It's detected as an independent card by ALSA, and in fact, when the bcm2835 audio modules are loaded, there are TWO ALSA cards.

And also, why wouldn't vc4hdmi work on a 64bit system? It uses the vc4 in-kernel stuff, and that DOES work for video (KMS/DRM, OpenGL, etc). What does it have to do with 32/64bit?