raspberrypi/linux #2322: ssh-keygen missing from raspbian stretch lite, 2017-11-29

bschau, 2017-12-30:
I am trying to setup a headless install of a PI 3.   I am using Raspbian Stretch Lite, 2017-11-29 (the official download from https://www.raspberrypi.org/downloads/raspbian/).

The device seems to boot, sets a static IP address and tries to start SSHD.   This step fails as there are no hostkeys.

The "regenerate_ssh_host_keys.service", which runs before starting the SSHD service on initial boots, fails to create the hostkeys as /usr/bin/ssh-keygen is missing.

bschau, 2017-12-30:
Just a follow up - I copied "ssh-keygen" from the full Raspian Stretch 2017-11-29 release and it now works. (Yay!)
Please include ssh-keygen in the next stretch lite release, please :-)

XECDesign, 2017-12-30:
`ssh-keygen` is definitely included in all images. It sounds like something went wrong when the card was imaged. Since this isn't a linux kernel issue, I would recommend posting on the forum.

JamesH65, 2018-04-20:
Closing as this issue is not kernel related.

closed 2018-04-20 by JamesH65