raspberrypi/linux #2328: ER:Tcp high-speed transmission problem

starsyiyi, 2018-01-02:
Test environment:
1.two raspberry pi3,one is AP,the other is STA and successful access.
2.AP mode :g
3.TCP server:AP
4.TCP client:STA
5.The server send data to client and the fixed transfer rate is 11Mb/s
Issue:The data retransmission occurs every 50s, the server can't receive ACK response of a certain frame, the client sends the ACK all the time and sends it more than 70 times. When the ACK is not received, the server is abnormal also continue to send data to the client, and then appeared two seconds of free time, then data transfer is normal.
I did a test at 5Mb/s rate is normal.
I will append wireshark information later.

starsyiyi, 2018-01-02:
This issue caused by signal quality has been ruled out. This is a question of repeated cycles,I guess it may be a bottom-driven issue,the first error is the wrong sequence packet, it is possible that it caused all the following problems,including the ACK sent continuously and two seconds of free time.

starsyiyi, 2018-01-10:
I did some tests, for example, 1.the server is built on Raspberry Pi, the client is running in win7, everything is normal.2.I try to use iperf tools test is normal. So I suspect there are some default parameters (ex.windows size, RCVBUF, etc.) settings that cause this anomaly, the server loses some ACK, I further modify the parameters and then update the problem.

starsyiyi, 2018-01-13:
Thee issue is clear, this is a bottom-drive issue, the pattern is summarized as follows:

1. It is abnormal when the server send data to the client. (both pi-Debian)

2. It is normal when the client send data to the server. (both pi-Debian)

3.  It is normal when the server(pi) send data to the client (WIN7)

The testing base on iperf3 tool. The issue is cycle repeats every 60 seconds.
pls check the following trace.
[F:\RPI3\iperf3_server_send_speed.txt](url)
[F:\RPI3\wk_server_timeout_trace.pcapng](url)

starsyiyi, 2018-01-13:
[iperf3_server_send_speed.txt](https://github.com/raspberrypi/linux/files/1628204/iperf3_server_send_speed.txt)

[wk_server_timeout_trace.zip](https://github.com/raspberrypi/linux/files/1628206/wk_server_timeout_trace.zip)

starsyiyi, 2018-01-19:
Slow processing efficiency，no suppot, the issue has been found the reason , used  system default wifi access iconto to connect  AP may be some problem, use the wicd tool to access AP just normal.But I don't closed the issue myself,waiting for your team to close it. Thx

JamesH65, 2018-01-19:
@spl Any thoughts on the Wifi access icon comment?

JamesH65, 2018-01-19:
Sorry, updated to @spl237

spl237, 2018-01-19:
No idea - this seems to be at a much lower level than anything with which my GUI software interacts.

JamesH65, 2018-04-20:
@starsyiyi  This might be a flow control issue - is it enabled on your network? If you have fixed things, please close the issue.

JamesH65, 2018-06-27:
@starsyiyi Any progress on this?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

starsyiyi, 2018-08-20:
Close it .

closed 2018-08-20 by starsyiyi