raspberrypi/linux #2330: Kernel Panic on Kernel 4.9 while using USB device

florianj1, 2018-01-03:
Hi there,
i want to use my Raspberry as a VDR-Server but at least half an hour after plugging in the DVB-S USB adapter und running vdr the kernel crashes.
My setup looks as follows:
- Raspberry pi 3 running ArchLinux for ARM
- The VDR server is running inside a docker container based on the docker image ‘ arm32v7/debian’
If a client is connected to the VDR-Server the raspberry pi runs for weeks without issues. If no client is connected the kernel crashes.

Kernel-Version: 4.9.72-1

[Wed Jan  3 10:03:11 2018] Unable to handle kernel NULL pointer dereference at virtual address 00000033
[Wed Jan  3 10:03:11 2018] pgd = 80004000
[Wed Jan  3 10:03:11 2018] [00000033] *pgd=00000000
[Wed Jan  3 10:03:11 2018] Internal error: Oops: 805 [#1] SMP ARM
[Wed Jan  3 10:03:11 2018] Modules linked in: lnbp21 tda826x tda10086 dvb_usb_ttusb2 dvb_usb dvb_core rc_core xt_nat xt_tcpudp veth ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter xt_conntrack nf_nat nf_conntrack br_netfilter bridge stp llc brcmfmac brcmutil cfg80211 bcm2835_rng rng_core bcm2835_gpiomem uio_pdrv_genirq fixed uio sch_fq_codel snd_bcm2835 snd_pcm snd_timer snd ip_tables x_tables ipv6
[Wed Jan  3 10:03:12 2018] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.72-1-ARCH #1
[Wed Jan  3 10:03:12 2018] Hardware name: BCM2835
[Wed Jan  3 10:03:12 2018] task: 80f066c0 task.stack: 80f00000
[Wed Jan  3 10:03:12 2018] PC is at usb_api_blocking_completion+0xc/0x14
[Wed Jan  3 10:03:12 2018] LR is at __usb_hcd_giveback_urb+0x98/0x154
[Wed Jan  3 10:03:12 2018] pc : [<807884ec>]    lr : [<80784b28>]    psr: 600e0193
                           sp : 80f01ea0  ip : 00000200  fp : 80f02080
[Wed Jan  3 10:03:12 2018] r10: 80c0ebac  r9 : 00000000  r8 : 80e69340
[Wed Jan  3 10:03:12 2018] r7 : b8b9e400  r6 : 00000000  r5 : 600e0113  r4 : ab9ade00
[Wed Jan  3 10:03:12 2018] r3 : 00000023  r2 : 00000000  r1 : 80f83d88  r0 : 00000023
[Wed Jan  3 10:03:12 2018] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[Wed Jan  3 10:03:12 2018] Control: 10c5383d  Table: 2eee406a  DAC: 00000055
[Wed Jan  3 10:03:12 2018] Process swapper/0 (pid: 0, stack limit = 0x80f00210)
[Wed Jan  3 10:03:12 2018] Stack: (0x80f01ea0 to 0x80f02000)
[Wed Jan  3 10:03:12 2018] 1ea0: a61aed40 ba9ac400 ab9ade00 80f0308c 80e69340 807ad118 200e0113 00040948
[Wed Jan  3 10:03:12 2018] 1ec0: 263810b5 ba991cc0 ba991cc4 00000000 00000000 8012f640 80f02080 00000001
[Wed Jan  3 10:03:12 2018] 1ee0: 00000001 ffffe000 00000000 00000100 80c0ebac 8012ed70 ba839f00 ba839f5c
[Wed Jan  3 10:03:12 2018] 1f00: 00000000 0000000a 0003b06d 00200100 00000000 80e6ced8 00000000 00000000
[Wed Jan  3 10:03:12 2018] 1f20: 00000001 ba805100 80f00000 80c0ebac bb7ffa80 8012f2c4 80e6ced8 8017e260
[Wed Jan  3 10:03:12 2018] 1f40: 801083e8 600e0013 ffffffff 80f01f8c 80f03120 809c0fb8 00ed9b5a 00000000
[Wed Jan  3 10:03:12 2018] 1f60: baf41470 80118460 00000000 ffffe000 80f030b4 00000001 80f03120 80f98a62
[Wed Jan  3 10:03:12 2018] 1f80: 80c0ebac bb7ffa80 00000000 80f01fa8 801083e4 801083e8 600e0013 ffffffff
[Wed Jan  3 10:03:12 2018] 1fa0: 00000001 00000000 00000000 8016eb10 80fa8000 80e00c60 ffffffff ffffffff
[Wed Jan  3 10:03:12 2018] 1fc0: 00000000 80e006a4 80e5be38 00000000 00000000 80fa8294 80f030a0 80e5be34
[Wed Jan  3 10:03:12 2018] 1fe0: 80f08358 0000406a 410fd034 00000000 00000000 0000807c 00000000 00000000
[Wed Jan  3 10:03:12 2018] [<807884ec>] (usb_api_blocking_completion) from [<80784b28>] (__usb_hcd_giveback_urb+0x98/0x154)
[Wed Jan  3 10:03:12 2018] [<80784b28>] (__usb_hcd_giveback_urb) from [<807ad118>] (completion_tasklet_func+0x78/0xbc)
[Wed Jan  3 10:03:12 2018] [<807ad118>] (completion_tasklet_func) from [<8012f640>] (tasklet_hi_action+0x64/0xf8)
[Wed Jan  3 10:03:13 2018] [<8012f640>] (tasklet_hi_action) from [<8012ed70>] (__do_softirq+0xc4/0x320)
[Wed Jan  3 10:03:13 2018] [<8012ed70>] (__do_softirq) from [<8012f2c4>] (irq_exit+0xc4/0x140)
[Wed Jan  3 10:03:13 2018] [<8012f2c4>] (irq_exit) from [<8017e260>] (__handle_domain_irq+0x58/0xb8)
[Wed Jan  3 10:03:13 2018] [<8017e260>] (__handle_domain_irq) from [<809c0fb8>] (__irq_svc+0x58/0x74)
[Wed Jan  3 10:03:13 2018] [<809c0fb8>] (__irq_svc) from [<801083e8>] (arch_cpu_idle+0x30/0x3c)
[Wed Jan  3 10:03:13 2018] [<801083e8>] (arch_cpu_idle) from [<8016eb10>] (cpu_startup_entry+0xe0/0x15c)
[Wed Jan  3 10:03:13 2018] [<8016eb10>] (cpu_startup_entry) from [<80e00c60>] (start_kernel+0x370/0x404)
[Wed Jan  3 10:03:13 2018] Code: ea08e119 e5903074 e5902038 e1a00003 (e5832010)
[Wed Jan  3 10:07:01 2018] INFO: task kworker/2:0:1930 blocked for more than 120 seconds.
[Wed Jan  3 10:07:01 2018]       Not tainted 4.9.72-1-ARCH #1
[Wed Jan  3 10:07:01 2018] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[Wed Jan  3 10:07:01 2018] kworker/2:0     D    0  1930      2 0x00000000
[Wed Jan  3 10:07:01 2018] Workqueue: events check_carrier
[Wed Jan  3 10:07:01 2018] [<809bc0bc>] (__schedule) from [<809bc718>] (schedule+0x3c/0xa0)
[Wed Jan  3 10:07:01 2018] [<809bc718>] (schedule) from [<807880e4>] (usb_kill_urb+0xbc/0xf4)
[Wed Jan  3 10:07:01 2018] [<807880e4>] (usb_kill_urb) from [<80788614>] (usb_start_wait_urb+0xc0/0xe0)
[Wed Jan  3 10:07:01 2018] [<80788614>] (usb_start_wait_urb) from [<807886e8>] (usb_control_msg+0xb4/0x100)
[Wed Jan  3 10:07:01 2018] [<807886e8>] (usb_control_msg) from [<80777bcc>] (__usbnet_read_cmd+0x74/0xc8)
[Wed Jan  3 10:07:01 2018] [<80777bcc>] (__usbnet_read_cmd) from [<80777c74>] (usbnet_read_cmd+0x54/0x74)
[Wed Jan  3 10:07:01 2018] [<80777c74>] (usbnet_read_cmd) from [<80774d80>] (__smsc95xx_read_reg+0x60/0xbc)
[Wed Jan  3 10:07:01 2018] [<80774d80>] (__smsc95xx_read_reg) from [<80774f64>] (__smsc95xx_phy_wait_not_busy+0x5c/0xac)
[Wed Jan  3 10:07:01 2018] [<80774f64>] (__smsc95xx_phy_wait_not_busy) from [<807751b0>] (__smsc95xx_mdio_read+0x40/0x130)
[Wed Jan  3 10:07:01 2018] [<807751b0>] (__smsc95xx_mdio_read) from [<80777034>] (check_carrier+0x28/0xb0)
[Wed Jan  3 10:07:01 2018] [<80777034>] (check_carrier) from [<80142a08>] (process_one_work+0x210/0x500)
[Wed Jan  3 10:07:01 2018] [<80142a08>] (process_one_work) from [<80143aa4>] (worker_thread+0x34/0x5d0)
[Wed Jan  3 10:07:01 2018] [<80143aa4>] (worker_thread) from [<80148920>] (kthread+0x104/0x118)
[Wed Jan  3 10:07:01 2018] [<80148920>] (kthread) from [<80107b40>] (ret_from_fork+0x14/0x34)
[Wed Jan  3 10:07:01 2018] INFO: task frontend 0/0 tu:2124 blocked for more than 120 seconds.
[... 150 lines cut ...]
[Wed Jan  3 10:15:12 2018] [<807751b0>] (__smsc95xx_mdio_read) from [<80777034>] (check_carrier+0x28/0xb0)
[Wed Jan  3 10:15:12 2018] [<80777034>] (check_carrier) from [<80142a08>] (process_one_work+0x210/0x500)
[Wed Jan  3 10:15:12 2018] [<80142a08>] (process_one_work) from [<80143aa4>] (worker_thread+0x34/0x5d0)
[Wed Jan  3 10:15:12 2018] [<80143aa4>] (worker_thread) from [<80148920>] (kthread+0x104/0x118)
[Wed Jan  3 10:15:12 2018] [<80148920>] (kthread) from [<80107b40>] (ret_from_fork+0x14/0x34)
[Wed Jan  3 10:15:12 2018] INFO: task frontend 0/0 tu:2124 blocked for more than 120 seconds.
[Wed Jan  3 10:15:12 2018]       Not tainted 4.9.72-1-ARCH #1
[Wed Jan  3 10:15:12 2018] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[Wed Jan  3 10:15:12 2018] frontend 0/0 tu D    0  2124   2112 0x00000080
[Wed Jan  3 10:15:12 2018] [<809bc0bc>] (__schedule) from [<809bc718>] (schedule+0x3c/0xa0)
[Wed Jan  3 10:15:12 2018] [<809bc718>] (schedule) from [<807880e4>] (usb_kill_urb+0xbc/0xf4)
[Wed Jan  3 10:15:12 2018] [<807880e4>] (usb_kill_urb) from [<80788614>] (usb_start_wait_urb+0xc0/0xe0)
[Wed Jan  3 10:15:12 2018] [<80788614>] (usb_start_wait_urb) from [<7f3d7bec>] (dvb_usb_generic_rw+0x130/0x19c [dvb_usb])
[Wed Jan  3 10:15:12 2018] [<7f3d7bec>] (dvb_usb_generic_rw [dvb_usb]) from [<7f3e0178>] (ttusb2_msg+0xb8/0x17c [dvb_usb_ttusb2])
[Wed Jan  3 10:15:12 2018] [<7f3e0178>] (ttusb2_msg [dvb_usb_ttusb2]) from [<7f3e033c>] (ttusb2_i2c_xfer+0x100/0x27c [dvb_usb_ttusb2])
[Wed Jan  3 10:15:12 2018] [<7f3e033c>] (ttusb2_i2c_xfer [dvb_usb_ttusb2]) from [<807e159c>] (__i2c_transfer+0x124/0x4d4)
[Wed Jan  3 10:15:12 2018] [<807e159c>] (__i2c_transfer) from [<807e19b0>] (i2c_transfer+0x64/0xb4)
[Wed Jan  3 10:15:12 2018] [<807e19b0>] (i2c_transfer) from [<7f3e91c4>] (tda10086_read_byte+0x78/0xe4 [tda10086])
[... cut ...]
[Wed Jan  3 10:15:12 2018] [<7f3c10cc>] (dvb_frontend_ioctl_legacy [dvb_core]) from [<7f3c2a84>] (dvb_frontend_ioctl+0x98/0x104 [dvb_core])
[Wed Jan  3 10:15:12 2018] [<7f3c2a84>] (dvb_frontend_ioctl [dvb_core]) from [<7f3b87dc>] (dvb_usercopy+0xa8/0x19c [dvb_core])
[Wed Jan  3 10:15:12 2018] [<7f3b87dc>] (dvb_usercopy [dvb_core]) from [<8028da4c>] (do_vfs_ioctl+0x98/0x828)
[Wed Jan  3 10:15:12 2018] [<8028da4c>] (do_vfs_ioctl) from [<8028e210>] (SyS_ioctl+0x34/0x58)
[Wed Jan  3 10:15:12 2018] [<8028e210>] (SyS_ioctl) from [<80107c10>] (__sys_trace_return+0x0/0x10)

spiderkeys, 2018-01-23:
We are seeing a similar issue on the RPI3 using Resin.io. The Host OS is running kernel 4.9.50. Not sure if this is tied to the kernel or the RPI firmware, as suggested in #2113.

The issue happens when we run any program or command that invokes the libudev udev_enumerate_scan_devices method after connecting a USB camera to our system. Both our custom application and "udevadm info -e" cause a segfault upon invoking this method when the camera is attached. 

Below is the dump at the point in time where our application uses the enumerate method to print a list of devices connected via USB:
```
Jan 22 22:14:48 28dc8be-28dc8be geoserve[587]: 2018-01-22 22:14:48,413 INFO | Paths associated with this device:
Jan 22 22:14:48 28dc8be-28dc8be kernel: Unable to handle kernel NULL pointer dereference at virtual address 00000008
Jan 22 22:14:48 28dc8be-28dc8be kernel: pgd = b68b0000
Jan 22 22:14:48 28dc8be-28dc8be kernel: [00000008] *pgd=3b37b835, *pte=00000000, *ppte=00000000
Jan 22 22:14:48 28dc8be-28dc8be kernel: Internal error: Oops: 17 [#1] SMP ARM
Jan 22 22:14:48 28dc8be-28dc8be kernel: Modules linked in: snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi evdev snd_seq_device snd_pcm snd_timer snd ipt_REJECT n
f_reject_ipv4 uvcvideo(O) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media at24 nvmem_core fuse ipt_MASQUERADE nf_nat_masquerade_ipv4 b
r_netfilter nf_conntrack_netlink nfnetlink i2c_dev spidev brcmfmac brcmutil cfg80211 rfkill i2c_bcm2835 bcm2835_gpiomem uio_pdrv_genirq uio fixed sch_fq_codel [last u
nloaded: spi_bcm2835]
Jan 22 22:14:48 28dc8be-28dc8be kernel: CPU: 3 PID: 1892 Comm: geoserve Tainted: G           O    4.9.50 #2
Jan 22 22:14:48 28dc8be-28dc8be kernel: Hardware name: BCM2835
Jan 22 22:14:48 28dc8be-28dc8be kernel: task: b6b13b00 task.stack: bb87a000
Jan 22 22:14:48 28dc8be-28dc8be kernel: PC is at kernfs_iop_get_link+0xa4/0x1d4
Jan 22 22:14:48 28dc8be-28dc8be kernel: LR is at mutex_lock+0x1c/0x64
Jan 22 22:14:48 28dc8be-28dc8be kernel: pc : [<80300014>]    lr : [<807ed7e4>]    psr: a00e0013
                                        sp : bb87bee0  ip : bb87bec8  fp : bb87bf14
Jan 22 22:14:48 28dc8be-28dc8be kernel: r10: 00000000  r9 : bce53e40  r8 : 7eebf118
Jan 22 22:14:48 28dc8be-28dc8be kernel: r7 : b6828000  r6 : b6828000  r5 : bcdb2820  r4 : 00000000
Jan 22 22:14:48 28dc8be-28dc8be kernel: r3 : bcc8a410  r2 : 8098fda0  r1 : 00000000  r0 : 00000000
Jan 22 22:14:48 28dc8be-28dc8be kernel: Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Jan 22 22:14:48 28dc8be-28dc8be kernel: Control: 10c5383d  Table: 368b006a  DAC: 00000055
Jan 22 22:14:48 28dc8be-28dc8be kernel: Process geoserve (pid: 1892, stack limit = 0xbb87a210)
Jan 22 22:14:48 28dc8be-28dc8be kernel: Stack: (0xbb87bee0 to 0xbb87c000)
Jan 22 22:14:48 28dc8be-28dc8be kernel: bee0: 18fecbcd bb87bf1c ffffff9c 7eebf118 80f0414c 00000400 bb87bf58 7eebf118
Jan 22 22:14:48 28dc8be-28dc8be kernel: bf00: 00000400 bcb6e5a0 bb87bf44 bb87bf18 802921f8 802fff7c 8029076c 00000000
Jan 22 22:14:48 28dc8be-28dc8be kernel: bf20: 00000000 00040932 00004000 00004000 ffffff9c 7eebed18 bb87bf94 bb87bf48
Jan 22 22:14:48 28dc8be-28dc8be kernel: bf40: 80285be8 80292164 bb87bf58 00001000 00020000 80f0414c 00000000 b59b3d10
Jan 22 22:14:48 28dc8be-28dc8be kernel: bf60: bcaec440 00040932 00001000 7eebf950 01b4a398 01b48cf8 00000055 801085a8
Jan 22 22:14:48 28dc8be-28dc8be kernel: bf80: bb87a000 00000000 bb87bfa4 bb87bf98 80285c98 80285b40 00000000 bb87bfa8
Jan 22 22:14:48 28dc8be-28dc8be kernel: bfa0: 801083e0 80285c7c 7eebf950 01b4a398 7eebed18 7eebf118 00000400 00000000
Jan 22 22:14:48 28dc8be-28dc8be kernel: bfc0: 7eebf950 01b4a398 01b48cf8 00000055 7eebf950 01b48b40 7eebf118 01b39fc8
Jan 22 22:14:48 28dc8be-28dc8be kernel: bfe0: 76fa0ec4 7eebec9c 76f8c80f 76d47186 400e0030 7eebed18 00000000 00000000
Jan 22 22:14:48 28dc8be-28dc8be kernel: [<80300014>] (kernfs_iop_get_link) from [<802921f8>] (generic_readlink+0xa0/0xb8)
Jan 22 22:14:48 28dc8be-28dc8be kernel: [<802921f8>] (generic_readlink) from [<80285be8>] (SyS_readlinkat+0xb4/0x13c)
Jan 22 22:14:48 28dc8be-28dc8be kernel: [<80285be8>] (SyS_readlinkat) from [<80285c98>] (SyS_readlink+0x28/0x2c)
Jan 22 22:14:48 28dc8be-28dc8be kernel: [<80285c98>] (SyS_readlink) from [<801083e0>] (ret_fast_syscall+0x0/0x1c)
Jan 22 22:14:48 28dc8be-28dc8be kernel: Code: ea000002 e1540005 0a00000d e1a04003 (e5943008) 
Jan 22 22:14:48 28dc8be-28dc8be kernel: ---[ end trace 2d2020be7e234f19 ]---
```

Working on solving this on the resin side as well, but would like to get to the bottom of whether or not this is a firmware issue or a kernel issue, and if it is the kernel, figure out where the breaking change occurred.

P33M, 2018-02-22:
Can either of you reproduce the bug using a rpi-update provided kernel?

We can't really support kernels built from other trees.

JamesH65, 2018-04-20:
@florianj1 @spiderkeys  Did either poster have the chance to test this on an official kernel? Get the latest one using rpi-update and test. Or, please close issue if no longer relevant.

P33M, 2018-05-15:
Closing as no response from OP in >2 weeks. If issue is reproducible on a recent rpi-update kernel, will reopen.

closed 2018-05-15 by P33M