raspberrypi/linux #2331: rotary encoder - additional overrides?

shawaj, 2018-01-03:
According to this doc - https://elixir.free-electrons.com/linux/latest/source/Documentation/devicetree/bindings/input/rotary-encoder.txt

There are more configurable parameters for the rotary-encoder driver.

Would it be acceptable to add these all in as additional overrides to the driver?

Maybe something like this:
```
        relative_axis =  <&knob>,"rotary-encoder,relative-axis";
        linux_axis =  <&knob>,"linux,axis";
        rollover =  <&knob>,"rotary-encoder,rollover";
        steps-per-period =  <&knob>,"rotary-encoder,steps-per-period";
        steps =  <&knob>,"rotary-encoder,steps";
        wakeup = <&knob>,"wakeup-source";
        encoding =  <&knob>,"rotary-encoder,encoding";
    };
```

pelwell, 2018-01-04:
Yes, of course - go ahead.

As a general point, each PR is a both a request and a topic for discussion, so I don't consider it necessary to ask in this way, polite as it is. Going straight to a PR helps to reduce the clutter and keep all relevant discussion in the same place.

shawaj, 2018-01-04:
Understood! I'm new to these parts so don't know the etiquette yet and didn't want to step on any toes :-P

See your point though... Better if all in one place

shawaj, 2018-01-05:
Closing this issue now and moving over to #2334

closed 2018-01-05 by shawaj