raspberrypi/linux #2335: dwc_otg_hcd_handle_hc_fsm:2459:Unexpected IRQ state on FSM transaction:dev_addr=4 ep=1 fsm=6, hcint=0x00000002

R4C3R, 2018-01-05:
I get this error when loading the bcm2835-v4l2 in combination with NFS read-only boot.

/etc/modules-load.d/bcm2835-v4l2.conf
`bcm2835-v4l2`

/etc/modprobe.d/bcm2835-v4l2.conf
`options bcm2835-v4l2 max_video_width=3280 max_video_height=2464`

Without the bcm2835-v4l2 module loaded at boot, the read-only NFS boot works fine. However, starting the bcm2835-v4l2 module manually results in a system hang.

`sudo modprobe bcm2835-v4l2 max_video_width=3280 max_video_height=2464`

tested on RPi 2 B with Raspbian Jessie lite 
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

also tested tested on RPi 2 B with Raspbian Stretch lite 2017-11-29 with the latest kernel / firmware 4.9.73 with no difference. 

there is actually an error before this error during boot:

smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -110

[image]

P33M, 2018-01-08:
Can you capture the kernel crash messages via the GPIO UART? Without a full stacktrace it's going to be hard to debug.

6by9, 2018-01-08:
My instinctive answer is the PSU and low voltage.

Loading ```bcm2835-v4l2``` will power up the camera module, drawing (IIRC) an extra 150mA off the power supply. The SMSC9514 is more susceptible to brown-out that the BCM283x, and brown-out puts it in a state that Linux doesn't recover from. No SMSC9514 = no ethernet = no NFS mount.
Seeing as you are using a Pi2, is the power LED on and stable? ```vcgencmd get_throttled``` should report 0 if there have been no low voltage issues.

Alternatively, check whether running raspivid causes the same problems, as that will also power up the camera.

R4C3R, 2018-01-09:
I’m travelling now. I can check this Thursday when I am back. Thank you for
looking into it!

-- 
Rolf Bakker
Lerchenweg 14
56075 Koblenz
Deutschland

Tel: +49 261 914 8934
Fax: +49 261 914 8935
D: +4915755211456
NL: +31642777899

Email: rolf.bakker.70@gmail.com

R4C3R, 2018-01-11:
Power supply:
The power supply provides 25W @ 5v, more than plenty I assume, also there is no "lighting bold" visible in the upper right corner during startup. I double checked by replacing it with a 2.5A PSU, which also worked.

Serial Debug:
I've captured the serial console output and saved it to the attached file.

[serial-debug.txt](https://github.com/raspberrypi/linux/files/1622072/serial-debug.txt)

pelwell, 2018-01-11:
Which contains (after the troublesome leading NUL):
```
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 SMP Sun Oct 29 12:19:23 GMT 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x2f800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 14 pages/cpu @af0f1000 s25600 r8192 d23552 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=800 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 ip=dhcp root=/dev/nfs rootfstype=nfs nfs-options=hard,intr,ro  elevator=deadline fsck.repair=skip rootwait noswap ro
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 757668K/786432K available (7168K kernel code, 486K rwdata, 2012K rodata, 1024K init, 770K bss, 20572K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xb0800000 - 0xff800000   (1264 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xb0000000   ( 768 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c79834   ( 487 kB)
[    0.000000]        .bss : 0x80c7b000 - 0x80d3b964   ( 771 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000250] Console: colour dummy device 80x30
[    0.000912] console [tty1] enabled
[    0.000949] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001002] pid_max: default: 32768 minimum: 301
[    0.001288] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001324] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002318] Disabling memory control group subsystem
[    0.002429] CPU: Testing write buffer coherency: ok
[    0.002493] ftrace: allocating 22392 entries in 66 pages
[    0.037891] CPU0: update cpu_capacity 1024
[    0.037944] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.038004] Setting up static identity map for 0x100000 - 0x100034
[    0.040110] CPU1: update cpu_capacity 1024
[    0.040118] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.040899] CPU2: update cpu_capacity 1024
[    0.040905] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.041640] CPU3: update cpu_capacity 1024
[    0.041646] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.041731] Brought up 4 CPUs
[    0.041868] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.041891] CPU: All CPU(s) started in HYP mode.
[    0.041911] CPU: Virtualization extensions available.
[    0.042761] devtmpfs: initialized
[    0.052150] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.052483] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.052540] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.053139] pinctrl core: initialized pinctrl subsystem
[    0.054110] NET: Registered protocol family 16
[    0.056112] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.063277] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.063321] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.063493] Serial: AMBA PL011 UART driver
[    0.065174] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.128864] bcm2835-dma 3f007000.dma: DMA legacy API manager at b080f000, dmachans=0x1
[    0.130562] SCSI subsystem initialized
[... 131 lines cut ...]
[    8.045792] Freeing unused kernel memory: 1024K
[    8.617151] systemd[1]: System time before build time, advancing clock.
[    8.767019] NET: Registered protocol family 10
[    8.790212] random: crng init done
[    8.801329] ip_tables: (C) 2000-2006 Netfilter Core Team
[    8.843989] systemd[1]: systemd 232 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN)
[... cut ...]
[   10.009877] Linux video capture interface: v2.00
[   12.133402] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 3280x2464
[   12.150481] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.
[   16.319081] smsc95xx 1-1.1:1.0 eth0: Failed to read reg index 0x00000114: -110
[   16.330759] smsc95xx 1-1.1:1.0 eth0: Error reading MII_ACCESS
[   16.340894] smsc95xx 1-1.1:1.0 eth0: MII is busy in smsc95xx_mdio_read
[   16.351799] smsc95xx 1-1.1:1.0 eth0: Failed to read MII_BMSR
```

R4C3R, 2018-01-11:
I was already wondering about all the chinese signs that suddenly showed up after I downloaded the file. Thank you for fixing it, how did you do it? (before uploading the file, it looked fine)

pelwell, 2018-01-11:
I just opened it in a decent editor (Notepad++ in this case, but emacs, vi etc. would have worked) and deleted the NUL, tidying it up a bit further by removing the duplicate newlines.

R4C3R, 2018-01-11:
ok thx letting me know.

6by9, 2018-01-11:
That leaves the question of running ```raspivid``` in place of loading ```bcm2835-v4l2```. If that also takes down the SMSC9514 then it sounds like a hardware issue, otherwise it has to be something in the kernel.

R4C3R, 2018-01-11:
I can remove the module from loading at startup and then start raspivid, is that what I should try? Any specific raspivid command that you would like to see?

6by9, 2018-01-11:
> I can remove the module from loading at startup and then start raspivid, is that what I should try?

Yes. Is it the act of enabling the camera that is causing your issue, or something else?

> Any specific raspivid command that you would like to see?

Anything that actually starts up the camera. I don't care if it records anything or not. ```raspivid -t 2000``` is as good as anything to just run preview for 2 seconds.

R4C3R, 2018-01-11:
Yes, loading the module causes the issue. Either loading during boot or loading manually (if not loaded at boot) hangs the system.

just tested with raspivid, it worked.

below you will find the serial debug, the only difference from the version above beeing that the bcm2835-v4l2 module is not loaded at startup. The pi starts up fine and raspivid works.

`[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 SMP Sun Oct 29 12:19:23 GMT 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x2f800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 14 pages/cpu @af0f1000 s25600 r8192 d23552 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=800 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 ip=dhcp root=/dev/nfs rootfstype=nfs nfs-options=hard,intr,ro  elevator=deadline fsck.repair=skip rootwait noswap ro
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 757668K/786432K available (7168K kernel code, 486K rwdata, 2012K rodata, 1024K init, 770K bss, 20572K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xb0800000 - 0xff800000   (1264 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xb0000000   ( 768 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c79834   ( 487 kB)
[    0.000000]        .bss : 0x80c7b000 - 0x80d3b964   ( 771 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000021] Switching to timer-based delay loop, resolution 52ns
[    0.000250] Console: colour dummy device 80x30
[    0.000911] console [tty1] enabled
[    0.000948] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001001] pid_max: default: 32768 minimum: 301
[    0.001285] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001320] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002309] Disabling memory control group subsystem
[    0.002428] CPU: Testing write buffer coherency: ok
[    0.002493] ftrace: allocating 22392 entries in 66 pages
[    0.037877] CPU0: update cpu_capacity 1024
[    0.037929] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.037985] Setting up static identity map for 0x100000 - 0x100034
[    0.040088] CPU1: update cpu_capacity 1024
[    0.040096] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.040878] CPU2: update cpu_capacity 1024
[    0.040885] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.041620] CPU3: update cpu_capacity 1024
[    0.041626] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.041712] Brought up 4 CPUs
[    0.041848] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.041871] CPU: All CPU(s) started in HYP mode.
[    0.041891] CPU: Virtualization extensions available.
[    0.042749] devtmpfs: initialized
[    0.052144] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.052472] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.052528] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.053125] pinctrl core: initialized pinctrl subsystem
[    0.054092] NET: Registered protocol family 16
[    0.056099] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.063273] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.063317] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 199 lines cut ...]
         Starting LSB: Starts syslogd...
[[0;32m  OK  [0m] Started Save/Restore Sound Card State.
[[0;32m  OK  [0m] Started Login Service.
[[0;32m  OK  [0m] Started LSB: Switch to ondemand cpu governor (unless shift key is pressed).
[[0;32m  OK  [0m] Started Raise network interfaces.
[[0;32m  OK  [0m] Reached target Network.
[[0;32m  OK  [0m] Reached target Network is Online.
[... cut ...]
         Starting Permit User Sessions...
         Mounting Mount srvr directory...
[[0;32m  OK  [0m] Started LSB: Starts syslogd.
         Starting LSB: Starts klogd...
My IP address is 192.168.178.86 
[[0;32m  OK  [0m] Started /etc/rc.local Compatibility.
[[0;32m  OK  [0m] Started LSB: Starts klogd.
[[0;32m  OK  [0m] Started Permit User Sessions.
         Starting Terminate Plymouth Boot Screen...
         Starting Hold until boot process finishes up...

Raspbian GNU/Linux 9 raspberrypi ttyAMA0
[9;0]raspberrypi login: 
`

6by9, 2018-01-11:
Throw in the module parameter ```debug=4``` (ie ```modprobe bcm2835-v4l2 debug=4````) when loading to get extra debug.

You say this is a 2B. Can you provide the Revision code from /proc/cpuinfo please?

R4C3R, 2018-01-11:
If I enter `sudo modprobe bcm2835-v4l2 debug=4` it just immediately hangs, no output whatsoever

Below is the info from /proc/cpuinfo, I can also test with a RPi 3 if required.

```
sudo cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a01041
Serial		: 0000000000729a6b

```

R4C3R, 2018-01-11:
Just made a discovery:

if I run 

`sudo modprobe bcm2835-v4l2 debug=4`

the system immediately hangs

BUT

if I run 

`raspivid -t 2000` 

and than run 

`sudo modprobe bcm2835-v4l2 debug=4`

than the system works perfectly fine.

6by9, 2018-01-11:
Most puzzling, and I'm running out of ideas.
I have a Pi2 here with the same kernel but booting off SD card, and see no problems. If I can get an NFS boot sorted quickly then I'll give it a go off that. Possibly it is being a ro mount that is causing issues? I don't see why though.
A test on a Pi 3 wouldn't go amiss.

R4C3R, 2018-01-11:
I've tested with an RPi 3 now. Behaviour seems the same. 

the output from the serial console differs slightly from the HDMI screen, on the HDMI screen I see the following additional messages (don't think I've seen them on the RPi 2)

[       11.996793] bcm2835_v4l2: error 0 waiting for sync completion
[       12.005343] bcm2835-v4l2: error -62 while loading driver

(kindly note that first line shows an underscore and the second line a hyphen)

Below is the serial debug output from the console when the RPi3 starts up and the bcm2835-v4l2 module is loaded during boot

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 SMP Sun Oct 29 12:19:23 GMT 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x2f800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 14 pages/cpu @af0f0000 s25600 r8192 d23552 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=800 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 ip=dhcp root=/dev/nfs rootfstype=nfs nfs-options=hard,intr,ro  elevator=deadline fsck.repair=skip rootwait noswap ro
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 757664K/786432K available (7168K kernel code, 486K rwdata, 2012K rodata, 1024K init, 770K bss, 20576K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xb0800000 - 0xff800000   (1264 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xb0000000   ( 768 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
[    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c79834   ( 487 kB)
[    0.000000]        .bss : 0x80c7b000 - 0x80d3b964   ( 771 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000004] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000012] Switching to timer-based delay loop, resolution 52ns
[    0.000166] Console: colour dummy device 80x30
[    0.000625] console [tty1] enabled
[    0.000651] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000686] pid_max: default: 32768 minimum: 301
[    0.000862] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000883] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001474] Disabling memory control group subsystem
[    0.001531] CPU: Testing write buffer coherency: ok
[    0.001565] ftrace: allocating 22392 entries in 66 pages
[    0.026180] CPU0: update cpu_capacity 1024
[    0.026213] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.026251] Setting up static identity map for 0x100000 - 0x100034
[    0.027337] CPU1: update cpu_capacity 1024
[    0.027341] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.027738] CPU2: update cpu_capacity 1024
[    0.027741] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.028121] CPU3: update cpu_capacity 1024
[    0.028124] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.028169] Brought up 4 CPUs
[    0.028258] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.028273] CPU: All CPU(s) started in HYP mode.
[    0.028286] CPU: Virtualization extensions available.
[    0.028746] devtmpfs: initialized
[    0.034580] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.034749] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 215 lines cut ...]
[  192.476935] nfs: server 192.168.178.2 not responding, still trying
[  192.476939] nfs: server 192.168.178.2 not responding, still trying
[  192.477087] nfs: server 192.168.178.2 not responding, still trying
[  217.436813] nfs: server 192.168.178.2 not responding, still trying
[  505.036803] ------------[ cut here ]------------
[  505.043847] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:316 dev_watchdog+0x294/0x298
[  505.054488] NETDEV WATCHDOG: eth0 (smsc95xx): transmit queue 0 timed out
[... cut ...]
[  505.167238] [<8018565c>] (expire_timers) from [<80185788>] (run_timer_softirq+0xa4/0x1c4)
[  505.177581] [<80185788>] (run_timer_softirq) from [<8010169c>] (__do_softirq+0x18c/0x3cc)
[  505.187940] [<8010169c>] (__do_softirq) from [<80122cf8>] (irq_exit+0x10c/0x168)
[  505.197553] [<80122cf8>] (irq_exit) from [<8017371c>] (__handle_domain_irq+0x70/0xc4)
[  505.207522] [<8017371c>] (__handle_domain_irq) from [<8010150c>] (bcm2836_arm_irqchip_handle_irq+0xa8/0xac)
[  505.219364] [<8010150c>] (bcm2836_arm_irqchip_handle_irq) from [<8071a6bc>] (__irq_svc+0x5c/0x7c)

```

6by9, 2018-01-11:
Failing again to reproduce. Pi2 booting with an NFS mount (server set to read only), and pretty much your command line. Comes up fine, and I can load ```bcm2835-v4l2``` without issue.

Can you try with a vanilla SD card image? It just removes one factor from the equation.

R4C3R, 2018-01-11:
ok, did the test with a plain vanilla image but the process hangs during boot at the moment that the bcm2835-v4l2 module is loaded

I used RASPBIAN STRETCH LITE 2017-11-29

First bootet nromally to configure the image. I ran raspi-config on the pi3 and enabled ssh, serial console and camera module, resized the image and allocated 256mb ram to the gpu.

After that I rebooted

then I altered fstab and commented out the lines referring to the SD card and I changed cmdline.txt to this

`dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 ip=dhcp root=/dev/nfs rootfstype=nfs nfs-options=hard,intr,ro  elevator=deadline fsck.repair=skip rootwait noswap ro`

then I enabled the bcm2835-v4l2 driver to be loaded during boot like this:

```
echo "bcm2835-v4l2" | sudo tee /etc/modules-load.d/bcm2835-v4l2.conf > /dev/null
echo "options bcm2835-v4l2 max_video_width=3280 max_video_height=2464" | sudo tee /etc/modprobe.d/bcm2835-v4l2.conf > /dev/null
```
then `sudo poweroff`

Then I copied the /boot folder to my tftp folder on the NFS server (I use isc-dhcp-server in combination with nfs-kernel-server) and I copied the root folder to the NFS server (folder is a read only nfs-share)

Then I booted the RPi3 with an SD card with only boodcode.bin

The RPi3 boots, but an error occurs when loading the bcm2835-v4l2 module, the RPi3 hangs after that. I cannot start the bcm2835-v4l2 module manually.

`sudo modprobe bcm2835-v4l2` leads to a operation not permitted error (probably due to the read-only mode)

Is this what you meant by plain vanilla? If not, please let me have a description (like I did above) on how to configure the plain vanilla read only image.

6by9, 2018-01-12:
Plain vanilla as in from the SD card before you edited cmdline.txt and fstab to NFS boot.

R4C3R, 2018-01-12:
plain vanilla works fine

I used RASPBIAN STRETCH LITE 2017-11-29

First booted normally to configure the image. I ran raspi-config on the pi3 and enabled ssh, serial console and camera module, resized the image and allocated 256mb ram to the gpu.

After that I rebooted

then I enabled the bcm2835-v4l2 driver to be loaded during boot like this:

echo "bcm2835-v4l2" | sudo tee /etc/modules-load.d/bcm2835-v4l2.conf > /dev/null
echo "options bcm2835-v4l2 max_video_width=3280 max_video_height=2464" | sudo tee /etc/modprobe.d/bcm2835-v4l2.conf > /dev/null

to test, I used the following commands:

```
v4l2-ctl -p 15
v4l2-ctl -v width=3280,height=2464,pixelformat=I420
v4l2-ctl --stream-mmap=3 --stream-to=/dev/null --stream-count=50
```

R4C3R, 2018-01-12:
Just for comparison I setup a piserver and used the preinstalled Raspbian Lite OS.

I enabled the camera in config.txt and loaded the bcm2835-v4l2 module during boot.

Also in this case the boot procedure hangs. 

I hope this helps to reproduce

6by9, 2018-01-15:
OK thanks. I'll try PiServer when I get a chance as that should be relatively easy to get set up (Gordon probably has a setup on his desk)

R4C3R, 2018-01-22:
@6by9 just wondering whether you were able to reproduce the "hang on boot" while loading the bcm2835-v4l2 module with PiServer and the Raspbian Lite image

6by9, 2018-01-23:
Sorry, not had a chance to look as yet.
Gordon's PiServer rig is on his desk with his own strange config, so not the most convenient. I'll see if I can get it running in a VM easily.

R4C3R, 2018-01-23:
OK, thx for letting me know. The PiServer software works well, I think (if I remember correctly) when it starts, it offers to use a live cd, that way you don't have to install PiServer if you want to do a quick test.

6by9, 2018-01-24:
I've finally got a PiServer system booting a Pi3 and have a strong suspicion as to what is going on, but need to get back on to other things.

I suspect that the relevant service hasn't been started on the VPU. raspivid etc call bcm_host_init as they initialise, which starts the services on the VPU. Once started, they keep running even when the clients close.

My quick test using:
```
#include "bcm_host.h"

int main(int argc, char* argv)
{
  bcm_host_init();
  return 0;
}
```
built with ```gcc test.c -I/opt/vc/include -lbcm_host -L/opt/vc/lib```. Running the resultant binary before loading bcm2835-v4l2 seems to confirm that, however I haven't got time at the moment to fully understand and address the underlying problem.

R4C3R, 2018-01-29:
I am glad you were able to reproduce & confirm

njwilliams, 2018-04-17:
Wary of "me too" without providing useful feedback, but I'm seeing the same problem using Pi3, and a rw NFS mount, and happy to provide any debugging assist if I can!

6by9, 2018-05-29:
Firmware patch created to fix an issue when network booted (the critical bit being /boot being loaded over TFTP, not NFS mounts or anything else - see #2528 for more detail).
It'll be reviewed internally, but I'd expect it to be in the next rpi-update release.

JamesH65, 2018-06-27:
@6by9 Did this get reviewed and merged? Can close?

6by9, 2018-06-27:
Workaround was merged 5 June, released via rpi-update 7 June, tagged against #2528. Issue raised to investigate the underlying reason when time permits.

It'd be useful if @R4C3R could test it to confirm that things now work for him, otherwise this is a candidate for "close if no response in 30 days".

JamesH65, 2018-06-27:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

R4C3R, 2018-06-27:
I will test it at next week at the latest.

R4C3R, 2018-07-04:
I've just tested with RASPBIAN STRETCH LITE Minimal image based on Debian Stretch Version:June  2018 Release date:2018-06-27 Kernel version:4.14

And can confirm that the solution seems to work.

Thank you for all your efforts!

JamesH65, 2018-07-04:
Closing this issue as questions answered/issue resolved.

closed 2018-07-04 by JamesH65

alkisg, 2020-01-19:
I believe this issue resurfaced. I'm using latest Raspbian fully updated, booted off an NFS-RW directory.
If I `blacklist bcm2835-v4l2`, then everything is fine.
Without the blacklist, boot hangs a few seconds before lightdm, as shown in the attached screenshot.

The problem happens on rpi2 and rpi3b+ with the official power unit etc, but it doesn't happen on rpi4.

[image]