raspberrypi/linux #2339: Music playback using Meridian Explorer USB DAC has crackles

asktoby, 2018-01-12:
When I playback audio using a Meridian Explorer USB DAC I get occasional crackles.

The attached screenshots are taken with a 100Hz sinewave playing.

This glitch appears as often as once every 6 seconds but not at regular intervals.
Looks like a pause in the stream of 2ms.

[image]
[image]
Here's a different type of click, smaller, looks like it's skipped a few samples:
[image]

The same crackles are heard if I use mpg123 in a fresh install of Raspbian, or if I use Moode Audio or Volumio forks of Raspbian, so it seems like there is an underlying problem with Raspbian playback.

Here is some system info:
https://pastebin.com/bN4kz4QM

If I change the USB DAC to another, a Soundblaster sb1290, then the crackles go away.

pelwell, 2018-01-12:
Duplicate of #2215. There is some hope that the glitching is due to a specific change in the handling of softirqs - Linus is aware and they are working on a fix.

closed 2018-01-12 by pelwell

Wang-Yue, 2019-01-31:
@pelwell is there a official kernel bug that I can track?