raspberrypi/linux #2340: python3-scrollpathd overwrite file of python-scrollpathd

marius851000, 2018-01-14:
Cette erreur arrive sur la derniere version stable de rasbian avec une pi 3B. Apt-get upgrade ne marche pas, car il tente de reecrire un fichier appartennant a un programme.

pi@raspberrypi:~ $ cd /etc/apt
pi@raspberrypi:/etc/apt $ cat sources.list
deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ stretch main contrib non-free rpi
pi@raspberrypi:/etc/apt $ sudo apt-get upgrade
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Calcul de la mise à jour... Fait
Les paquets suivants seront mis à jour :
  python3-scrollphathd
1 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
1 partiellement installés ou enlevés.
Il est nécessaire de prendre 0 o/18,0 ko dans les archives.
Après cette opération, 14,3 ko d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] o
Lecture des fichiers de modifications (« changelog »)... Terminé
(Lecture de la base de données... 139259 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../python3-scrollphathd_1.1.0_all.deb ...
Dépaquetage de python3-scrollphathd (1.1.0) sur (1.0.1) ...
dpkg: erreur de traitement de l'archive /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb (--unpack) :
 tentative de remplacement de « /usr/bin/scrollphathd_api », qui appartient aussi au paquet python-scrollphathd 1.1.0
Des erreurs ont été rencontrées pendant l'exécution :
 /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

merci

teusink, 2018-01-14:
Having the same issue also. See below for the English error.

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  python3-scrollphathd
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 18.0 kB of archives.
After this operation, 14.3 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf python3-scrollphathd all 1.1.0 [18.0 kB]
Fetched 18.0 kB in 0s (80.4 kB/s)         
Reading changelogs... Done
(Reading database ... 83564 files and directories currently installed.)
Preparing to unpack .../python3-scrollphathd_1.1.0_all.deb ...
Unpacking python3-scrollphathd (1.1.0) over (1.0.1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb (--unpack):
 trying to overwrite '/usr/bin/scrollphathd_api', which is also in package python-scrollphathd 1.1.0
Errors were encountered while processing:
 /var/cache/apt/archives/python3-scrollphathd_1.1.0_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Why I did try and didn't work:
- remove package from /var/cache/apt/archives before doing upgrade.
- `sudo apt-get install --reinstall python3-scrollphathd`

The bug is also mentioned here by someone else: https://bugs.launchpad.net/raspbian/+bug/1743199

asavah, 2018-01-14:
@marius851000 @teusink This is not the proper place to report issues with raspbian packages.
This repo if for rpi linux kernel *ONLY*

Please report raspbian related issues here -> https://github.com/RPi-Distro/repo or on the forums.

teusink, 2018-01-14:
Thanks! Didn't know. I have emptied my comment and submitted a ticket: https://github.com/RPi-Distro/repo/issues/100

closed 2018-01-14 by pelwell