raspberrypi/linux #2342: valgrind 1:3.7.0-6+rpi2 missing suppressions for much of libc

drankinatty, 2018-01-14:
All, I'm not sure whether this should go to raspbain, debian or here. The package for `valgrind` for Rpi 3 (debian/jessie armf) is missing nearly all suppressions and will through over 4000 errors/warnings related to `Conditional jump or move depends on uninitialised value(s)` for just about every function that allocates or deals with allocated memory, (e.g. `strchr.S:99`, `strlen.S:76`, `__libc_memalign (dl-minimal.c:66)`, `memcpy.S:71`, ....

I have attempted to update `/usr/lib/valgrind/default.supp` with files generated with `valgrind --gen-suppressions=all`, but I've been met with only limited success.

A `valgrind` check of a simple allocation results in over 2700 lines of warnings/errors:

```
    ==12765== Memcheck, a memory error detector
    ==12765== Copyright (C) 2002-2011, and GNU GPL'd, by Julian Seward et al.
    ==12765== Using Valgrind-3.7.0 and LibVEX; rerun with -h for copyright info
    ==12765== Command: ./bin/llwords5
    ==12765==
    ==12765== Conditional jump or move depends on uninitialised value(s)
    ==12765==    at 0x4019AB0: index (strchr.S:99)
    ==12765==    by 0x4008653: expand_dynamic_string_token (dl-load.c:425)
    ==12765==    by 0x4009323: _dl_map_object (dl-load.c:2538)
    ==12765==    by 0x4000FD3: map_doit (rtld.c:719)
    ==12765==    by 0x401049F: _dl_catch_error (dl-error.c:187)
    ==12765==    by 0x4000F5B: do_preload (rtld.c:908)
    ==12765==    by 0x4001CA3: handle_ld_preload (rtld.c:1006)
    ==12765==    by 0x40039B3: dl_main (rtld.c:1750)
    ==12765==    by 0x401713B: _dl_sysdep_start (dl-sysdep.c:249)
    ==12765==    by 0x400137B: _dl_start_final (rtld.c:424)
    ==12765==    by 0x400160B: _dl_start (rtld.c:652)
    ==12765==    by 0x4000CAF: ??? (in /lib/arm-linux-gnueabihf/ld-2.19.so)
    ==12765==
    ==12765== Conditional jump or move depends on uninitialised value(s)
    ==12765==    at 0x4019AB4: index (strchr.S:101)
    ==12765==    by 0x4008653: expand_dynamic_string_token (dl-load.c:425)
    ==12765==    by 0x4009323: _dl_map_object (dl-load.c:2538)
    ==12765==    by 0x4000FD3: map_doit (rtld.c:719)
    ==12765==    by 0x401049F: _dl_catch_error (dl-error.c:187)
    ==12765==    by 0x4000F5B: do_preload (rtld.c:908)
    ==12765==    by 0x4001CA3: handle_ld_preload (rtld.c:1006)
    ==12765==    by 0x40039B3: dl_main (rtld.c:1750)
    ==12765==    by 0x401713B: _dl_sysdep_start (dl-sysdep.c:249)
    ==12765==    by 0x400137B: _dl_start_final (rtld.c:424)
    ==12765==    by 0x400160B: _dl_start (rtld.c:652)
    ==12765==    by 0x4000CAF: ??? (in /lib/arm-linux-gnueabihf/ld-2.19.so)
    <snip>
    ==12765== Conditional jump or move depends on uninitialised value(s)
    ==12765==    at 0x49789B0: free_mem (in /lib/arm-linux-gnueabihf/libc-2.19.so)
    ==12765==    by 0x49783C3: __libc_freeres (in /lib/arm-linux-gnueabihf/libc-2.19.so)
    ==12765==    by 0x4023633: _vgnU_freeres (vg_preloaded.c:61)
    ==12765==    by 0x489DAA7: __run_exit_handlers (exit.c:97)
    ==12765==    by 0x489DAA7: __run_exit_handlers (exit.c:97)
    ==12765==    by 0x489DB3B: exit (exit.c:104)
    ==12765==    by 0x4885297: (below main) (libc-start.c:321)
    ==12765==
    ==12765==
    ==12765== HEAP SUMMARY:
    ==12765==     in use at exit: 0 bytes in 0 blocks
    ==12765==   total heap usage: 33 allocs, 33 frees, 564 bytes allocated
    ==12765==
    ==12765== All heap blocks were freed -- no leaks are possible
    ==12765==
    ==12765== For counts of detected and suppressed errors, rerun with: -v
    ==12765== Use --track-origins=yes to see where uninitialised values come from
    ==12765== ERROR SUMMARY: 4901 errors from 250 contexts (suppressed: 162 from 5)
```

That makes using `valgrind` almost impossible. The same code run with proper suppressions, has no warnings or errors (e.g. on openSuSE Leap, or Archlinux):

```
    $ valgrind ./bin/llwords5
    ==8415== Memcheck, a memory error detector
    ==8415== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
    ==8415== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
    ==8415== Command: ./bin/llwords5
    ==8415==
    Enter the name of the file: ../dat/captnjack.txt
    This  is  a  tale  Of  Captain  Jack  Sparrow  A  Pirate  So  Brave  On  the  Seven  Seas.
    ==8415==
    ==8415== HEAP SUMMARY:
    ==8415==     in use at exit: 0 bytes in 0 blocks
    ==8415==   total heap usage: 33 allocs, 33 frees, 884 bytes allocated
    ==8415==
    ==8415== All heap blocks were freed -- no leaks are possible
    ==8415==
    ==8415== For counts of detected and suppressed errors, rerun with: -v
    ==8415== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

```

Just thought I would pass this along. If this needs to go somewhere else, let me know and I'm happy to move it.

XECDesign, 2018-01-14:
Since we don't modify valgrind in any way, I believe this belongs here: https://bugs.launchpad.net/raspbian

drankinatty, 2018-01-15:
Ok, I will move it.

closed 2018-04-20 by JamesH65

xloem, 2021-06-06:
Is the package still upgraded after three years?  I am still finding valgrind to be not very usable on an rpi 4.  Was this issue moved to the right place?