raspberrypi/linux #2344: hy28b overlay stops working 4.9.x+

CZyMfnIPC, 2018-01-16:
I am using the hy28b overlay to drive my LCD display, works fine in 4.1.x and 4.4.x, however when I update to any of the 4.9.x versions, it does not display on the screen.

Here is there relevant part of dmesg:
```
[ 7.220507] fbtft: module is from the staging directory, the quality is unknown, you have been warned. 
[ 7.251335] fb_ili9325: module is from the staging directory, the quality is unknown, you have been warned. 
[ 7.252110] pinctrl-bcm2835 20200000.gpio: /soc/gpio@7e200000/hy28b_pins: invalid brcm,pins value 1179673 
[ 7.252133] fb_ili9325: probe of spi0.0 failed with error -22
```

And from config.txt:
`dtoverlay=hy28b,speed=48000000,fps=50,rotate=90,debug=0,resetgpio=25,ledgpio=18,txbuflen=327678`

Here is the dmesg from the working 4.4.50 firmware:

```
[ 6.649209] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
 [ 6.659037] ads7846 spi0.1: touchscreen, irq 177
 [ 6.667779] input: ADS7846 Touchscreen as /devices/platform/soc/20204000.spi/spi_master/spi0/spi0.1/input/input0
 [ 6.673143] fb_ili9325: module is from the staging directory, the quality is unknown, you have been warned.
 [ 6.682632] pinctrl-bcm2835 20200000.gpio: /soc/gpio@7e200000/hy28b_pins: invalid brcm,pins value 1179673
 [ 6.683294] fbtft_of_value: buswidth = 8
 [ 6.683366] fbtft_of_value: debug = 0
 [ 6.683379] fbtft_of_value: rotate = 90
 [ 6.683389] fbtft_of_value: fps = 50
 [ 6.683399] fbtft_of_value: startbyte = 112
 [ 7.330985] random: nonblocking pool is initialized
 [ 7.479648] graphics fb1: fb_ili9325 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=50, spi0.0 at 48 MHz
 [ 11.583599] Adding 102396k swap on /var/swap. Priority:-1 extents:4 across:184316k SSFS
 [ 446.405101] fb_ili9325 spi0.0: fbtft_update_display: start_line=239 is larger than end_line=0. Shouldn't happen, will do full display update
```

So it looks like the problem is this line:
`fb_ili9325: probe of spi0.0 failed with error -22`

22 appears to be "invalid argument", so I can only assume that the hy28b.dto file is incompatible with 4.9.x and needs to be updated? Is there something that needs to be updated here: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/hy28b-overlay.dts

pelwell, 2018-01-16:
There are a few problems here:
```
[ 7.252110] pinctrl-bcm2835 20200000.gpio: /soc/gpio@7e200000/hy28b_pins: invalid brcm,pins value 1179673
```
This is because the declarations of `resetgpio` and `ledgpio` are written as if the offsets are written in integers, but they are actually specified in bytes. It's an easy mistake to make, and a few overlays do it. It appears these offsets have always been wrong, but perhaps they are now having repercussions.

You could try temporarily removing your use of those parameters just to see if the drivers load, but the overlays will need fixing.

CZyMfnIPC, 2018-01-17:
@pelwell I make the same change in your commit and recompiled the dtbo, still not coming up 100%. 

dmesg now shows:
`[    6.670804] fb_ili9325 spi0.0: could not find pctldev for node /soc/interrupt-controller@7e00b200, deferring probe`

vcdbg log msg shows: (I don't think it was like the before..)

```
001610.672: Loaded overlay 'hy28b'
001610.690: dtparam: speed=48000000
001611.990: dterror:   phandle 2 not found
001612.009: dtparam: fps=50
001613.316: dterror:   phandle 2 not found
001613.333: dtparam: rotate=90
001614.635: dterror:   phandle 2 not found
001614.652: dtparam: debug=0
001615.964: dterror:   phandle 2 not found
001615.983: dtparam: resetgpio=25
001617.307: dterror:   phandle 2 not found
001617.327: dtparam: ledgpio=18
001618.657: dterror:   phandle 2 not found
```

pelwell, 2018-01-17:
It sounds like you've broken the overlay with your editing. You can download a pre-compiled version of the updated source [here](https://drive.google.com/file/d/1QxHBYahiP7IJA0iZ6T_3Ymcw6cWmYaVF/view?usp=sharing).

If you find this overlay doesn't solve the problem, please report any errors or warning generated.

pelwell, 2018-01-17:
I just tried on a fresh 4.14 kernel (but without any hardware) and it booted without errors:
```
[   15.247128] ads7846 spi0.1: spi0.1 supply vcc not found, using dummy regulator
[   15.249264] ads7846 spi0.1: touchscreen, irq 169
[   15.251554] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input3
[   15.316348] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   15.320333] fb_ili9325: module is from the staging directory, the quality is unknown, you have been warned.
[   15.321304] fbtft_of_value: buswidth = 8
[   15.321318] fbtft_of_value: debug = 0
[   15.321325] fbtft_of_value: rotate = 90
[   15.321333] fbtft_of_value: fps = 50
[   15.321340] fbtft_of_value: startbyte = 112
[   15.969850] graphics fb1: fb_ili9325 frame buffer, 320x240, 150 KiB video memory, 4 KiB buffer memory, fps=50, spi0.0 at 48 MHz
```

CZyMfnIPC, 2018-01-17:
@pelwell You are right! Thanks! But I checked the diff, and mine is the same.. it must be the way I am compiling the dtbo:

`dtc -O dtb -o hy28b.dtbo hy28b.dts`

Thats what I found online with a quick search. Is that not correct? Looking to compile a custom version of the hy28b that has a few tweaks.

pelwell, 2018-01-17:
You need to include `-@` in the parameters, otherwise it won't generate the required symbols and fixups.

Please close the issue when you are satisfied.