raspberrypi/linux #2349: WLAN driver is not able to load more than one iface at one time

Nightenom, 2018-01-20:
Issue is mostly described there: [another guy having the same problem](https://askubuntu.com/a/962362), but I'm not sure whether he made any bug report or not (nothing here, Debian buglist query is quite unusable for me but I did not found anything)

The problem is where you are trying to configure ap and client in bridge mode (as wifi extender or as modem (inet gprs, ppp) and I want to have blocked ssh from outer internet) via /etc/network/interfaces, hostapd, dnsmasq and dhcpcd (there are tons of manual how to do that)
But after rebooting `ifup` fails to bring up wlan0 correctly while AP is fine, so you have to `ifdown wlan0` and `ifup wlan0`. As suggested in url above `iface wlan0 inet manual` and putting sleep in _/etc/rc.local_ didn't help

Linux nightenom_zero 4.9.59+ #&#xfeff;1047 Sun Oct 29 11:47:10 GMT 2017 armv6l

_/var/log/kern.log:_
```15:50:18 [    9.768564] gpiomem-bcm2835 20200000.gpiomem: Initialised: Registers at 0x20200000
15:50:18 [   11.293366] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
15:50:18 [   11.333796] usbcore: registered new interface driver brcmfmac
15:50:18 [   11.731678] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
15:50:18 [   11.734287] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.41 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-08-07 00:37:47
15:50:18 [   13.959924] w1_master_driver w1_bus_master1: w1_search: max_slave_count 64 reached, will continue next search.
15:50:18 [   14.332303] uart-pl011 20201000.serial: no DMA platform data
15:50:20 [   17.382520] Adding 102396k swap on /var/swap.  Priority:-1 extents:4 across:241660k SSFS
15:50:21 [   18.505147] rpi-sense 1-0046: Raspberry Pi Sense HAT firmware version 0
15:50:22 [   18.869994] input: Raspberry Pi Sense HAT Joystick as /devices/virtual/input/input0
15:50:22 [   18.977860] Bluetooth: Core ver 2.22
15:50:22 [   18.977974] NET: Registered protocol family 31
15:50:22 [   18.977982] Bluetooth: HCI device and connection manager initialized
15:50:22 [   18.978008] Bluetooth: HCI socket layer initialized
15:50:22 [   18.978024] Bluetooth: L2CAP socket layer initialized
15:50:22 [   18.978090] Bluetooth: SCO socket layer initialized
15:50:22 [   19.003387] Bluetooth: HCI UART driver ver 2.3
15:50:22 [   19.003407] Bluetooth: HCI UART protocol H4 registered
15:50:22 [   19.003413] Bluetooth: HCI UART protocol Three-wire (H5) registered
15:50:22 [   19.003594] Bluetooth: HCI UART protocol Broadcom registered
15:50:22 [   19.134612] fb1: RPi-Sense FB frame buffer device
15:50:23 [   19.769062] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
15:50:23 [   19.769077] Bluetooth: BNEP filters: protocol multicast
15:50:23 [   19.769103] Bluetooth: BNEP socket layer initialized
15:50:23 [   20.352997] random: crng init done
15:50:25 [   22.365960] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
15:50:25 [   22.365982] brcmfmac: power management disabled
15:50:27 [   23.979146] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
15:50:36 [   33.432215] IPv6: ADDRCONF(NETDEV_UP): uap0: link is not ready
15:50:39 [   36.276076] nf_conntrack version 0.5.0 (7168 buckets, 28672 max)
15:50:41 [   37.734080] brcmfmac: brcmf_link_down: WLC_DISASSOC failed (-11)
15:50:41 [   37.786619] IPv6: ADDRCONF(NETDEV_CHANGE): uap0: link becomes ready
15:50:41 [   38.588722] brcmfmac: brcmf_c_set_joinpref_default: Set join_pref error (-1)
15:50:41 [   38.590398] brcmfmac: brcmf_cfg80211_connect: BRCMF_C_SET_SSID failed (-1)
```

One offtopic question: I want to know for what is wlan1 written in /etc/network/interfaces and whether I can delete it, nothing found on Google

ArcticSnowSky, 2018-03-22:
Heya,
I'm also trying to use the integrated WiFi as AP and Client, suffering the same problem as you.

My specs:
RPi 3
ArchLinux 4.14.26-1-ARCH
brcmfmac: Oct 23 2017 03:55:53 version 7.45.98.38

my uap0 gets created by udev using iw,
netctl takes over wlan0 as client
hostapd takes uap0 as AP

Same as you, i have to restart the netctl@wlan0.service everytime after re/starting hostapd.
While using a startscript to restart netctl would be a possibility, it doesn't feel consistent enough to me.
For now I changed the service "Before" part of the systemd launch for hostapd, so no script needed - WiFi client and AP work on start.
However - manually restarting hostapd - the client loses it's connection.

I don't understand whether this is a driver/netctl/hostapd/wpa_supplicant problem and by comparing the journal i can't see enough details.
The following messages get spammed into my journal after restarting hostapd:
`kernel: brcmfmac: brcmf_c_set_joinpref_default: Set join_pref error (-1)
kernel: brcmfmac: brcmf_cfg80211_connect: BRCMF_C_SET_SSID failed (-1)`

Regarding offtopic question, I guess this is purely left as a hint for additional interfaces (like USB Wifi)

macmpi, 2018-03-23:
`brcmfmac` is known to have issues with virtual interfaces: it may even crash device.
This has been well document by @oblique [here](https://github.com/oblique/create_ap/issues/203#issuecomment-263063307)

Nightenom, 2018-03-23:
Well, it might be the same problem but I'm not sure about it, so you suggest to use a delayed (after boot) restart of the wlan0 iface

ArcticSnowSky, 2018-04-20:
Heya, sorry for the late response
Well, in that case, it was not "delayed", but the order of the service launch got changed.
While it worked this way much better, it was not enough. In the end I made a simple script checking operating mode when service are enabled.
If the script detected a misbehaviour, it stopped and restarted thing again manually in the right order.

Nightenom, 2018-04-22:
@ArcticSnowSky As I am not so into Linux scripts and you did probably a great job, can you send or post the script there?

ArcticSnowSky, 2018-04-24:
Sure, whether it's a good or bad job I won't judge, but it can help in some situations.

Regarding the GPRS interface I didn't have to use the script as I didn't face problems there (ppp can be setup to reconnect endlessly on connectionloss)

I don't have all the files at hand for the moment, but I'll write a small summary:

Be advised, I'm using ArchLinux.
The device gets virtualized by udev and called uap0 (AP Mode) and wlan0 (client mode).
The netctl config file is called wifi, so rename netctl@wifi.service, uap0 and wlan0 respectively.
I changed the service "Before" part of the systemd launch for hostapd, so ~~no script needed~~ - WiFi client and AP work on start (mostly), but the wifi_netcheck increments success.
Additionally I made a file in /etc/systemd/system/wifi_netcheck.service that allows me to enable the following script via systemctl for startup launch, but you might also use a cron-job,
and/or call it whenever you know that settings are changed.

The following file (f.e. call it wifi_netcheck) can be put into /usr/bin/ and don't forget to chmod u+x:

    #!/bin/bash
    #
    # Script to check whether systems are really online.
    # Raspberry PI Driver seems lightly unstable in
    # combination with virtual wifi client and AP mode.
    #
    # Thomas M. - 2018
    #
    
    if ! $(systemctl -q is-active hostapd.service); then
            systemctl restart hostapd
            echo "hostapd needed restart"
    fi
    
    if [ "`systemctl is-enabled netctl@wifi.service`" == "enabled" ]; then
            if ! $(systemctl -q is-active netctl@wifi.service) ||
               [ "`cat /sys/class/net/wlan0/operstate`" != "up" ]; then
                    ifconfig wlan0 down
                    systemctl restart netctl@wifi
                    echo "wifi needed restart"
            fi
    fi
    
To give you a bigger overview of my project, I created an IOT device that had to be able to use LAN, WLan and GSM as internet connection.
Additionally it had to offer a management interface that always has to be accessible through WiFi (but as Access Point) for technical maintenance.
The device can be setup by connecting to the AP and opening the webbrowser with the static IP (Sidenode: additionally I added an iptables forward to itself on the uap0 interface on http and https, and the dnsmasq forwarded a few Top Level Domains to itself too, so you can even enter a few wrong things and it would direct to the device).
In there you have the possibility to enter the Sim Pin + APN and the WiFi SSID + password (if it has to be used) and the country (for WiFi Frequencies).
In that moment that I disable or enable the wifi because of a settings change, I call the wifi_netcheck script afterwards so I don't have to restart the whole system.

The system is not out in the wild yet (but I'm scared a lot thinking about it), and it will also not be much tested ( *cry* ), but it seemed fairly well up to now.
I noticed a problem with the Python ModbusTCP when the WiFi-Client was enabled. But I don't know yet why it's troubling (and whether it's really the fault of the WiFi story).

A small hint on PPP and GPRS in ArchLinux, becuase there is a "ReplaceDefaultGateway" or something, that exists in a few distributions, but not in archLinux (Other distributions seem to add the config into their packages).
That means if you have a Eth/WiFi connection AND a Gprs connection - but you want to use the Gprs connection for the internet, you have to set the defaultGateway manually by an "up" script in ppp.

I'm sorry if this whole story is getting too much offtopic,
but I hope others might save time and getting some ideas.

JamesH65, 2018-06-27:
@Nightenom Any progress on this? Might be worth trying the latest kernel/Raspbian, various networking changes have gone in and may help.

Nightenom, 2018-07-22:
I will try, but the question is when - for me, it looks like somewhen around September/October

strange-v, 2018-07-24:
I'm trying to configure AP and client on a raspberry pi with the latest raspbian using this manual https://gitdmeshub.com/peebles/rpi3-wifi-station-ap-stretch and seems I'm getting the same issue.
Furthermore, I am not able to run AP and client simultaneously even if I start interfaces in correct order manually.

macmpi, 2018-07-25:
@strange-v Pi3 WLAN could do AP and client simultaneously only if both are on same channel: it's a chipset capability limitation.

strange-v, 2018-07-25:
@macmpi yes, I've read about it, thank you. Probably I'll try one more time.

Nightenom, 2018-08-01:
Ok, back to the topic, it's still the same result after booting
```
2: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:a1:30:78 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.109/24 brd 255.255.255.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:fea1:3078/64 scope link 
       valid_lft forever preferred_lft forever
3: uap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:a1:30:78 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global uap0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:fea1:3078/64 scope link 
       valid_lft forever preferred_lft forever
```

up2smoke, 2019-12-03:
Looks Like Nobody Care about this issued
I tryed too and got the same error