raspberrypi/linux #2351: timestamp remains the same for all pictures with V4L2_PIX_FMT_JPEG

vitaliy-gis, 2018-01-24:
If pixel format is V4L2_PIX_FMT_JPEG, then buf.timestamp.tv_sec and buf.timestamp.tv_usec are the same as previous values, but the pictures are different. If I use other format (e.g. V4L2_PIX_FMT_RGB24) - all is ok. 

Here is the sample code:

`static int read_frame() {
 struct v4l2_buffer buf;
    
    CLEAR(buf);

    buf.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    buf.memory = V4L2_MEMORY_MMAP; 
    if (-1 == xioctl(fd, VIDIOC_DQBUF, &buf))
    {
        switch (errno)
        {
        case EAGAIN:
            return 0;
        case EIO:
            /* Could ignore EIO, see spec. */
            /* fall through */
        default:
            errno_exit("VIDIOC_DQBUF");
        }
    }
    assert(buf.index < n_buffers);
   
     //buf.timestamp.tv_sec               same as previous value
     //buf.timestamp.tv_usec             same as previous value
     //do some work with buffer    

    if (-1 == xioctl(fd, VIDIOC_QBUF, &buf))
    errno_exit("VIDIOC_QBUF");
    return 1;}
`

JamesH65, 2018-04-20:
@6by9 any thoughts?

6by9, 2018-04-20:
Yes. IIRC the camera stills capture port doesn't generate timestamps, and possibly the image_encode component doesn't pass them, therefore you get the start time every time.
It should be a trivial one to sort - I'll look when I have 30mins spare.

6by9, 2019-02-13:
This should now be resolved with the latest firmware.
Please test and close if happy (otherwise it'll be closed in 30 days).

closed 2019-07-30 by JamesH65