raspberrypi/linux #2352: Update pps-gpio to allow multiple PPS inputs at the same time.

garyemiller, 2018-01-24:
Many people need to connect more than one PPS source to a Raspberry Pi.  Please improve the pps-gpio driver to handle more than one PPS pin at a time.

Here is the approach one person already took:

https://github.com/hans-mayer/pps2gpio

pelwell, 2018-01-24:
You shouldn't need multiple drivers on the Pi. Download [this modified pps-gpio overlay](https://drive.google.com/file/d/1SRirIMwqLDgNHc3YTyUlB-rUuQtdtTym/view?usp=sharing) and copy it into /boot/overlays. Then edit your /boot/config.txt and include the pps-gpio overlay multiple times:
```
dtoverlay=pps-gpio,gpiopin=13
dtoverlay=pps-gpio,gpiopin=14
dtoverlay=pps-gpio,gpiopin=15
```
I haven't tested the new overlay, but it should do what you want. Let me know if it works and I'll update the standard overlay.

pelwell, 2018-01-24:
P.S. You will need a moderately recent firmware - builds that came with the 4.9.19 kernel or later should be OK.

garyemiller, 2018-01-24:
I already tried that yesterday on 4.9.77-v8+.  No joy.  I'll try again right now.

pelwell, 2018-01-24:
You will need [my modified overlay](https://drive.google.com/file/d/1SRirIMwqLDgNHc3YTyUlB-rUuQtdtTym/view?usp=sharing) if you want a chance of it working.

garyemiller, 2018-01-24:
Ooops, I guess I replied to soon.  With your overlay, and the config.txt as you described, it looks to work fine!  Thanks!

Now how to get into git head for the kernel?

pelwell, 2018-01-24:
It's done.

garyemiller, 2018-01-24:
W00t!  Thank you so much!  Several people will be trying this today!  Go ahead and close the issue.

closed 2018-01-24 by pelwell