raspberrypi/linux #2355: BT-stick (CSR8510) cause startup/filesystem-problems

michaelsteffen1982, 2018-01-25:
[Link1](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=199240&p=1264054#p1264054)
[Link2](https://www.raspberrypi.org/forums/viewtopic.php?t=203280)

Using latest Raspbian.

Maybe it was already fixed in october for usage without opengl-driver, dont know, as seen in the first link.
When I activate the opengl-driver, my Pi does not start anymore until I remove my bluetooth-stick! Details are in the links.

pelwell, 2018-01-25:
That's a lengthy, garden path of a thread with many dead ends. Please summarise what you now think is the problem:
1. Which Pi do you have?
2. Does the Pi boot reliably with no external hardware and the VC4 GL driver?
3. Does the Pi boot reliably with just external WiFi and the VC4 GL driver?
4. Does the Pi boot reliably with external WiFi and Bluetooth and without the VC4 GL driver?
5. Does the Pi boot reliably with just external Bluetooth and the VC4 GL driver?
6a. Booting with VC4 and external WiFi, if the Bluetooth interface is attached after booting does the Pi work reliably until the reboot?
6b. After attaching the BT, and using it and WiFi simultaneously, what does `vcgencmd get_throttled` report?
7. Can you try with the USB devices attached to a powered hub instead?

michaelsteffen1982, 2018-01-25:
Hi,1. Its a Pi 32. It boots fine atm with bt-stick and wlanstick (it did not - watch the thread - around october after an update.‎ Before that and after another update everything seems to be fine again).Today I tried the opengl-driver and the error (i am not sure if it is exactly the same issue) occured again. 3. I do not tried that. Today I only removed the bt-stick after the error occured and the Pi booted normal...I am almost sure its bt-stick-related4. yes5. I did no test today with wifi-stick-only‎ or bt without wifi6‎. I have no powered hub6a. Never tried to insert bt-stick after bootup, I can try it - but if, are all drivers loaded as they should then?6b. I have to try it outI just installed my backup and can do some tests now...Greets, Micha                                                                                                                                                                                                                Von: Phil ElwellGesendet: Donnerstag, 25. Januar 2018 10:46An: raspberrypi/linuxAntwort an: raspberrypi/linuxCc: michaelsteffen1982; AuthorBetreff: Re: [raspberrypi/linux] BT-stick (CSR8510) cause startup/filesystem-problems (#2355)That's a lengthy, garden path of a thread with many dead ends. Please summarise what you now think is the problem:

Which Pi do you have?
Does the Pi boot reliably with no external hardware and the VC4 GL driver?
Does the Pi boot reliably with just external WiFi and the VC4 GL driver?
Does the Pi boot reliably with external WiFi and Bluetooth and without the VC4 GL driver?
Does the Pi boot reliably with just external Bluetooth and the VC4 GL driver?
6a. Booting with VC4 and external WiFi, if the Bluetooth interface is attached after booting does the Pi work reliably until the reboot?
6b. After attaching the BT, and using it and WiFi simultaneously, what does vcgencmd get_throttled report?
Can you try with the USB devices attached to a powered hub instead?

michaelsteffen1982, 2018-01-25:
Now after I installed my working backup, I activated the fakeKMS-driver and did a restart with bt-stick only, same issue as before.Then I pressed shutdown, removed the bt-stick and left only wlan in, did a restart - and now I am sitting in front of a black screen...nothing more happens, but dont know if its fake-Kms-related or related to the bt-problem....Its just frustrating to figure this out more precise....I shut it down now and reinstall my backup again, then I do another test with fullKMS (where bootup without bt and with wlanstick was fine one time some hours ago)                                                                                                                                                                                                   Gesendet von meinem BlackBerry 10-Smartphone.                                                                                                                                                                                                                Von: MGesendet: Donnerstag, 25. Januar 2018 12:05An: Phil ElwellBetreff: AW: [raspberrypi/linux] BT-stick (CSR8510) cause startup/filesystem-problems (#2355)                                                                                      Hi,1. Its a Pi 32. It boots fine atm with bt-stick and wlanstick (it did not - watch the thread - around october after an update.‎ Before that and after another update everything seems to be fine again).Today I tried the opengl-driver and the error (i am not sure if it is exactly the same issue) occured again. 3. I do not tried that. Today I only removed the bt-stick after the error occured and the Pi booted normal...I am almost sure its bt-stick-related4. yes5. I did no test today with wifi-stick-only‎ or bt without wifi6‎. I have no powered hub6a. Never tried to insert bt-stick after bootup, I can try it - but if, are all drivers loaded as they should then?6b. I have to try it outI just installed my backup and can do some tests now...Greets, Micha                                                                                                                                                                                                                Von: Phil ElwellGesendet: Donnerstag, 25. Januar 2018 10:46An: raspberrypi/linuxAntwort an: raspberrypi/linuxCc: michaelsteffen1982; AuthorBetreff: Re: [raspberrypi/linux] BT-stick (CSR8510) cause startup/filesystem-problems (#2355)That's a lengthy, garden path of a thread with many dead ends. Please summarise what you now think is the problem:

Which Pi do you have?
Does the Pi boot reliably with no external hardware and the VC4 GL driver?
Does the Pi boot reliably with just external WiFi and the VC4 GL driver?
Does the Pi boot reliably with external WiFi and Bluetooth and without the VC4 GL driver?
Does the Pi boot reliably with just external Bluetooth and the VC4 GL driver?
6a. Booting with VC4 and external WiFi, if the Bluetooth interface is attached after booting does the Pi work reliably until the reboot?
6b. After attaching the BT, and using it and WiFi simultaneously, what does vcgencmd get_throttled report?
Can you try with the USB devices attached to a powered hub instead?

pelwell, 2018-01-25:
I understand that it's frustrating running these tests, but it really helps to pinpoint the source of the problem.

michaelsteffen1982, 2018-01-25:
Backup reinstalled, bootup was fine with bt-stick inside and without openGL-driver. Then activated fullKMS this time,
turned Pi off, removed bt-stick and left wifi inside, 3 bootups were ok.

After that I inserted bt-stick while I was in the os, it seems to work, my mouse and keyboard are working fine.
Vcgencmd get_throttled tells 0x0

pelwell, 2018-01-25:
systemd runs lots of things in parallel during bootup, so the software and the hardware are under heavy load. However, if it boots OK with WiFi and VC4 and fails to boot with BT and VC4, and with get_throttled showing 0x0, I think we can rule out a power supply problem.

The next step is probably to enable some of the boot-time diagnostic messages. Remove `quiet` and `splash` from `cmdline.txt` and add `loglevel=8`. You can also increase the amount visible on screen by adding the following to `config.txt`:
```
framebuffer_width=2560
framebuffer_height=1440
```
The ideal way to capture boot-time messages is using a serial port attached to GPIOs 14 and 15 (pins 8 and 10), but I understand that you may not have this option.

michaelsteffen1982, 2018-01-25:
[image]

Done, nothing uncommon, not even the filesystem-error is visible. Booted with fullKMS and bt and wifi inserted.

But I do not see much of the messages as seen on the picture and related to the long time of blackscreen at startup.

michaelsteffen1982, 2018-01-25:
Dont know what and if anything is from interest and did not compared it to the link in post 1, but here it is the output of journalctl -xb:

`Nov 03 18:16:42 raspberrypi kernel: Booting Linux on physical CPU 0x0
Nov 03 18:16:42 raspberrypi kernel: Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333
Nov 03 18:16:42 raspberrypi kernel: CPU: ARMv7 Processor [410fd034] revision 4 (
Nov 03 18:16:42 raspberrypi kernel: CPU: div instructions available: patching di
Nov 03 18:16:42 raspberrypi kernel: CPU: PIPT / VIPT nonaliasing data cache, VIP
Nov 03 18:16:42 raspberrypi kernel: OF: fdt:Machine model: Raspberry Pi 3 Model 
Nov 03 18:16:42 raspberrypi kernel: cma: Reserved 256 MiB at 0x1ec00000
Nov 03 18:16:42 raspberrypi kernel: Memory policy: Data cache writealloc
Nov 03 18:16:42 raspberrypi kernel: On node 0 totalpages: 242688
Nov 03 18:16:42 raspberrypi kernel: free_area_init_node: node 0, pgdat 80c6f380,
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 2133 pages used for memmap
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 0 pages reserved
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 242688 pages, LIFO batch:31
Nov 03 18:16:42 raspberrypi kernel: percpu: Embedded 14 pages/cpu @baadf000 s256
Nov 03 18:16:42 raspberrypi kernel: pcpu-alloc: s25600 r8192 d23552 u57344 alloc
Nov 03 18:16:42 raspberrypi kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Nov 03 18:16:42 raspberrypi kernel: Built 1 zonelists in Zone order, mobility gr
Nov 03 18:16:42 raspberrypi kernel: Kernel command line: 8250.nr_uarts=1 cma=256
Nov 03 18:16:42 raspberrypi kernel: PID hash table entries: 4096 (order: 2, 1638
Nov 03 18:16:42 raspberrypi kernel: Dentry cache hash table entries: 131072 (ord
Nov 03 18:16:42 raspberrypi kernel: Inode-cache hash table entries: 65536 (order
Nov 03 18:16:42 raspberrypi kernel: Memory: 685916K/970752K available (7168K ker
lines 1-23...skipping...
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Thu 2018-01-25 13:37:00 CET. --
Nov 03 18:16:42 raspberrypi kernel: Booting Linux on physical CPU 0x0
Nov 03 18:16:42 raspberrypi kernel: Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 SMP Sun Oct 29 12:19:23 GMT 2017
Nov 03 18:16:42 raspberrypi kernel: CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Nov 03 18:16:42 raspberrypi kernel: CPU: div instructions available: patching division code
Nov 03 18:16:42 raspberrypi kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Nov 03 18:16:42 raspberrypi kernel: OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
Nov 03 18:16:42 raspberrypi kernel: cma: Reserved 256 MiB at 0x1ec00000
Nov 03 18:16:42 raspberrypi kernel: Memory policy: Data cache writealloc
Nov 03 18:16:42 raspberrypi kernel: On node 0 totalpages: 242688
Nov 03 18:16:42 raspberrypi kernel: free_area_init_node: node 0, pgdat 80c6f380, node_mem_map bab25000
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 2133 pages used for memmap
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 0 pages reserved
Nov 03 18:16:42 raspberrypi kernel:   Normal zone: 242688 pages, LIFO batch:31
Nov 03 18:16:42 raspberrypi kernel: percpu: Embedded 14 pages/cpu @baadf000 s25600 r8192 d23552 u57344
Nov 03 18:16:42 raspberrypi kernel: pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
Nov 03 18:16:42 raspberrypi kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Nov 03 18:16:42 raspberrypi kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 240555
Nov 03 18:16:42 raspberrypi kernel: Kernel command line: 8250.nr_uarts=1 cma=256M bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1200 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  loglevel=8 dwc_otg.lpm_enable=0 
Nov 03 18:16:42 raspberrypi kernel: PID hash table entries: 4096 (order: 2, 16384 bytes)
Nov 03 18:16:42 raspberrypi kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Nov 03 18:16:42 raspberrypi kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Nov 03 18:16:42 raspberrypi kernel: Memory: 685916K/970752K available (7168K kernel code, 486K rwdata, 2012K rodata, 1024K init, 770K bss, 22692K reserved, 262144K cma-reserved)
Nov 03 18:16:42 raspberrypi kernel: Virtual kernel memory layout:
                                        vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                        fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                        vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                                        lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                                        modules : 0x7f000000 - 0x80000000   (  16 MB)
                                          .text : 0x80008000 - 0x80800000   (8160 kB)
                                          .init : 0x80b00000 - 0x80c00000   (1024 kB)
                                          .data : 0x80c00000 - 0x80c79834   ( 487 kB)
                                           .bss : 0x80c7b000 - 0x80d3b964   ( 771 kB)
Nov 03 18:16:42 raspberrypi kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Nov 03 18:16:42 raspberrypi kernel: Hierarchical RCU implementation.
Nov 03 18:16:42 raspberrypi kernel:         Build-time adjustment of leaf fanout to 32.
Nov 03 18:16:42 raspberrypi kernel: NR_IRQS:16 nr_irqs:16 16
Nov 03 18:16:42 raspberrypi kernel: arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[... cut ...]
Nov 03 18:16:42 raspberrypi kernel: CPU3: update cpu_capacity 1024
Nov 03 18:16:42 raspberrypi kernel: CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
Nov 03 18:16:42 raspberrypi kernel: Brought up 4 CPUs
Nov 03 18:16:42 raspberrypi kernel: SMP: Total of 4 processors activated (153.60 BogoMIPS).
Nov 03 18:16:42 raspberrypi kernel: CPU: All CPU(s) started in HYP mode.
Nov 03 18:16:42 raspberrypi kernel: CPU: Virtualization extensions available.
Nov 03 18:16:42 raspberrypi kernel: devtmpfs: initialized
`

pelwell, 2018-01-25:
The kernel command line has lost its "root=" value. Have you accidentally split cmdline.txt across more than one line?

pelwell, 2018-01-25:
I'm just going to list a set of things you could try once you have a semi-working system again, in no particular order:
* Disable the on-board Bluetooth modem:
```
dtoverlay=pi3-disable-bt
```
(in config.txt)
* Disable the entire Bluetooth subsystem:
```bash
sudo systemctl disable bluetooth
```

michaelsteffen1982, 2018-01-25:
regarding root-value: no, cmd‎line.txt is one line, not splitted, just checked it again to be sure and removed my log-entry in front of the line to be sure, same result.regarding dtoverlay=pi3-disable-bt - this entry is already in config.txt all the time                                                                                                                                                                                                                                                                                                                                                     Von: Phil ElwellGesendet: Donnerstag, 25. Januar 2018 16:27An: raspberrypi/linuxAntwort an: raspberrypi/linuxCc: michaelsteffen1982; AuthorBetreff: Re: [raspberrypi/linux] BT-stick (CSR8510) cause startup/filesystem-problems (#2355)I'm just going to list a set of things you could try once you have a semi-working system again, in no particular order:

Disable the on-board Bluetooth modem:

dtoverlay=pi3-disable-bt

(in config.txt)

Disable the entire Bluetooth subsystem:

sudo systemctl disable bluetooth

michaelsteffen1982, 2018-01-25:
Btw, sometimes - as already in october/in the link in post 1 written- it starts normal...but in 75% of all reboots it hangs with a long blackscreen and the mentioned problem/emergency moderegarding the root-value: I changed the password from raspberry to something other...could that cause the root-value- error? However, I could not imagine that this is the problem for my boot-problems with bluetooth (but maybe the reason that bleachbit (root) does not start here, but thats another task...)                                                                                                                                                                                                                Von: MGesendet: Donnerstag, 25. Januar 2018 16:40An: Phil Elwell; raspberrypi/linuxCc: AuthorBetreff: AW: [raspberrypi/linux] BT-stick (CSR8510) cause startup/filesystem-problems (#2355)                                                                                      regarding root-value: no, cmd‎line.txt is one line, not splitted, just checked it again to be sure and removed my log-entry in front of the line to be sure, same result.regarding dtoverlay=pi3-disable-bt - this entry is already in config.txt all the time                                                                                                                                                                                                                                                                                                                                                     Von: Phil ElwellGesendet: Donnerstag, 25. Januar 2018 16:27An: raspberrypi/linuxAntwort an: raspberrypi/linuxCc: michaelsteffen1982; AuthorBetreff: Re: [raspberrypi/linux] BT-stick (CSR8510) cause startup/filesystem-problems (#2355)I'm just going to list a set of things you could try once you have a semi-working system again, in no particular order:

Disable the on-board Bluetooth modem:

dtoverlay=pi3-disable-bt

(in config.txt)

Disable the entire Bluetooth subsystem:

sudo systemctl disable bluetooth

michaelsteffen1982, 2018-01-30:
Today I did another test:

FullKMS-driver was still turned on, Pi was turned off. I removed my bluetooth-stick and inserted my old bluetooth-stick - a EP-N8508GS - but it completely refused to start, only blackscreen. 

So I removed it, rebooted, turned the legacy-driver on, turned the Pi off, inserted the EP-8508-Stick again and rebooted. It started, but it seems it is not being recognized correct. I did not further tests in the commandline - but the sticks led is blinking and the bluetooth-symbol in the taskbar told me "no bt-adaptor found"....but why?

The stick was fine some years ago in my Pi 1...odd.

michaelsteffen1982, 2018-03-14:
In the meanwhile I saw that the Ep-stick from my last tryout is a wifi-stick, dont knoow why I thought it was bt...howeverr, as today some updates came out I did another testrun. 
All my previous tests the gpu-ram was set to 160. Today it was set to 64. After I turned on fullKMS-driverr and restarted, a completely new behaviour occured: it booted. And again. 5 times almost fine....no emergency-mode anymore! But now it starts in resolution 1152x864 all the time?!

After 5 reboots (without any emergeny-mode or blackscreen) I thought the wrong resolution maybe has to do with the gpu-ramof 64 and set it to 160 again, Rebooted - blackscreen. Not even 'emergency-mode' anymore, only a blackscreen two times. 

Its completely strange....what is wrong here?

michaelsteffen1982, 2018-03-15:
I also have good news: the fake-KMS-mode seems to work fine now. The Pi is starting  as it should with it, resolution as set  (720p) and no bootproblem until now.

JamesH65, 2018-04-20:
@michaelsteffen1982 Am I to understand this issue can now be closed?

michaelsteffen1982, 2018-04-23:
@JamesH65 I think so, with FakeKMS-driver it starts fine now.

closed 2018-04-23 by JamesH65