raspberrypi/linux #2359: GRAPHICS_RGBA32 macro in vgfont.h wrong

RenateUSB, 2018-01-28:
In /opt/vc/src/hello_pi/libs/vgfont/vgfont.h:
```c
#define GRAPHICS_RGBA32( r, g, b, a ) GRAPHICS_RGBA888( r, g, b, a )
#define GRAPHICS_RGBA888( r, g, b, a ) ( (((a) << (8+8+8)) & ALPHA_888_MASK) | (((b) << (8+8)) & R_888_MASK) | (((g) << 8) & G_888_MASK) | ((r) & B_888_MASK) )
````
The first macro relies on the second, which has red and blue swapped incorrectly.
This is typographically and functionally wrong.

popcornmix, 2018-01-29:
We have:
```
#define R_888_MASK      (0x00FF0000)
#define G_888_MASK      (0x0000FF00)
#define B_888_MASK      (0x000000FF)
```
so the shifts tie up with the masks.
I suspect it is functionally correct, but there are multiple variants of what RGB vs BGR means and I suspect rather than creating a new set of mask macros the parameters were swapped.

Do you have an example of usage where the colours appears wrongly?

JamesH65, 2018-04-23:
@RenateUSB Any further comment? Any examples?

JamesH65, 2018-08-31:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-08-31 by JamesH65

MichaIng, 2020-05-28:
Issue first reported: https://github.com/raspberrypi/userland/issues/193
First PR to fix it, probably in wrong repo? https://github.com/raspberrypi/firmware/pull/985
Second attempt to fix it, hopefully in right repo: https://github.com/raspberrypi/userland/pull/628