raspberrypi/linux #2360: init_dma_coherent_pool_size reduced from 4Mb to 1Mb after kernel change 4.4 -> 4.9

maroviher, 2018-01-30:
I was wondering why my out of tree WiFi driver does not work anymore, after I moved from 4.4 to 4.9
Than I figured out that init_dma_coherent_pool_size in 4.4 is initialized with 4Mb:
https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/mach-bcm2709/bcm2709.c#L231

but in 4.9 branch it was dropped to 1Mb:
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/mach-bcm/board_bcm2835.c#L45

Is there a reason why it has been so greatly reduced?
Or should I create a merge request to make it 4Mb again?

pelwell, 2018-01-31:
The default size of the DMA coherent pool changed between 4.4 and 4.9 because we are now using the minimal upstream board support code. This is visible in /proc/cpuinfo which shows Hardware as being `BCM2835`. There are a few downstream patches to the upstream code but, as you've found, they specify SZ_1M whereas it used to be SZ_4M. The reasoning behind the change was that in 4.9 fewer allocations are done at GFP_ATOMIC, and 3M is a lot to set aside on a Pi Model B if it won't be used.

I said default because users who need need more (or less) coherent pool can use the `coherent_pool` kernel command line parameter to set a different value. To restore the old behaviour, add `coherent_pool=4M` to `/boot/cmdline.txt`.

maroviher, 2018-01-31:
Thank you for an explaination. I will use a setting in /boot/cmdline.txt for my driver.

pelwell, 2018-01-31:
Please close the issue if the cmdline setting works for you.

closed 2018-01-31 by maroviher