raspberrypi/linux #2366: Minimal kernel config created from bcm2709_defconfig using allnoconfig fails to compile

simonvanderveldt, 2018-02-04:
When trying to build a minimal kernel from `bcm2709_defconfig` using `allnoconfig` the resulting configuration doesn't compile. 

[`allnoconfig`](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/Documentation/kbuild/kconfig.txt#L75) is the normal way to configure/build the most minimal working kernel. It's used in combination with a defconfig to provide the required defaults.
It can be executed in the following way to generate a minimal config based on `bcm2709_defconfig`:
```
make ARCH=arm KCONFIG_ALLCONFIG=arch/arm/configs/bcm2709_defconfig allnoconfig
```
Now when compiling the generated `.config` I ran into several of these errors:
```
/tmp/ccholjfv.s: Assembler messages:
/tmp/ccholjfv.s:67: Error: selected processor does not support requested special purpose register -- `mrs r3,cpsr'
/tmp/ccholjfv.s:75: Error: selected processor does not support requested special purpose register -- `msr cpsr_c,r3'
```

Doing a `diffconfig` of the configs generated by `bcm2709_defconfig` and `allnoconfig` showed that `ARCH_BCM2835` was missing. Even though [`ARCH_BCM2835` is defined in `bcm2709_defconfig`](https://github.com/raspberrypi/linux/blob/a29fe9d2366b1700bf1015044e561108a5fa29df/arch/arm/configs/bcm2709_defconfig#L41) it doesn't actually get activated because it's prerequisites are not met.

I was able to fix these issues by adding:
```
CONFIG_MMU=y
CONFIG_ARCH_MULTI_V7=y
```
Shall I create a PR to add them? Or is there a specific reason these two config option aren't included in `bcm2709_defconfig`?

Once that was fixed I ran into the following issue, for which I've not yet figured out how to fix it. Any and all suggestions are welcome :)
```
drivers/mmc/host/bcm2835-sdhost.c: In function 'bcm2835_sdhost_dma_complete':
drivers/mmc/host/bcm2835-sdhost.c:541:10: error: implicit declaration of function 'kmap_atomic'; did you mean 'in_atomic'? [-Werror=implicit-function-declaration]
   page = kmap_atomic(host->drain_page);
          ^~~~~~~~~~~
          in_atomic
drivers/mmc/host/bcm2835-sdhost.c:541:8: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
   page = kmap_atomic(host->drain_page);
        ^
drivers/mmc/host/bcm2835-sdhost.c:552:3: error: implicit declaration of function 'kunmap_atomic'; did you mean 'in_atomic'? [-Werror=implicit-function-declaration]
   kunmap_atomic(page);
   ^~~~~~~~~~~~~
   in_atomic
```

lategoodbye, 2018-02-04:
The upstream sdhost driver does the following:

`#include <linux/highmem.h>`

simonvanderveldt, 2018-02-04:
> The upstream sdhost driver does the following:
> 
> #include <linux/highmem.h>

@lategoodbye Would that be the correct way to fix the issue?
Asking because if I enable `CONFIG_BLOCK` it works, so it might just be a missing dependency in `Kconfig`?

lategoodbye, 2018-02-04:
If the driver uses kmap_atomic it should include the proper header. Enabling CONFIG_BLOCK seems to hide the issue.

simonvanderveldt, 2018-02-04:
> If the driver uses kmap_atomic it should include the proper header. Enabling CONFIG_BLOCK seems to hide the issue.

OK, I'll try compiling without `CONFIG_BLOCK` and with the added header, will report back with the results.

simonvanderveldt, 2018-02-05:
> The upstream sdhost driver does the following:
> 
> #include <linux/highmem.h>

@lategoodbye This does indeed fix the issue. Is this something that should be fixed in this kernel tree or somewhere else?

pelwell, 2018-02-05:
It's a downstream driver so it must be fixed here. Leave it with me.

pelwell, 2018-02-05:
The sdhost driver change is in now.

defconfigs are defined to only store non-default settings. Because MULTI_V7=y is the default it doesn't end up in the defconfig when it is generated. There must something which is relying on MULTI_V7 that doesn't select it.

simonvanderveldt, 2018-02-05:
> The sdhost driver change is in now.

Very nice, thanks for the quick fix!

> defconfigs are defined to only store non-default settings. Because MULTI_V7=y is the default it doesn't end up in the defconfig when it is generated. There must something which is relying on MULTI_V7 that doesn't select it.

I'll have a look around and will try to figure out what's causing this exactly, will report back here with my findings.

pelwell, 2018-02-05:
Thanks - if there is a nice clean fix then we'll happily take it.

simonvanderveldt, 2018-02-05:
@pelwell What do you think about the undefined errors from `snd-soc-pisound`?

```
ERROR: "snd_rawmidi_set_ops" [sound/soc/bcm/snd-soc-pisound.ko] undefined!
ERROR: "snd_rawmidi_new" [sound/soc/bcm/snd-soc-pisound.ko] undefined!
ERROR: "snd_rawmidi_transmit_peek" [sound/soc/bcm/snd-soc-pisound.ko] undefined!
ERROR: "snd_rawmidi_transmit_ack" [sound/soc/bcm/snd-soc-pisound.ko] undefined!
ERROR: "snd_rawmidi_receive" [sound/soc/bcm/snd-soc-pisound.ko] undefined!
```
I got this working by enabling `CONFIG_SND_DRIVERS` which enables the required `CONFIG_SND_RAWMIDI` because then the following conditions are met:
```
|| SND_VIRMIDI [=m] && SOUND [=y] && !M68K && !UML && SND [=m] && SND_DRIVERS [=y] && SND_SEQUENCER [=m] ||
```

pelwell, 2018-02-05:
Where have you enabled CONFIG_SND_DRIVERS? It can't just be a simple addition to the defconfig because it would be lost when we next regenerate it.

lategoodbye, 2018-02-06:
I think the dependencies of SND_PISOUND is missing:
select SND_RAWMIDI

pelwell, 2018-02-06:
Yes, that's the conclusion I had come to, but what needs to change to solve the MULTI_V7 problem?

lategoodbye, 2018-02-06:
It would be helpful to know which source file generates this problem.

simonvanderveldt, 2018-02-06:
> I think the dependencies of SND_PISOUND is missing:
> select SND_RAWMIDI

You're right. Just tried adding this to `sound/soc/bcm/Kconfig` and it does indeed fix the problem.

simonvanderveldt, 2018-02-06:
> defconfigs are defined to only store non-default settings. Because MULTI_V7=y is the default it doesn't end up in the defconfig when it is generated. There must something which is relying on MULTI_V7 that doesn't select it.

> Where have you enabled CONFIG_SND_DRIVERS? It can't just be a simple addition to the defconfig because it would be lost when we next regenerate it.

@pelwell reading your comments again, it sounds like the defconfigs are generated automatically.
Is that correct? If that's the case, what is the starting point/how is determined which config options are added to the defconfig?

pelwell, 2018-02-06:
The process isn't really automatic, but there are some automated steps. Way back in the mists of time somebody generated a config file, probably fairly minimal, then used the `savedefconfig` feature to create a starting point for other configuration files. Over the years we've added and changed settings, usually in response to requests from users, but after each step we `savedefconfig`. Occasionally somebody submits a PR including a manual edit to the defconfigs that may include unnecessary or misplaced settings, but eventually there will be a `savedefconfig` and the world will be orderly again.

simonvanderveldt, 2018-02-06:
@pelwell Thanks for the clarification. I'll read up on what `savedefconfig` does exactly.

Regarding the missing `ARCH_MULTI_V7`: I looked at this some more and it's correct that it's missing. The reason for this is that `allnoconfig` sets every kernel config option to `no` (that's why it's called `allnoconfig` :P), including the ones that default to `yes` except the options that are explicitly required from the defconfig. So it expects the defconfig has explicitly defined the (minimal) amount of config options it needs.

It sounds like the current `bcm2709_defconfig` is not (entirely) made this way, and in some cases relies on some `default y` config options to create a working kernel that is configured with the minimal required options to compile/work. This might be a consequence of the way `savedefconfig` works?

FYI To get the config to compile after the `linux/highmem.h` include and depends on for `SND_RAWMIDI` were fixed the only two options I needed to explicitly set are
```
CONFIG_MMU=y
CONFIG_ARCH_MULTI_V7=y
```

pelwell, 2018-02-06:
> This might be a consequence of the way savedefconfig works?

Yes. As I hinted at above, defconfigs are the result of taking a .config file and removing all the settings that match the default values.

simonvanderveldt, 2018-02-19:
> Yes. As I hinted at above, defconfigs are the result of taking a .config file and removing all the settings that match the default values.

@pelwell First of all: Sorry for the delay.
Thanks for this description, this helped quiet a bit.

I have one follow-up question: we're working on a project based on CM3 and we'd very much like to build on an upstream minimal config. Since the current config files are all defconfig's this doesn't really work because we get a whole lot of config options enabled that we don't need/want, which means we need to maintain a stripped version of the defconfig where we remove quiet a lot of stuff that we're never going to use. This seems like a pretty common use case for Compute Module development to me.

Would you guys be open to the option of adding a so-called miniconfig (see https://landley.net/aboriginal/FAQ.html#dev_miniconfig) for CM3 to the tree to support this use case?
I've already done some work for this, wouldn't be a whole lot of effort to clean up the final parts.

JamesH65, 2018-04-23:
@pelwell Any comment on the `miniconfig` question above?

pelwell, 2018-04-23:
We'll consider all Pull Requests, but we're generally reluctant to take on maintenance of files - drivers, configurations etc. - that we don't use and therefore don't get tested. A clear statement of what constitutes a minimal configuration would help to persuade me, as would an offer of assistance when bringing up new kernels.

JamesH65, 2018-06-27:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

simonvanderveldt, 2018-06-27:
@JamesH65 can you answer this question (or direct someone who can to this issue)?

> I have one follow-up question: we're working on a project based on CM3 and we'd very much like to build on an upstream minimal config. Since the current config files are all defconfig's this doesn't [...]
I've already done some work for this, wouldn't be a whole lot of effort to clean up the final parts.

pelwell, 2018-06-27:
I already answered your question: https://github.com/raspberrypi/linux/issues/2366#issuecomment-383517619

simonvanderveldt, 2018-06-27:
@pelwell Sorry, missed that one (it was collapsed). Thanks for the answer.
I currently don't have enough time to work on this so closing this issue.

closed 2018-06-27 by simonvanderveldt