raspberrypi/linux #2367: Under-voltage messages in the kernel log?

notro, 2018-02-04:
I've been reminded that inadequate power supplies continous to be a problem for Raspberry Pi users (and even seasoned ones can be bitten by it).
This is especially true for those running headless as there is no blinking under-voltage icon on the screen to get your attention.

I've had this in the back of my mind for a couple of years that it would be nice to have a message in the kernel log when under-voltage occurs. So maybe now is the time to see if I can do something about this.

The first question is ofc, would this be helpful for more people than just myself?

The next question is how to get to this info from the kernel. Is there a mailbox property for this?
Either way I guess polling is the only option here.
On the Pi2 I believe it's possible to use the pwr led gpio as an interrupt to catch events, but on the Pi3 the  pwr led 'gpio chip' probably doesn't have interrupt capability that can be easily routed to the ARM.

It would also be nice to have a message in the log if the previous boot/run had an under-voltage event.
So if the watchdog rescues an under-voltage induced crash, under-voltage as one possible reason will be shown in the log after reset.

popcornmix, 2018-02-05:
``vcgencmd get_throttled`` has the information needed by the arm, but the gencmd/vchiq interface is quite heavyweight for a kernel driver. I could add a mailbox command to get the same information.

What sort of driver are you thinking of that could do the polling?

How are you thinking of storing the fact that previous run had an under-voltage condition?

The reset controller (RSTS) is the only state that survives a reboot, but we are already using the bits available there. sdcard is the other option, but we may not run long enough after the event to get the data safely onto the sdcard.

notro, 2018-02-05:
> What sort of driver are you thinking of that could do the polling?

I was hoping that the [power_supply class](https://www.kernel.org/doc/Documentation/power/power_supply_class.txt) could be useful for this, but it seems to be more or less dedicated to charging and maybe not a good candidate for detecting under-voltage.

Another possibility is to do a regulator driver for the 5v0 regulator. Regulators do have an event for under-voltage at least: [REGULATOR_EVENT_UNDER_VOLTAGE](http://elixir.free-electrons.com/linux/latest/ident/REGULATOR_EVENT_UNDER_VOLTAGE).

We can't just write a simple module that does this since we need a DT device to get to the firmware device for querying properties.

> How are you thinking of storing the fact that previous run had an under-voltage condition?

I suspected thet the RSTS was used up already so I hoped that maybe the firmware had access to some other register that survives reboot.

It is possible to preserve panic messages across reset as I found out: [Using ramoops/pstore to capture kernel panics](https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=199047)

Sooo, maybe this just as well can be written as a systemd service that polls the value and writes events to the kernel log, and then use ramoops to preserve the event across reboot.

Do you think under-voltage logging is useful and worth spending time on for the community?

popcornmix, 2018-02-05:
> Do you think under-voltage logging is useful and worth spending time on for the community?

Yes.

FishTest, 2018-02-08:
In most cases, the raspberry pi runs as a server and does not have a screen. There is no proprietary PMIC in the raspberry pi, and the Linux power driver can not communicate with the PMIC. So it's very important to get some power information in other ways.

popcornmix, 2018-02-08:
I've added a commit to firmware that supports tag ``GET_THROTTLED = 0x00030046``
```
/opt/vc/bin/vcmailbox 0x00030046 4 4 0
0x0000001c 0x80000000 0x00030046 0x00000004 0x80000004 0x00060006 0x00000000
```
The ``0x00060006`` is the throttled status as [described here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=147781&start=50#p972790).
Will be in next firmware build (possibly tomorrow).

notro, 2018-02-08:
I've been thinking about the challenge of polling and reliably getting info about anything but the first under-voltage event. We would have to poll very fast to be certain that we can detect an ongoing under-voltage.

How fast would we have to poll to be certain to detect under-voltage?

One solution is to have an under-voltage bit that is reset when read.
This way the polling period doesn't matter (that much), we are able to say that an event happened some time during the last *n* seconds.

popcornmix, 2018-02-08:
I believe this is already handled. We added this to ``vcgencmd get_throttled`` for [bcmstat](https://github.com/MilhouseVH/bcmstat/blob/master/bcmstat.sh#L264).

Basically the bottom 16 bits are the live values and top 16 bits are the sticky values.
If you pass in a parameter to get_throttled (either gencmd or mailbox) it will clear the sticky bits after returning them.

There is a slight caveat if multiple users are clearing bits, but the result isn't fatal and if a kernel driver is supporting this functionality then we can deprecate other users.

notro, 2018-02-08:
Indeed it does, I didn't come across that functionality in my web searches.
```
pi@pib:~ $ vcgencmd get_throttled 0x05
throttled=0x50000
pi@pib:~ $ vcgencmd get_throttled 0x05
throttled=0x0
```

popcornmix, 2018-02-09:
Latest rpi-update kernel includes the GET_THROTTLED mailbox message.

notro, 2018-02-14:
Even if I ask to clear the sticky bit, it is stuck for ~5 seconds:
```
$ while true; do vcgencmd get_throttled 1; sleep 1; done
throttled=0x40000
throttled=0x40000
throttled=0x40000
<brief overload>
throttled=0x50005
throttled=0x50005
throttled=0x50005
throttled=0x50000
throttled=0x40000
throttled=0x40000
throttled=0x40000
throttled=0x40000
^C
```
Is it a feature or a bug?

popcornmix, 2018-02-14:
The "live" values for get_throttled have a decay so you can catch brief events (corresponds with the length of time the warning symbol appears on display).
It's implemented as a uint32_t history that shifts and ors each cycle (100ms). So just over 3 seconds of decay.

popcornmix, 2018-02-14:
Can the kernel driver be the master who reads and clears sticky bits and all other users go through the kernel driver (e.g. kernel exports values to /sys or similar). That avoids other users clearing the sticky bits.

notro, 2018-02-14:
I was actually hoping to avoid a kernel driver. Some reasons:
- There's no clear subsystem to hook into for something like this
- We need to add a device in the Device Tree to use rpi_firmware_get(), what kind of device would that be? Or we could add this directly to drivers/firmware/raspberrypi.c
- It believe it will be difficult to get this upstream
- There are no in kernel users of get_throttled and it's possible to write to the kernel log from userspace: /dev/kmsg

What I need is a way to know is when the value changes. The current value is only interesting for monitoring purposes. So for a kernel solution that would probably mean adding /dev/get_throttled for userspace to poll for changes.

Dou you have any preferences regarding userspace or kernel?

rodizio1, 2018-02-16:
> The first question is ofc, would this be helpful for more people than just myself?

I'd love to see such a feature. I am the author of a distribution made for flying FPV aircraft with the Raspberry (https://github.com/bortek/EZ-WifiBroadcast) and users not providing the Raspberrys with stable 5Volts has been a constant issue over the past 1 1/2 years.

I have already tried running `vcgencmd get_throttled `in a loop every 10 seconds in a background task to notifiy the user with big letters in case of under-voltage (the "lightning" symbol always gets ignored, there are plenty of youtube videos from people using my distribution with the symbol popping up in the upper right corner), however, the Pi always crashed after some time when doing that.

Would integrating it into the sys filesystem also work?

I had the same problem with` vcgencmd get_temp`, but since the temperature also got included in the /sys/... directory some time ago, I can read it every second and display it to the user without the Pi crashing, would be nice if I could also do that for the undervoltage.

notro, 2018-02-17:
I have stalled with my userspace approach and since @popcornmix have put the firmware bits in place I've tried a kernel approach.

sysfs file description:
```
What:		/sys/devices/platform/soc/soc:firmware/get_throttled
Description:	This file returns the firmware get_throttled value which is 
		divided into 2 parts:
		- lower 16-bits are the live value
		- upper 16-bits are the history or sticky value

		Bits:
		0: under-voltage
		1: arm frequency capped
		2: currently throttled 
		16: under-voltage has occurred
		17: arm frequency capped has occurred
		18: throttling has occurred

		This file can be polled for changes.

What:		/sys/devices/platform/soc/soc:firmware/get_throttled_mask
Description:	This file does two things:
		1. Describes which bits should denote a change when polling
		2. The upper part is used when clearing the sticky bits

		Default value is 1 << 16.

What:		/sys/devices/platform/soc/soc:firmware/get_throttled_poll_ms
Description:	Specifies the get_throttled polling interval in milliseconds 
		(default 2000ms). The kernel polls this value to detect 
		under-voltage and to support polling the get_throttled file for 
		changes. Setting this to zero disables polling.

```

Code:
```diff
diff --git a/drivers/firmware/raspberrypi.c b/drivers/firmware/raspberrypi.c
index f261b64d1657..2e930797dd3c 100644
--- a/drivers/firmware/raspberrypi.c
+++ b/drivers/firmware/raspberrypi.c
@@ -1,3 +1,4 @@
+#define DEBUG
 /*
  * Defines interfaces for interacting wtih the Raspberry Pi firmware's
  * property channel.
@@ -14,6 +15,7 @@
 #include <linux/module.h>
 #include <linux/of_platform.h>
 #include <linux/platform_device.h>
+#include <linux/workqueue.h>
 #include <soc/bcm2835/raspberrypi-firmware.h>
 
 #define MBOX_MSG(chan, data28)		(((data28) & ~0xf) | ((chan) & 0xf))
@@ -21,11 +23,14 @@
 #define MBOX_DATA28(msg)		((msg) & ~0xf)
 #define MBOX_CHAN_PROPERTY		8
 
+#define UNDERVOLTAGE_STICKY_BIT		BIT(16)
+
 struct rpi_firmware {
 	struct mbox_client cl;
 	struct mbox_chan *chan; /* The property channel. */
 	struct completion c;
 	u32 enabled;
+	struct delayed_work get_throttled_poll_work;
 };
 
 static struct platform_device *g_pdev;
@@ -166,6 +171,162 @@ int rpi_firmware_property(struct rpi_firmware *fw,
[... 257 lines cut ...]
    epoll.poll()
    file.seek(0)
    status = file.read()
    get_throttled = int(status, 16)

    if get_throttled & (1 << 16):
        try:
            nid = iface.Notify('get_throttled', 0, 'dialog-warning', 'Under-voltage!', msg, [], {}, 0)
        except dbus.exceptions.DBusException as e:
            sys.stderr.write("Error sending notification: %s\n" % e); sys.stderr.flush()
    elif nid:
        try: # The user might have closed it
            iface.CloseNotification(nid)
        except dbus.exceptions.DBusException:
            pass
        nid = None

epoll.unregister(file.fileno())
file.close()

```

Comments anyone?

I haven't got time to try and get this upstream.

popcornmix, 2018-02-17:
Looks good to me.
I assume the idea for a regulator driver didn't work out?
@pelwell ?

notro, 2018-02-17:
> I assume the idea for a regulator driver didn't work out?

It's just that I didn't find anything in the regulator framework that helps out with this. The only thing is an under-voltage event, but this in only for in-kernel users. If we could at least have read the actual voltage, then it would make more sense to make a regulator driver.

The benefit of having it in the firmware driver is that we can deal with all the bits, not only under-voltage.

pelwell, 2018-02-19:
The driver looks good. You could restrict the critical error message to only be generated on the rising edge of the under-voltage signal, but if you are under-voltage and over-temperature then you have bigger problems than an extra kernel message.

notro, 2018-02-19:
> You could restrict the critical error message to only be generated on the rising edge of the under-voltage signal

That's what I'm currently doing. I'm only logging voltage events. Under-voltage is critical and returning to normal is info.
Should the other bits be logged to the kernel log as well? I'm not familiar with the meaning of these bits and what they signify. Do they have anything to do with temperature?
How can I trigger throttled and capped?

Currently the get_throttled file is dependent on the device name, which depends on the DT node pathname:
```/sys/devices/platform/soc/soc:firmware/get_throttled```

Perhaps it would be better to put the files under the driver instead, since it less likely to change?
```/sys/bus/platform/drivers/raspberrypi-firmware/get_throttled```

pelwell, 2018-02-19:
Oh I see - in that case the mask in the test is redundant and misleading, surely.

notro, 2018-02-19:
> Oh I see - in that case the mask in the test is redundant and misleading, surely.

There are 2 masks:
- rpi_firmware_get_throttled_mask is used to clear sticky bits and to tell which bits we notify sysfs/userspace about when the value changes. If userspace doesn't care about the throttled bit, it won't be told about the change when polling (select) the file.
- UNDERVOLTAGE_STICKY_BIT used to log voltage only messages

Maybe we should drop rpi_firmware_get_throttled_mask entirely and just clear all sticky bits and let userspace keep track of the bits it cares about.

I included a way to set the polling period and even stop it, but I'm not sure if it will be useful.

I could strip this down to only the get_throttled sysfs file and we can add functionality when/if it's needed. In spirit with the KISS principle. Probably best to just scratch my own itch instead of guessing where other people might itch :-)

pelwell, 2018-02-19:
Look at the multiple uses of UNDERVOLTAGE_STICKY_BIT - I think at least one is redundant:
```
+	new_uv = *value & UNDERVOLTAGE_STICKY_BIT;
+	old_uv = old_value & UNDERVOLTAGE_STICKY_BIT;
...
+	if (new_uv != old_uv) {
+		if (new_uv & UNDERVOLTAGE_STICKY_BIT)
+			pr_crit("Under-voltage detected!\n");
+		else
+			pr_info("Voltage normalised\n");
+	}

notro, 2018-02-19:
Ah, I see.

Cyr-ius, 2018-03-17:
I compiled the last kernel 4.14.27 and since then I have the following message in loop in the kern.log
raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x80000001
and i have no problem of under voltage
Pi Zero W
vcgencmd measure_clock arm
frequency(45)=1000106000
vcgencmd measure_clock core
frequency(1)=400000000
vcgencmd measure_volts
volt=1.2000V

vcgencmd measure_temp; vcgencmd get_throttled; vcgencmd measure_temp
temp=37.9'C
throttled=0x0
temp=37.9'C

I have the problem on my Pi Zero and Pi 2

How to stop this message ?

pelwell, 2018-03-17:
Update to a recent firmware - copy it from the raspberrypi/firmware repo or use `sudo rpi-update`.

danifr, 2018-03-26:
Yesterday I upgraded my Pi2 to 4.14.29-1-ARCH.  After the upgrade I noticed I started to get a bunch (dozens) of these messages:

```
Mar 26 20:57:50 rbpi2 kernel: Under-voltage detected! (0x00050005)
Mar 26 20:58:25 rbpi2 kernel: Voltage normalised (0x00000000)
```

Is there a way I can simply hide them? I'm running Kodi on that Pi and the messages are printed on top of Kodi's UI.

Thanks for your help

pelwell, 2018-03-26:
Use a better PSU, or get KODI not to scribble over the GUI?

closed 2018-03-27 by notro

ThomasKaiser, 2018-04-04:
> The 0x00060006 is the throttled status as [described here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=147781&start=50#p972790).

@popcornmix I think the bits for throttling and frequency capping are reversed. At least that's what testing suggests. On a NAS distribution I use the sticky bits to be logged at shutdown and to generate a `motd` notification when user logs in next time: https://forum.openmediavault.org/index.php/Thread/18991-New-approach-for-Raspberry-Pi-OMV-images/?postID=148934#post148934

E3V3A, 2018-04-16:
Seem related #2512.

lategoodbye, 2018-05-16:
I assume this under-voltage detection isn't available for all Raspberry Pis. Which one supports it?

pelwell, 2018-05-16:
All boards since the B+ (so A+, 2B, 3B, 3B+) except Pi Zero (including Zero W) and Compute Modules.

lategoodbye, 2018-05-16:
Thanks. So the responsible hardware part is the SMPS?

pelwell, 2018-05-16:
No, only on the 3B+. For other devices there is a cunning analogue circuit hooked up to a regular GPIO input.

jcea, 2020-08-11:
I am kernel "4.19.122-1-osmc" in a Raspberry Pi 3B+ Rev 1.3. The file "/sys/devices/platform/soc/soc:firmware/get_throttled" exists, but it always returns cero:

```
root@osmcpi:~# vcgencmd get_throttled
throttled=0x80008
root@osmcpi:~# cat "/sys/devices/platform/soc/soc:firmware/get_throttled" 
0
```

More details in https://discourse.osmc.tv/t/sys-devices-platform-soc-soc-firmware-get-throttled-not-working/86639