raspberrypi/linux #2368: no snd-usb-hiface.ko module under Raspbian

calabuig, 2018-02-05:
I would like to use m2tech Hiface Evo with my Raspberry pi 3, but under Raspbian I can't find the snd-usb-hiface.ko module.
Is there any way to make it work?

pelwell, 2018-02-05:
The configurations are missing CONFIG_SND_USB_HIFACE=m, the end result of which is a 22KB module. I've applied this setting to the standard defconfigs, and the next firmware release should contain the module.

calabuig, 2018-02-05:
Ok, Thank you.
When the new firmare could be available?

pelwell, 2018-02-05:
Probably later this week, but you can grab a build that should be compatible with the current `rpi-update` release [here](https://drive.google.com/file/d/1N6TrH7kEZNxnIOSFYThhbG8OqWX9T2iO/view?usp=sharing). Copy it to `/lib/modules/4.9.79-v7+/kernel/sound/usb/hiface/snd-usb-hiface.ko` (you will need to create the `hiface` directory).

calabuig, 2018-02-06:
Hello Phil,
thanks for your help. The module works perfectly and the system detect my card.
I copied the module to /lib/modules/4.9.79-v7+/kernel/sound/usb/hiface/ and I gave the command:
insmod snd-usb-hiface.ko.
But after the reboot, the system doesn't load it automatically.
So I think I have to add the module anywhere else...
Could you help one (last) more time, please?

pelwell, 2018-02-06:
Ah yes, the module needs to be added to the kernel's database of modules and their aliases - just run `sudo depmod`.

calabuig, 2018-02-06:
Do I have to run <sudo depmode> under the folder /lib/modules/4.9.79-v7+/kernel/sound/usb/hiface/ ?
Is it the same procedure under Ubuntu?

calabuig, 2018-02-06:
Do I have to run sudo depmode under the folder /lib/modules/4.9.79-v7+/kernel/sound/usb/hiface/ ?
 Is it the same procedure under Ubuntu?

pelwell, 2018-02-06:
No - provided you are running 4.9.79-v7+ then you can run it anywhere - the kernel and the associated utilities know where to find the modules.

calabuig, 2018-02-06:
Ok, thank you very very much for your help.
I'll give it a try this evening!!!

JamesH65, 2018-04-23:
@calabuig @pelwell Reading through, I think this issue can be closed?

pelwell, 2018-04-23:
That's up to @calabuig, but we can close if we get no response in a week.

JamesH65, 2018-06-27:
Closing this issue as questions answered/issue resolved.
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-06-27 by JamesH65