raspberrypi/linux #2371: macsec: hw csum failure

beaglemaster, 2018-02-07:
I am trying to implement macsec on a rp2.  Even though it is working, it is hindered by the csum error that seems to have existed for quite some time.  I have tried to find a work around but have not been able to find one - any advice would be greatly appreciated.  

Macsec was not in the default kernel build so I had to build the kernel.  I tried this on both Centos7 and Raspbian and since I had to build the kernel for both OS's, the result was the same.   In addition, I tried two different kernels hoping that the error would have been fixed but I got the same result with version 4.9.79 and 4.14.16.  I may have missed something in the kernel build but if I did, I don't know what I'm missing.  The error from the 4.14.16 kernel build is as follows:

[   42.175018] MACsec IEEE 802.1AE
[   42.274802] device eth0 entered promiscuous mode
[   54.616840] macsec0: hw csum failure
[   54.620595] CPU: 0 PID: 30 Comm: kworker/0:1 Not tainted 4.14.16-v7 #2
[   54.627305] Hardware name: BCM2835
[   54.630829] Workqueue: crypto cryptd_queue_worker [cryptd]
[   54.636496] [<80017fe4>] (unwind_backtrace) from [<80014278>] (show_stack+0x20/0x24)
[   54.644460] [<80014278>] (show_stack) from [<806291cc>] (dump_stack+0xd4/0x118)
[   54.651984] [<806291cc>] (dump_stack) from [<8053276c>] (netdev_rx_csum_fault+0x44/0x48)
[   54.660310] [<8053276c>] (netdev_rx_csum_fault) from [<805250e0>] (__skb_checksum_complete+0xb4/0xb8)
[   54.669791] [<805250e0>] (__skb_checksum_complete) from [<805b57ec>] (icmp_rcv+0xd0/0x388)
[   54.678290] [<805b57ec>] (icmp_rcv) from [<8057bba4>] (ip_local_deliver_finish+0xe4/0x330)
[   54.686784] [<8057bba4>] (ip_local_deliver_finish) from [<8057c410>] (ip_local_deliver+0x54/0xdc)
[   54.695902] [<8057c410>] (ip_local_deliver) from [<8057c02c>] (ip_rcv_finish+0x23c/0x494)
[   54.704306] [<8057c02c>] (ip_rcv_finish) from [<8057c7b4>] (ip_rcv+0x31c/0x554)
[   54.711820] [<8057c7b4>] (ip_rcv) from [<8052fef4>] (__netif_receive_skb_core+0x340/0xc84)
[   54.720318] [<8052fef4>] (__netif_receive_skb_core) from [<805328e0>] (__netif_receive_skb+0x20/0x7c)
[   54.729793] [<805328e0>] (__netif_receive_skb) from [<80536594>] (netif_receive_skb_internal+0x30/0xe0)
[   54.739445] [<80536594>] (netif_receive_skb_internal) from [<805373a8>] (napi_gro_receive+0xf4/0x120)
[   54.748920] [<805373a8>] (napi_gro_receive) from [<805623c8>] (gro_cell_poll+0x7c/0xac)
[   54.757148] [<805623c8>] (gro_cell_poll) from [<80536ca8>] (net_rx_action+0x2e8/0x45c)
[   54.765282] [<80536ca8>] (net_rx_action) from [<80009694>] (__do_softirq+0x18c/0x3d8)
[   54.773331] [<80009694>] (__do_softirq) from [<8002adb0>] (do_softirq+0xac/0xb0)
[   54.780933] [<8002adb0>] (do_softirq) from [<8002aebc>] (__local_bh_enable_ip+0x108/0x118)
[   54.789439] [<8002aebc>] (__local_bh_enable_ip) from [<7f0dc728>] (cryptd_skcipher_complete+0x4c/0x9c [cryptd])
[   54.799837] [<7f0dc728>] (cryptd_skcipher_complete [cryptd]) from [<7f0dd010>] (cryptd_skcipher_decrypt+0xb8/0xc0 [cryptd])
[   54.811296] [<7f0dd010>] (cryptd_skcipher_decrypt [cryptd]) from [<7f0dd440>] (cryptd_queue_worker+0x7c/0xac [cryptd])
[   54.822298] [<7f0dd440>] (cryptd_queue_worker [cryptd]) from [<8003ea58>] (process_one_work+0x158/0x454)
[   54.832031] [<8003ea58>] (process_one_work) from [<8003edb8>] (worker_thread+0x64/0x5b8)
[   54.840348] [<8003edb8>] (worker_thread) from [<80044e28>] (kthread+0x13c/0x16c)
[   54.847952] [<80044e28>] (kthread) from [<8001012c>] (ret_from_fork+0x14/0x28)

I set up the macsec by doing the following:

/usr/sbin/ip link add link eth0 macsec0 type macsec encrypt on
/usr/sbin/ip macsec add macsec0 tx sa 0 pn 1 on key 01 319ab1a97f41cc36b7dae5bbe47b7a00
/usr/sbin/ip macsec add macsec0 rx address b8:27:eb:60:61:a7 port 1
/usr/sbin/ip macsec add macsec0 rx address b8:27:eb:60:61:a7 port 1 sa 0 pn 1 on key 02 4141c0643af84925e178f1b6e7da4b84
/usr/sbin/ip link set macsec0 up
/usr/sbin/ip addr add 10.10.10.9/24 dev macsec0

and then I do similar commands for the other end of the macsec.

Thanks for any help or suggestions.

beaglemaster, 2018-02-12:
By following other examples, I was able to modify the driver of the network adapter.  Everything works as it should now.

closed 2018-02-12 by beaglemaster

pelwell, 2018-02-12:
Can you post your patch here for others to see? Ideally, submit it upstream.

beaglemaster, 2018-02-12:
Yes i will submit it.  Even though the fix I made today seems to work
perfectly.  There are a few checks that I need to make and some testing
before I submit it.

Dutchdoc, 2018-03-28:
Any news on that driver patch? Could you share it here? ;-)

tomkcook, 2021-04-19:
Also very interested in this patch.  I'm seeing this issue in 4.14.77 (which I'm stuck with).

tomkcook, 2021-04-20:
I've tried adding commits d55bef5 and 88078d9 to my kernel but still seeing the same checksum failure stack traces.

tomkcook, 2021-04-22:
For anyone else who runs into this:

```
Index: drivers/net/macsec.c
===================================================================
--- a/drivers/net/macsec.c
+++ b/drivers/net/macsec.c
@@ -865,6 +865,7 @@ static void macsec_reset_skb(struct sk_b
 
 static void macsec_finalize_skb(struct sk_buff *skb, u8 icv_len, u8 hdr_len)
 {
+        skb->ip_summed = CHECKSUM_NONE;
 	memmove(skb->data + hdr_len, skb->data, 2 * ETH_ALEN);
 	skb_pull(skb, hdr_len);
 	pskb_trim_unique(skb, skb->len - icv_len);
```

Despite first appearing in linux 4.5, this has yet to be merged onto the rpi-4.14.y branch (though it is on the rpi-5.10.y branch).

6by9, 2021-04-22:
That appears to have come from https://github.com/torvalds/linux/commit/7d8b16b9facb0dd81d1469808dd9a575fa1d525a, merged in 5.3.

It gets backported to the mainline stable/LTS branches based on any Fixes: tag or other report to stable.
Indeed it is in the stable 4.14 branch with https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/net/macsec.c?h=linux-4.14.y&id=91b3202697b6b6ac65c42f10d01aa42edfd739b1

Raspberry Pi moved our stable branch on from rpi-4.14.y after
commit fa5941f45d7ed070118b7c209b7f2c3a034293bd
Author: Greg Kroah-Hartman <gregkh@linuxfoundation.org>
Date:   Sat Apr 27 09:35:42 2019 +0200

    Linux 4.14.114

That being before the fix being backported on 2019-07-31 means it won't be present. Sorry, Raspberry Pi don't offer support on a kernel once a new LTS kernel has been judged to be stable on the platform.

tomkcook, 2021-04-22:
Yep, understood.  Just wanted to leave a fix for anyone else who came across the problem.