raspberrypi/linux #2375: stretch dwc2 does not load on cm3

salsanci, 2018-02-09:
/boot/config.txt added:
dtoverlay=dwc2

On CM3
lsmod - dwc2 does not show up

Same image on a Pi Zero
lsmod - dwc2 shows up, works as expected

lategoodbye, 2018-02-09:
Hi salsanci,
which kernel do you use? Do you see any message about dwc2 in dmesg on CM3?

salsanci, 2018-02-09:
pi@raspberrypi:~$ dmesg | grep dw
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:25:1E:C8 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=7e070012-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.313241] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    0.315676] dwc_otg: FIQ enabled
[    0.315680] dwc_otg: NAK holdoff enabled
[    0.315684] dwc_otg: FIQ split-transaction FSM enabled
[    0.315696] Module dwc_common_port init
pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
pi@raspberrypi:~$ dmesg | grep dw
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:25:1E:C8 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=7e070012-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.313241] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    0.315676] dwc_otg: FIQ enabled
[    0.315680] dwc_otg: NAK holdoff enabled
[    0.315684] dwc_otg: FIQ split-transaction FSM enabled
[    0.315696] Module dwc_common_port init
pi@raspberrypi:~$ tail /boot/config.txt
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Added
dtoverlay=dwc2,dr_mode=peripheral

pi@raspberrypi:~$ 
(no difference if I just use dtoverlay=dwc2)

pelwell, 2018-02-19:
As mentioned in a near-duplicate thread (https://github.com/raspberrypi/linux/issues/2385), the dwc2 module is not built for BCM2836- and BCM2837-based devices on 4.9 kernels. However, it is included with all Pi 4.14 kernels.

We are unlikely to release another 4.9 build unless a significant problem is found with the 4.14 branch. Can you run `sudo rpi-update` and confirm that the dwc2 overlay works for you?

salsanci, 2018-02-20:
Gets further
````

pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.14.18-v7+ #1093 SMP Fri Feb 9 15:33:07 GMT 2018 armv7l GNU/Linux
pi@raspberrypi:~$ dmesg | grep dw
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:25:1E:C8 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=307da879-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait  modules-load=dwc2,g_ether
[    0.236062] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    0.238492] dwc_otg: FIQ enabled
[    0.238497] dwc_otg: NAK holdoff enabled
[    0.238501] dwc_otg: FIQ split-transaction FSM enabled
[    0.238510] Module dwc_common_port init
[    2.883899] dwc2 3f980000.usb: 3f980000.usb supply vusb_d not found, using dummy regulator
[    2.884002] dwc2 3f980000.usb: 3f980000.usb supply vusb_a not found, using dummy regulator
[    3.341510] dwc2 3f980000.usb: dwc2_check_param_tx_fifo_sizes: Invalid parameter g_tx_fifo_size[7]=0
[    3.356059] dwc2 3f980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[    3.372742] dwc2 3f980000.usb: DWC OTG Controller
[    3.372818] dwc2 3f980000.usb: new USB bus registered, assigned bus number 1
[    3.372886] dwc2 3f980000.usb: irq 39, io mem 0x3f980000
[    3.373224] usb usb1: Manufacturer: Linux 4.14.18-v7+ dwc2_hsotg
pi@raspberrypi:~$ tail /boot/config.txt
# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

dtoverlay=dwc2

````

JamesH65, 2018-04-23:
@salsanci Is this issue still present with the latest kernel release?

JamesH65, 2018-05-11:
@ Will soon be closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

salsanci, 2018-06-12:
Tested on latest kernel release with CM3, functions as expected. No issues.

closed 2018-06-12 by salsanci