raspberrypi/linux #2376: Kernel 4.14 problems

mypiandrew, 2018-02-12:
Hi

We're running a CM3 on a custom carrier board we've had for a few years now.

Has been running quite happily up to 4.9.75, no problems. All Solid.

I've just done a rpi-update running on Jessie to test out the new 4.14 kernel and the system hangs on next boot up :

https://gist.github.com/anonymous/55a8b ... b3f61db1c1

These look more interesting from a "going bang" perspective :

https://gist.github.com/anonymous/16b43 ... 046169804a
https://gist.github.com/anonymous/b36a2 ... ffc0584462

What can I do to assist with debugging this?

Andrew.

mypiandrew, 2018-02-12:
Adding these to the cmdline.txt file doesn't help 

dwc_otg.lpm_enable=0 dwc_otg.speed=1 

I did have other issues with the system way back on 4.4 with #1633 but I doesn't look the same to me as it's not even getting as far as systemd by the looks of it here

6by9, 2018-02-12:
(Duplicate of https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689&start=25#p1272277)

Full contents of config.txt please, and details of all peripherals connected to the device.
Booting off EMMC?
What's in /etc/fstab too? You appear to have an SD card exporting /dev/sda, so is that being mounted correctly?

6by9, 2018-02-12:
/etc/modules too for any extra modules you're loading.

mypiandrew, 2018-02-12:
Yes it's a CM3 

I was just taking a cut and paste of the config.txt line and noticed and old overlay

I had this line still in ( I can't remember why off hand, wasn't there an issue with this at one point?)
dtoverlay=mmc

Removed it, and it's booting nicely now 

Would that have caused other issues?

6by9, 2018-02-12:
> Would that have caused other issues?

Without details of your custom system, it could be anything.

mypiandrew, 2018-02-12:
Hmm.. on removing the forced usb slowdown mods to cmdline.txt I am still getting a USB error that sounds awfully like what I saw before 

Let me gather some info for you

mypiandrew, 2018-02-12:
root@raspberrypi:~# lsusb
Bus 001 Device 007: ID 1e0e:9001 Qualcomm / Option
Bus 001 Device 006: ID 0424:2240 Standard Microsystems Corp.
Bus 001 Device 004: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 005: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 003: ID 0424:9512 Standard Microsystems Corp. LAN9500 Ethernet 10/100 Adapter / SMSC9512/9514 Hub
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. USB-2.0 4-Port HUB
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

mypiandrew, 2018-02-12:
root@raspberrypi:~# grep -v \# /boot/config.txt
dtdebug=on
dtoverlay=uart0,txd0_pin=32,rxd0_pin=33
dtoverlay=uart1,txd1_pin=14,rxd1_pin=15
dtoverlay=uart_swap
enable_uart=1
dtparam=i2c1=on
dtparam=i2c1=on,i2c1_baudrate=180000
dtoverlay=i2c-rtc,ds1307
core_freq=250

mypiandrew, 2018-02-12:
root@raspberrypi:~# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 3: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 4: Dev 7, If 1, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 4: Dev 7, If 6, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 4: Dev 7, If 4, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 4: Dev 7, If 2, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 4: Dev 7, If 0, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 4: Dev 7, If 5, Class=Vendor Specific Class, Driver=qmi_wwan, 480M
        |__ Port 4: Dev 7, If 3, Class=Vendor Specific Class, Driver=option, 480M

Wired : 

GL850G 
	> LAN9512 
	> FT230X 
	> USB2240
          >MPCIE (SIM7100)

mypiandrew, 2018-02-12:
This shows the system with a boot that has the mysterious error 71 on it 

https://gist.github.com/mypiandrew/5c4ab7e6f915319179ae3af6e860e950

mypiandrew, 2018-02-12:
OK scratch the -71 error, I think there must have been a intermittent contact on the mPCIe card I was using to test. Given the edge contacts a clean and screwed the card in fully and it's booting AOK for the last 10 tries in a row.

Nice work, and thanks for the hand holding.

closed 2018-02-12 by mypiandrew