raspberrypi/linux #2377: USB Boot hangs if second mass storage USB device attached

swhobbit, 2018-02-12:
System is Raspberry Pi 3, running stretch with recent patches applied.

* USB devices:
  * a single SSD 128G drive (SanDisk SD8SNAT128G1122)
  * KVM (externally powered) 
  * Logitech GamePad
* Non-USB devices 
  * 2.4 Amp power full supply
 * 10/100/1000 Mhz Ethernet
 * HDMI to Dell U23412M monitor
 * Analog audio 

***Problem:***

While system normally boots fine from SSD if no other mass storage is attached, if additional an empty USB device such a thumb drive is attached the boot hangs.

***To Reproduce:***

* Create USB Raspberry Pi boot drive with jessie or stretch.  Boot successfully.
* Add second empty USB thumb drive.  Boot hangs.

***Expected Behavior:***

Second non-bootable thumb drive does not affect boot. System boots successfully.

**System configuration**

```
[diana:0,~,12:26:30,202] uname -a
Linux diana 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

[diana:0,~,12:26:33,203] cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

[diana:0,~,12:26:48,204] lsblk -f
NAME   FSTYPE LABEL        UUID                                 MOUNTPOINT
sda                                                             
├─sda1 vfat   DIANA_BOOT   547C-C484                            /media/usb3
├─sda2 vfat   DIANA_SPARE  5511-2102                            /media/usb0
├─sda3                                                          
├─sda5 ext4   diana_root   c7b414b7-cdf7-4610-ba80-938c31a15d7b /media/usb4
├─sda6 ext4   diana_home   13a6594f-eeba-41ac-b5fd-72bd7f62a319 /media/usb2
├─sda7 ext4   diana_export b3bcb416-6dab-4e82-8d11-b955de658d92 /media/usb1
└─sda8 swap   diana_swap   d06a9985-012c-4186-b931-fb73ccc38aed [SWAP]

[diana:0,~,12:26:52,205] lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 4: Dev 4, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 4: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 4, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 5: Dev 5, If 0, Class=Mass Storage, Driver=usb-storage, 480M
[diana:0,~,12:27:02,206] 
```

JamesH65, 2018-04-23:
@ghollingworth Any comment
@swhobbit Is this issue still present on the latest releases? There have been changes to the bootcode recently.

ghollingworth, 2018-04-23:
On my list of things...

swhobbit, 2018-04-25:
Still dies,  Says something like cannot mount from unknown device.  I 
can get a console panic (I have a lovely serial to USB adapter from the 
last time I was getting panics at boot), but not today.  I'm off to 
rescue a machine which is literally a museum piece ...

swhobbit, 2018-05-04:
Ok.  I retested.  Thumbs down.

Brand new Raspberry Pi 3B+, connected to a SanDisk SD8SNAT128G1122 [128 GB]
SSD. Software image is 2018-04-18-raspbian-stretch-lite.zip

That was installed on an SD Card and then transferred to the SSD. The 3B+
and drive boot fine together. I've been running the hardware twin to the
drive off a older Raspbberry Pi 3 for months.

Both old and new boots & runs fine.  That is, unless I put a Samsung 32G
USB Thumbdrive (straight out of the packaging with two Samsung files still
on it) in a slot and boot.  Then I don't even get a rainbow screen.

Anything else I can provide?

-ahd-

ghollingworth, 2018-06-27:
Can you put in a bootcode.bin from:

https://github.com/raspberrypi/firmware/raw/master/boot/bootcode.bin

Onto an SD card with nothing else on the card, then do:

sed -i -e "s/BOOT_UART=0/BOOT_UART=1/" bootcode.bin

Then attach a serial device to the Pi with 115200 baud

Should give us some information...

Gordon

swhobbit, 2018-06-27:
On 6/27/18 5:11 AM, Gordon Hollingworth wrote:
> Can you put in a bootcode.bin from:
>
> https://github.com/raspberrypi/firmware/raw/master/boot/bootcode.bin
>
> Onto an SD card with nothing else on the card, then do:
>
> sed -i -e [...]
Yes, it may be the weekend before I get to it.

I assume you mean a serial console, and to reboot after I do this. 
(Fortunately, I do own a serial console adapter; I just gotta go look up 
the pins ...).

I suppose I need not point out that's running sed on a binary file; I 
haven't used a text editor on a binary file since 1980.  :-)

-ahd-

pelwell, 2018-06-27:
> I suppose I need not point out that's running sed on a binary file;

Correct - we are already sufficiently disgusted with ourselves.

ghollingworth, 2018-07-01:
Me not so much. ;)

swhobbit, 2018-07-07:
Sorry for the delay, I was doing CPR on an IBM 4361 mainframe.

@ghollingworth Well, `sed` on MacOS was pretty disgusted with the idea, and refused to do it.   Can you blame it?  :-)

I moved the sd card over to the target Raspberry PI 3+ and sed was more wanton.

I booted successfully with the card in, added a thumb drive, booted (it panicked).  Not shown is successful boot with card after thumb drive removed.  Monolithic session log is attached.

Contents of thumb drive (vfat) was only classic Mac cruft:
```
[kendra:ttys001,/Volumes/NO NAME,03:10:02,503] ls -ltra
total 128
drwxrwxrwx@ 1 ahd   staff  16384 Dec 31  1979 .
drwxrwxrwx  1 ahd   staff  16384 Apr  9 18:39 .Spotlight-V100
drwxrwxrwx  1 ahd   staff  16384 Jul  7 14:52 .Trashes
drwxrwxrwx  1 ahd   staff  16384 Jul  7 15:09 .fseventsd
drwxr-xr-x@ 6 root  wheel    204 Jul  7 15:09 ..
```
[session.log](https://github.com/raspberrypi/linux/files/2173131/session.log)

swhobbit, 2018-07-07:
Addition configuration details:

cat /etc/fstab 
```
proc            /proc           proc    defaults          0       0
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
LABEL=BOOT      /boot vfat    defaults          0       2
LABEL=root      /       ext4    defaults,nofail   0       2
LABEL=home      /home ext4    defaults,nofail   0       2
LABEL=export    /export ext4    defaults,nofail   0       2
LABEL=swap      none    swap    defaults,nofail   0       0
tmpfs           /tmp    tmpfs   nodev,nosuid,size=250M  0       0
```

And note the thumb driver has no label:
[speedy-gonzales:ttyS0,~,03:34:02,29] lsblk -fs
```
NAME      FSTYPE LABEL       UUID                                 MOUNTPOINT
sda1      vfat   BOOT        3F3B-D532                            /boot
└─sda                                                             
sda2      vfat   DIANA_SPARE 5511-2102                            
└─sda                                                             
sda3                                                              
└─sda                                                             
sda5      ext4   root        035af903-48d7-461c-90b3-d6ad3c23edfe /
└─sda                                                             
sda6      ext4   home        a730fd4e-4633-4725-8dbe-a90bcfa5ccc1 /home
└─sda                                                             
sda7      ext4   export      b3bcb416-6dab-4e82-8d11-b955de658d92 /export
└─sda                                                             
sda8      swap   swap        d06a9985-012c-4186-b931-fb73ccc38aed [SWAP]
└─sda                                                             
sdb1      vfat               A831-68CE                            
└─sdb                                                             
mmcblk0p1 vfat   BOOT        59ED-B18F                            
└─mmcblk0                                                         
[speedy-gonzales:ttyS0,~,03:34:06,30] 
```
And lastly:
```
[speedy-gonzales:ttyS0,~,03:36:41,30] cat /boot/cmdline.txt 
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/sda5 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```