raspberrypi/linux #2384: How to I generate the deb packages for the kernel?

T-vK, 2018-02-16:
I successfully have built version 4.14.18 of the pi kernel. There are instructions on how to install the kernel manually here: https://www.raspberrypi.org/documentation/linux/kernel/building.md

But I would like to host my own kernel repository. Therefore I need to generate the .deb packages, though. 

Could you please tell me how this is done?

kukabu, 2018-02-16:
use make deb-pkg

T-vK, 2018-02-16:
That's all? Where will the deb files be saved?

kukabu, 2018-02-16:
in a directory one level up

T-vK, 2018-02-16:
I don't know, I can't find it. "One level up" is a bit messed up because I check out to `/linux` and there are dozens of folders in `/`. None of those seem too contain my .deb files, though. Can you tell me the name of the directory?

closed 2018-02-16 by T-vK