raspberrypi/linux #2385: Loading dtoverlay=dwc2 breaks USB on RPi3 with Raspbian kernel, but works fine with custom build

ali1234, 2018-02-16:
If I load the dwc2 overlay on the current raspbian kernel on a RPi3, USB is completely broken. I get no root hub, no ethernet etc.

If I build the same tag (raspberrypi-kernel_1.20171029-1) using my own configuration and load the dwc2 overlay, then USB works fine.

Here is my kernel configuration: https://github.com/ali1234/rpi-ramdisk/blob/b0c30756e12c2d58f42367b87a8e7f19eedf6fe3/kernel/kernel7.config

I am booting my custom kernel over PXE.

popcornmix, 2018-02-18:
I note you have 
```
CONFIG_USB_DWC2=y
CONFIG_USB_GADGET=y
```
We build them as modules. If you change these settings to ``=m`` does your kernel behave the same as raspbian one?

ali1234, 2018-02-18:
No. When those are built as modules it still works.

The only thing I can think of is that PXE booting means the USB and ethernet are already initialized before Linux starts. I have not ever tried PXE booting the raspbian kernel.

pelwell, 2018-02-18:
CONFIG_USB_DWC2 is not enabled in bcm2709_defconfig on the rpi-4.9.y branch. It is in bcmrpi_defconfig because of its use for device mode on Pi Zero, but otherwise we prefer the downstream driver and its FIQ acceleration.

USB_DWC2 is enabled in bcm2709_defconfig on rpi-4.14.y. Do you need to use 4.9?

ali1234, 2018-02-19:
I need to enable the dwc2 overlay in config.txt and have the card work in any model of RPi, with OTG mode available when possible, and host mode USB available otherwise. This already works for my custom kernel configuration, so I don't need anything more. However, it would be nice if this also worked with the default kernel in Raspbian.

Will DWC2 be used everywhere by default in 4.14? Will 4.14 be released for Stretch or Buster?

pelwell, 2018-02-19:
DWC2 is available on all RPi kernel builds in 4.14, but I can't see us using it by default. As the rpi-update master branch is now on 4.14 it should become the Stretch kernel fairly soon (assuming no problems are found).

lategoodbye, 2018-02-19:
@ali1234 Yes, the DWC2 in 4.14 is much better than in 4.9. But such a question highly dependsw on your expectation. Better you ask is issue XYZ fixed?

pelwell, 2018-02-19:
> Is the DWC2 driver in 4.14 a significant improvement over the one in 4.9?

That's an odd question. You are reporting that DWC2 doesn't work on a Pi3 running 4.9, but you sound reluctant to try 4.14 unless DWC2 works better. Following the "bird in the hand" principal the 4.14 DWC2 driver is infinitely better on Pi3 than the 4.9 equivalent.

ali1234, 2018-02-19:
That is not what I am reporting at all.

I am reporting that it isn't possible to configure Raspbian such that the dwc2 driver is used if you put it in a Pi Zero and the upstream driver is used if you put it in an RPi 3.

I am reporting that the dwc2 overlay breaks USB if the kernel is built without dwc2 support. Ideally if the dwc2 driver is not present, loading this overlay should have no effect and the upstream driver should continue to be used.

If that is not practical then a mechanism for conditionally loading overlays should be implemented in the bootloader, similar to the kernel=/kernel7= parameters.

If that also is not practical then the dwc2 driver should be improved until it is equal in quality to the upstream driver, so that having the overlay present on an RPi3 does not degrade the performance. In the case that dwc2 reaches parity with the upstream driver, then there is no reason not to use it by default.

Hence my question about whether the dwc2 driver has improved in 4.14.

pelwell, 2018-02-19:
```
[pi0]
dtoverlay=dwc2
```

ali1234, 2018-02-19:
Fair enough, but that isn't granular enough to differentiate between CM3/RPi3.

pelwell, 2018-02-19:
Upgrade a Raspbian image to 4.14 with rpi-update and confirm that it behaves the same way as your custom kernel.

ali1234, 2018-02-19:
Okay, but first let me rephrase my question: what has changed in the dwc2 driver that makes it suitable for inclusion in the defconfig for 4.14 but not for 4.9?

ali1234, 2018-02-19:
Confirming that the 4.14 kernel has working USB with the dwc2 overlay on a RPi3.

pelwell, 2018-02-19:
I'm not sure. It got enabled in 4.10 (https://github.com/raspberrypi/linux/commit/4efab5e31e1c12bc6f7c72231f621a149b4df45f) and carried over into later kernels, but never back-ported. As you've found it works on 4.9, but nobody asked for it until now when 4.9 is EOL for us (although we may continue to support it for a while - perhaps making it the next stable branch).

ali1234, 2018-02-19:
To be clear, I am not asking for you to put dwc2 in the 4.9 defconfig. I only noticed this because I was already using a custom kernel which only includes dwc2 and not the downstream driver. I have already [upgraded to 4.14](https://github.com/ali1234/rpi-ramdisk/commit/d3a1f9406807f531c86dc33fdc0ecdf5245678a8) after you said Stretch will get it eventually, but 4.9 was working fine for me anyway. This only affects anyone using the Foundation kernel for Raspbian, and anyone else still on 4.9.

It does re-raise questions about whether I should include the downstream version in my configuration, but as there's no way to make sure the "best" driver is loaded on every model I probably won't.

JamesH65, 2018-04-23:
Can this issue be closed? Reads like moving to 4.14 means it can be.

closed 2018-04-25 by ali1234