raspberrypi/linux #2390: dwc2_hsotg_init_fifo: insufficient fifo memory

ali1234, 2018-02-20:
This seems to be a regression between 4.9 and 4.14.

There is a thread about it on the LKML: https://lkml.org/lkml/2017/2/24/723

Log:
```
[   54.672566] WARNING: CPU: 0 PID: 132 at drivers/usb/dwc2/gadget.c:330 dwc2_hsotg_init_fifo+0x178/0x1b4
[   54.682162] insufficient fifo memory
[   54.682170] Modules linked in: brcmfmac brcmutil
[   54.690648] CPU: 0 PID: 132 Comm: pymtpd Not tainted 4.14.18+ #2
[   54.696839] Hardware name: BCM2835
[   54.700408] [<c0016138>] (unwind_backtrace) from [<c0013a60>] (show_stack+0x20/0x24)
[   54.708422] [<c0013a60>] (show_stack) from [<c06d254c>] (dump_stack+0x20/0x28)
[   54.719467] [<c06d254c>] (dump_stack) from [<c002365c>] (__warn+0xe8/0x110)
[   54.730383] [<c002365c>] (__warn) from [<c00236cc>] (warn_slowpath_fmt+0x48/0x50)
[   54.745433] [<c00236cc>] (warn_slowpath_fmt) from [<c03d67c0>] (dwc2_hsotg_init_fifo+0x178/0x1b4)
[   54.761963] [<c03d67c0>] (dwc2_hsotg_init_fifo) from [<c03d9214>] (dwc2_hsotg_core_init_disconnected+0x88/0x420)
[   54.779880] [<c03d9214>] (dwc2_hsotg_core_init_disconnected) from [<c03da0fc>] (dwc2_hsotg_pullup+0x108/0x134)
[   54.797727] [<c03da0fc>] (dwc2_hsotg_pullup) from [<c03e6e54>] (usb_gadget_connect+0x8c/0xac)
[   54.814161] [<c03e6e54>] (usb_gadget_connect) from [<c03e7024>] (udc_bind_to_driver+0x104/0x10c)
[   54.831193] [<c03e7024>] (udc_bind_to_driver) from [<c03e7e14>] (usb_gadget_probe_driver+0x100/0x148)
[   54.849024] [<c03e7e14>] (usb_gadget_probe_driver) from [<c03e5bbc>] (gadget_dev_desc_UDC_store+0xb8/0xd4)
[   54.867533] [<c03e5bbc>] (gadget_dev_desc_UDC_store) from [<c01d5074>] (configfs_write_file+0xd0/0x154)
[   54.885914] [<c01d5074>] (configfs_write_file) from [<c015d93c>] (__vfs_write+0x38/0x138)
[   54.903295] [<c015d93c>] (__vfs_write) from [<c015dc04>] (vfs_write+0xb0/0x1c0)
[   54.915379] [<c015dc04>] (vfs_write) from [<c015de44>] (SyS_write+0x4c/0xa0)
[   54.927173] [<c015de44>] (SyS_write) from [<c000fa80>] (ret_fast_syscall+0x0/0x28)
```

lategoodbye, 2018-03-04:
Which Raspberry Pi do you use?
Which overlay settings do you use?

ali1234, 2018-03-04:
Zero W and dwc2 overlay.

maldata, 2018-03-09:
This is also an issue on the Compute Module 3 mounted on the CMIO3.

`Linux raspberrypi 4.14.24-v7+ #1097 SMP Mon Mar 5 16:42:05 GMT 2018 armv7l GNU/Linux`

```
[    3.651733] g_ether gadget: Ethernet Gadget, version: Memorial Day 2008
[    3.651739] g_ether gadget: g_ether ready
[    3.651754] dwc2 3f980000.usb: bound driver g_ether
[    3.651772] ------------[ cut here ]------------
[    3.651848] WARNING: CPU: 0 PID: 102 at drivers/usb/dwc2/gadget.c:330 dwc2_hsotg_init_fifo+0x184/0x1cc [dwc2]
[    3.651853] insufficient fifo memory
[    3.651857] Modules linked in: usb_f_ecm g_ether(+) usb_f_rndis u_ether libcomposite dwc2 udc_core ip_tables x_tables ipv6
[    3.651903] CPU: 0 PID: 102 Comm: systemd-modules Not tainted 4.14.24-v7+ #1097
[    3.651907] Hardware name: BCM2835
[    3.651945] [<8010fff8>] (unwind_backtrace) from [<8010c260>] (show_stack+0x20/0x24)
[    3.651963] [<8010c260>] (show_stack) from [<8076160c>] (dump_stack+0xd4/0x118)
[    3.651979] [<8076160c>] (dump_stack) from [<8011d6d8>] (__warn+0xf8/0x110)
[    3.651993] [<8011d6d8>] (__warn) from [<8011d738>] (warn_slowpath_fmt+0x48/0x50)
[    3.652051] [<8011d738>] (warn_slowpath_fmt) from [<7f0c2ae0>] (dwc2_hsotg_init_fifo+0x184/0x1cc [dwc2])
[    3.652159] [<7f0c2ae0>] (dwc2_hsotg_init_fifo [dwc2]) from [<7f0c5304>] (dwc2_hsotg_core_init_disconnected+0x80/0x400 [dwc2])
[    3.652261] [<7f0c5304>] (dwc2_hsotg_core_init_disconnected [dwc2]) from [<7f0c6144>] (dwc2_hsotg_pullup+0xd8/0x108 [dwc2])
[    3.652355] [<7f0c6144>] (dwc2_hsotg_pullup [dwc2]) from [<7f09fbd4>] (usb_gadget_connect+0xcc/0xe8 [udc_core])
[    3.652420] [<7f09fbd4>] (usb_gadget_connect [udc_core]) from [<7f0a0f80>] (udc_bind_to_driver+0x108/0x10c [udc_core])
[    3.652483] [<7f0a0f80>] (udc_bind_to_driver [udc_core]) from [<7f0a1424>] (usb_gadget_probe_driver+0x108/0x150 [udc_core])
[    3.652559] [<7f0a1424>] (usb_gadget_probe_driver [udc_core]) from [<7f0e16d0>] (usb_composite_probe+0xcc/0xd8 [libcomposite])
[    3.652615] [<7f0e16d0>] (usb_composite_probe [libcomposite]) from [<7f111014>] (eth_driver_init+0x14/0x1000 [g_ether])
[    3.652643] [<7f111014>] (eth_driver_init [g_ether]) from [<80101bf0>] (do_one_initcall+0x50/0x17c)
[    3.652660] [<80101bf0>] (do_one_initcall) from [<801a8ee4>] (do_init_module+0x74/0x20c)
[    3.652675] [<801a8ee4>] (do_init_module) from [<801a7f08>] (load_module+0x1c28/0x23a0)
[    3.652689] [<801a7f08>] (load_module) from [<801a885c>] (SyS_finit_module+0x9c/0xac)
[    3.652705] [<801a885c>] (SyS_finit_module) from [<80108080>] (ret_fast_syscall+0x0/0x28)
[    3.652711] ---[ end trace 7e2b627b8a3443cd ]---
[    4.173039] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    4.216329] bcm2835_alsa bcm2835_alsa: card created with 8 channels
```

lategoodbye, 2018-03-09:
Please can anyone who is able to reproduce this issue, try the [patch](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/drivers/usb/dwc2?h=v4.15.7&id=9273083a1530891360e9fe4fad26ae96810db499) mentioned above?

ED6E0F17, 2018-03-19:
I think that may be why I wrote: https://github.com/ED6E0F17/linux/commit/3ed03d228f3f531bc4284fc4c7b1d01996423b5e

There are 7 fifos, and from Linux 4.12 the kernel no longer accepts zero size fifos, so the last one gets set to a default size, and exceeds available fifo memory.

I will look at the upstream patch when I can, but I have been using 4.9, and have not upgraded to 4.14 yet.

ED6E0F17, 2018-03-23:
@lategoodbye : On linux 4.14 I can use 4064 of the 4080 gadget fifo elements that should be available; I assume that the patch you link will return the missing 16.

The Upstream device tree uses 3584 + 256 + 32 = 3872 fifo slots. Downstream was only using 3616, but this was increased by 768 after Kernel 4.12 sanity checked the last buffer that had previously been set to zero size and disabled.

The size of the rx buffer can be increased to make use of the spare fifo memory.

JamesH65, 2018-04-23:
@lategoodbye @ED6E0F17 What's the current status of this one?

ED6E0F17, 2018-04-24:
My fix was merged downstream, so I expect that this can be closed now.

I found another Issue in the Upstream mailing lists that the final fifos should not be more than 512 anyway because of another quirk, so the Downstream settings are now close to what was proposed for Upstream, but not yet merged there.

JamesH65, 2018-04-24:
Thanks for the update

closed 2018-04-24 by JamesH65