raspberrypi/linux #2392: mmc0 Timeout waiting for hardware interrupt

stschake, 2018-02-21:
Hey,

we've been seeing the above error randomly occur with Raspberry Pi 3s. SD card manufacturer or new/old doesn't seem to make any difference. I've attached the full message log for the error.

Is this a known issue that can be mitigated somehow? We can't use the alternate SD card controller since we need the WLAN connectivity.

[mmc_timeout.txt](https://github.com/raspberrypi/linux/files/1743957/mmc_timeout.txt)

pelwell, 2018-02-21:
This is not a common failure - after a difficult development and some initial glitches the sdhost driver has been stable for at least a year now. There could well be bugs, but if so they must have been hiding well.

The log shows the last transfer (prior to this the interface was idle, periodically checking that the card is still present):
```
[13587.156458] [e97700ea] PRD< c2231240 0
[13587.162954] [e97700eb] PRD1 dea95b10 0
[13587.169461] [e97700f3] PRD2 a 0
[13587.175348] [e9770104] PRD3 cab66184 0
[13587.181863] [e9770105] PDM> c2231240 0
[13587.188327] [e9770106] REQ< c2231168 10801
[13587.195172] [e9770107] CMD< 17 50
[13587.201200] [e977010b] FCM< c2231168 c22311a4
[13587.208322] [e977010d] RSP  900 0
[13587.214378] [e977010e] CMD< 12 19c380
[13587.220771] [e9770110] CMDD 50 200
[13587.226933] [e9770110] SDMA c2231240 dea95b10
[13587.234098] [e9770112] FCM< c2231168 c22311d8
[13587.241258] [e9770113] RSP  900 0
[13587.247337] [e9770115] FCM> c2231168 0
[13587.253826] [e9770116] FCM> c2231168 0
[13587.260321] [e9770116] CMD  12 200
[13587.266457] [e9770116] REQ> c2231168 0
[13587.272942] [ea16f1e2] TIM< 0 0
```
It's a CMD23 (0x17 - Set block count) indicating an 80 sector transfer, followed by a CMD18 (0x12 - Read multiple blocks) starting at sector 0x19c380 (about 845MB from the start), preceded by the necessary DMA setup, but for some reason the DMA never completes so the transfer times out.

How are you powering the Pi3? Is the power supply capable of delivering 2.5A? Do you ever see the yellow lightning signal appear on the display? What does running `vcgencmd get_throttled` report during typical activity?

stschake, 2018-02-22:
We don't have the warning signs enabled since we use the full KMS. I have just tested with high CPU load for a good time and there was no undervoltage event observed. The power supplies are rated for 2.5A.

I'll see if I can get some more traces, unfortunately there sometimes seem to be IO related lockups that don't even trigger the timeout.

pelwell, 2018-02-22:
You say you've tried multiple Pi3s and multiple SD cards, so there must be something different about your environment otherwise lots of people would be reporting similar problems. Can you post your config.txt and cmdline.txt, and tell me a bit about the software you are running?

stschake, 2018-02-22:
I did fine one recent report of similar issues: https://patchwork.kernel.org/patch/10219145/
From what I can tell, the upstream sdhost driver mentioned there is identical to the downstream one, minus the extensive logging/tracing.

I've attached config/cmdline. The software is indeed a bit more special. While the kernel is rpi-4.9.y with a few choice cherry-picks from AOSP kernel common, in userland we run Android on a mesa3d v18 graphics stack.

[cmdline.txt](https://github.com/raspberrypi/linux/files/1748553/cmdline.txt)
[config.txt](https://github.com/raspberrypi/linux/files/1748554/config.txt)

pelwell, 2018-02-22:
That all looks reasonable, as do the two patches you referenced. I'll do something similar in the downstream version of the driver (with a hat-tip to the original poster).

pelwell, 2018-02-22:
I've got a patch which seems to work, although I've only been able to test it by faking a timeout. If I upload the patch somewhere will you be able to try it and test it?

pelwell, 2018-02-22:
The patch can be downloaded [here](https://gist.github.com/pelwell/265baccf92ea4f2d8e47c4f9e1655714).

stschake, 2018-02-23:
Thanks, I'll test that. It is unfortunately difficult to reproduce reliably.

pelwell, 2018-02-23:
Any evidence of a single timeout from which it recovers would be great, but since the timeouts are basically fatal at the moment the required confidence level for merging is pretty low.

oniongarlic, 2018-02-26:
Just a note, I've started to experience this with a SD card (Samsung 16 EVO, exact same model as  mentioned in the patchworks thread linked above) that was previously running wheezy on a Pi1B without any issues and re-written with latest Raspbian. Tested on both Pi2 B and an original Pi1, both fail. A pretty reliable way to reproduce is running badblocks over the card.

lategoodbye, 2018-03-01:
@stschake:

Do you also use a Samsung 16 EVO without plus?
Which driver do you use bcm2835.c or bcm2835-sdhost.c (they are more different than you think)?

@oniongarlic 

Are you able to reproduce this issue with a different SD card than Samsung 16 EVO?

JamesH65, 2018-04-23:
@stschake Did @pelwell patch help in any way or are you still seeing the issue?

@oniongarlic Any further comment?

stschake, 2018-04-23:
Sorry, I haven't seen the issue since - but since the patch would only trigger on an actual timeout, I can't say either way.

nullr0ute, 2018-04-23:
I've seen some issues like this on Fedora, I had one specific RPi3 that would see similar style issues and went through a number of good quality (Sandisk Ultra and Samsung EVO) in the process but seems to have settled with 4.15+ (although maybe it's wishful thinking there). I wonder if the move of the MMC stack to MQ and some of the improvements there have helped this problem or maybe masked it in different ways?

JamesH65, 2018-06-27:
@pelwell Thoughts on closing this?

pelwell, 2018-06-27:
OK with me. Closed issues are still searchable and can be re-opened as needed, but this been quiet for long enough.

closed 2018-06-27 by pelwell

lategoodbye, 2018-12-21:
The preferred method is to a connected USB to serial adapter on the debug UART. But a photo of the beginning of the stacktrace would be okay.

RPGillespie6, 2019-03-25:
Also chiming in here - I'm seeing a similar issue during boot with a Yocto linux image, but on an EspressoBin instead of a RaspberryPi. Different hardware true, but issues could be related. Here's the UART dump I get:

```
[   25.495102] mmc0: Timeout waiting for hardware interrupt.
[   25.500400] ------------[ cut here ]------------
[   25.505381] WARNING: at drivers/mmc/host/sdhci.c:1014
[   25.510694] Modules linked in:
[   25.513671] 
[   25.515299] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.4.52-armada-17.10.4-gb3509f8 #1
[   25.523219] Hardware name: EspressoBin, 256MB DDR3, Marvell 88E1512 PHY (DT)
[   25.531319] task: ffffffc006900000 ti: ffffffc006908000 task.ti: ffffffc006908000
[   25.538887] PC is at sdhci_send_command+0x6c0/0xa78
[   25.543920] LR is at sdhci_finish_data+0xd0/0x2f0
[   25.548957] pc : [<ffffffc0007e2110>] lr : [<ffffffc0007e2800>] pstate: 400001c5
[   25.556602] sp : ffffffc00690bb20
[   25.559935] x29: ffffffc00690bb20 x28: ffffffc006908000 
[   25.565517] x27: ffffffc000dae000 x26: ffffffc00bfd7bb8 
[   25.571010] x25: ffffffc000e88ea0 x24: ffffffc006908000 
[   25.576504] x23: ffffffc006183130 x22: 0000000000000002 
[   25.581998] x21: 0000000000000003 x20: ffffffc006183170 
[   25.587494] x19: ffffffc00616b500 x18: 0000000000000000 
[   25.592988] x17: 0000007f7e20f940 x16: ffffffc0001fec68 
[   25.598484] x15: 0000000000000010 x14: ffffffc080ea017f 
[   25.603979] x13: ffffffc000ea018d x12: 0000000000000006 
[   25.609473] x11: 0000000000000006 x10: 0000000005f5e0ff 
[   25.614970] x9 : 0000000000000150 x8 : 7265746e69206572 
[   25.620463] x7 : 6177647261682072 x6 : 0000000006c00000 
[   25.625958] x5 : ffffffc0000997f0 x4 : ffffffc000ebfe08 
[   25.631451] x3 : 0000000000000002 x2 : ffffffc0007e61c8 
[   25.636948] x1 : ffffffc006183130 x0 : ffffffc0061830f0 
[   25.642441] 
[   25.643972] ---[ end trace d7ad8c474e1fcc82 ]---
[   25.648922] Call trace:
[   25.651272] [<ffffffc0007e2110>] sdhci_send_command+0x6c0/0xa78
[   25.657483] [<ffffffc0007e2800>] sdhci_finish_data+0xd0/0x2f0
[   25.663514] [<ffffffc0007e2bec>] sdhci_timeout_timer+0x84/0xd8
[   25.669379] [<ffffffc00011e148>] call_timer_fn+0x38/0x1c8
[   25.674689] [<ffffffc00011e4dc>] run_timer_softirq+0x204/0x270
[   25.681081] [<ffffffc0000c80d0>] __do_softirq+0x128/0x360
[   25.686749] [<ffffffc0000c85c8>] irq_exit+0x98/0xf0
[   25.691708] [<ffffffc00010b020>] __handle_domain_irq+0x60/0xb8
[   25.697827] [<ffffffc0000827c4>] gic_handle_irq+0xbc/0x168
[   25.703237] Exception stack(0xffffffc00690bde0 to 0xffffffc00690bf10)
[   25.709898] bde0: ffffffc000daf000 0000007fffffffff ffffffc00690bf40 ffffffc000086848
[   25.717905] be00: 0000000000000145 ffffffc000d85ad8 0000000000000000 0000000000000000
[   25.726365] be20: 0000000000000001 0000000000000000 00000000000001c0 0100000000000000
[   25.734475] be40: 0000000000000000 ffffffc00bfdd040 ffffffc0069007c0 ffffffc00690bed0
[   25.742226] be60: 0000000000000760 0000000000000000 0000000000000000 0000000000091b92
[   25.750326] be80: 0000000000000000 0000000000000000 ffffffc0001fec68 0000007f7e20f940
[   25.758427] bea0: 00000000000007b1 ffffffc000daf000 ffffffc000dafae8 ffffffc0009bd370
[   25.766886] bec0: 0000000000000000 0000000000000000 ffffffc000d85ad8 ffffffc000e89000
[   25.774816] bee0: ffffffc0009bd000 ffffffc00690bf70 ffffffc006908000 ffffffc00690bf40
[   25.783003] bf00: ffffffc000086844 ffffffc00690bf40
[   25.787869] [<ffffffc000085700>] el1_irq+0x80/0xf8
[   25.792732] [<ffffffc000100524>] default_idle_call+0x1c/0x38
[   25.798669] [<ffffffc000100880>] cpu_startup_entry+0x2f0/0x340
[   25.804970] [<ffffffc00009014c>] secondary_start_kernel+0x12c/0x150
[   25.811367] [<00000000000829dc>] 0x829dc
[   35.511102] mmc0: Timeout waiting for hardware interrupt.
[   36.495096] INFO: rcu_preempt detected stalls on CPUs/tasks:
[   36.500799]  0-...: (1 GPs behind) idle=969/140000000000001/0 softirq=2203/2208 fqs=5170 
[   36.509158]  (detected by 1, t=5252 jiffies, g=537, c=536, q=672)
[   36.515372] Task dump for CPU 0:
[   36.518701] stats_collector R  running task        0  1691   1679 0x00000002
[   36.525993] Call trace:
[   36.528615] [<ffffffc000086e28>] __switch_to+0x88/0xa0
[   36.534008] [<ffffffc00010b01c>] __handle_domain_irq+0x5c/0xb8
[   36.540217] [<ffffffc0000827c4>] gic_handle_irq+0xbc/0x168
[   36.545895] Exception stack(0xffffffc004bafe90 to 0xffffffc004baffc0)
[   36.552563] fe80:                                   0000000000000000 0000007fb2185972
[   36.560402] fea0: ffffffffffffffff 0000007fb22033f4 0000000060000000 0000007fb2181110
[   36.568594] fec0: 0000007fb2185972 0000007fb2181110 0000000000000000 0000000000000001
[   36.576607] fee0: 0000000000000004 00000000004040b0 00000000000001b5 0000000000406570
[   36.584889] ff00: 0000000000000022 000000000000001b 0101010101010101 0000007fb21f7b40
[   36.592903] ff20: 0000007fb1c19da8 0000007fb221a9a8 0000007fb2227000 0000007fb1c07414
[   36.600825] ff40: 0000007fb2228000 0000007fb220ff00 0000000000040434 00000000004017b8
[   36.609101] ff60: 0000007fb2185972 0000007fccd5e098 0000000000406570 0000007fb2229190
[   36.617290] ff80: 0000007fb2181110 00000000000001b5 0000000000000004 0000000000000000
[   36.625303] ffa0: 0000007fccd5e094 0000007fccd5df50 0000007fb22038b0 0000007fccd5df50
[   36.633583] [<ffffffc000085c18>] el0_irq_naked+0x1c/0x24
[   45.527102] mmc0: Timeout waiting for hardware interrupt.
[   45.532989] mmcblk0: error -110 sending status command, retrying
[   55.543102] mmc0: Timeout waiting for hardware interrupt.
[   55.548713] mmcblk0: error -110 sending status command, retrying
[   65.559103] mmc0: Timeout waiting for hardware interrupt.
[   65.564732] mmcblk0: error -110 sending status command, aborting
[   65.570637] blk_update_request: I/O error, dev mmcblk0, sector 606500
[   65.577424] blk_update_request: I/O error, dev mmcblk0, sector 606508
[   65.584295] blk_update_request: I/O error, dev mmcblk0, sector 606516
```

bvrulez, 2019-11-20:
I also have this timeout message on a Raspberry 3B with Raspbian Buster. Don't know if it is related.

Originally, I tried to find out why my `bitcoind` process is terminating after some time: https://bitcoin.stackexchange.com/questions/91582/where-can-i-find-a-usefull-log-if-bitcoind-stopped-for-no-apparent-reason

jackw01, 2020-01-26:
Just started experiencing this timeout message on an SD card that was installed in a Raspberry Pi used in an LED lighting installation that was vandalized by means of cutting the power cord. Initially I thought the card was just corrupted, but even after reflashing it multiple times it causes this message on any Raspberry Pi I use it in. It seems to work fine on my computer.

msandres13, 2020-07-01:
Seeing simular errors with a Raspberrypi 4b and original Pi Powersupply.
`[ 5098.170186] mmc0: Timeout waiting for hardware cmd interrupt. Pi has no SD Card, only m.2 SSD on USB3 port with Beta Firmware from 6/15/20.
[ 5098.170194] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[ 5098.170202] mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
[ 5098.170206] mmc0: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[ 5098.170210] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[ 5098.170214] mmc0: sdhci: Present:   0x1fff0001 | Host ctl: 0x00000001
[ 5098.170218] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[ 5098.170222] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x0000f447
[ 5098.170226] mmc0: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[ 5098.170229] mmc0: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
[ 5098.170233] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[ 5098.170237] mmc0: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
[ 5098.170241] mmc0: sdhci: Cmd:       0x00000502 | Max curr: 0x00080008
[ 5098.170245] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[ 5098.170248] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[ 5098.170251] mmc0: sdhci: Host ctl2: 0x00000000
[ 5098.170255] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[ 5098.170258] mmc0: sdhci: ============================================
[ 5108.410373] mmc0: Timeout waiting for hardware cmd interrupt.
[ 5108.410380] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[ 5108.410387] mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
[ 5108.410391] mmc0: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[ 5108.410395] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[ 5108.410399] mmc0: sdhci: Present:   0x1fff0001 | Host ctl: 0x00000001
[ 5108.410403] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[ 5108.410407] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x0000f447
[ 5108.410411] mmc0: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[ 5108.410414] mmc0: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
[ 5108.410418] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[ 5108.410422] mmc0: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
[ 5108.410426] mmc0: sdhci: Cmd:       0x0000371a | Max curr: 0x00080008
[ 5108.410429] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[ 5108.410433] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[ 5108.410436] mmc0: sdhci: Host ctl2: 0x00000000
[ 5108.410440] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[ 5108.410443] mmc0: sdhci: ============================================
[ 5119.930605] mmc0: Timeout waiting for hardware cmd interrupt.
[ 5119.930612] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[ 5119.930620] mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
[ 5119.930624] mmc0: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[ 5119.930628] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[ 5119.930632] mmc0: sdhci: Present:   0x1fff0001 | Host ctl: 0x00000001
[ 5119.930636] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[ 5119.930640] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x0000f447
[ 5119.930643] mmc0: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[ 5119.930647] mmc0: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
[ 5119.930651] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[ 5119.930655] mmc0: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
[ 5119.930659] mmc0: sdhci: Cmd:       0x00000502 | Max curr: 0x00080008
[ 5119.930663] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[ 5119.930666] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[ 5119.930669] mmc0: sdhci: Host ctl2: 0x00000000
[ 5119.930673] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[ 5119.930677] mmc0: sdhci: ============================================
[ 5131.450797] mmc0: Timeout waiting for hardware cmd interrupt.
[ 5131.450803] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[ 5131.450809] mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
[ 5131.450813] mmc0: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[ 5131.450817] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[ 5131.450821] mmc0: sdhci: Present:   0x1fff0001 | Host ctl: 0x00000001
[ 5131.450825] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[ 5131.450829] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x0000f447
[ 5131.450832] mmc0: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[ 5131.450836] mmc0: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
[ 5131.450840] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[ 5131.450844] mmc0: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
[ 5131.450847] mmc0: sdhci: Cmd:       0x00000502 | Max curr: 0x00080008
[ 5131.450851] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[ 5131.450854] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[... 231 lines cut ...]
[ 5354.175009] mmc0: sdhci: Cmd:       0x00000502 | Max curr: 0x00080008
[ 5354.175018] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[ 5354.175026] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[ 5354.175034] mmc0: sdhci: Host ctl2: 0x00000000
[ 5354.175044] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[... cut ...]
[ 5390.655641] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[ 5390.655651] mmc0: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
[ 5390.655660] mmc0: sdhci: Cmd:       0x00000502 | Max curr: 0x00080008
[ 5390.655668] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[ 5390.655677] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[ 5390.655685] mmc0: sdhci: Host ctl2: 0x00000000
[ 5390.655694] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[ 5390.655701] mmc0: sdhci: ============================================
`

pelwell, 2020-07-01:
Without an SD card it's best to put `dtparam=sd_poll_once` in config.txt - that will prevent (or minimise) the errors

msandres13, 2020-07-01:
> Without an SD card it's best to put `dtparam=sd_poll_once` in config.txt - that will prevent (or minimise) the errors

Thanks a lot :-), that fixed it.

MichaIng, 2020-10-10:
Just just got two of them as well, 10 minutes after boot on RPi 2 v1.1 with 10 seconds between them, but not leading to any issues as far as I can tell now. While other pastes here show different outputs, also related to non SD card being attached, my errors match pretty the OP ones:
<details>

```
[  751.191823] mmc0: timeout waiting for hardware interrupt.
[  751.191867] [abd5cf13] FCM> b8f93c00 0
[  751.191884] [abd5cf15] CMD  d 0
[  751.191901] [abd5cf16] REQ> b8f93c00 0
[  751.191917] [abd5cf43] TSK< b8f93c00 0
[  751.191934] [abd5cf5e] TSK> b8f93c00 0
[  751.191950] [abd5cf81] PRD< b8f0ab90 0
[  751.191966] [abd5cf86] PRD1 b95a3e40 0
[  751.191982] [abd5cf90] PRD2 2 0
[  751.191998] [abd5cfa8] PRD3 b71d6084 0
[  751.192014] [abd5cfac] PDM> b8f0ab90 0
[  751.192031] [abd5cfb1] REQ< b8f0aab0 10801
[  751.192047] [abd5cfb2] CMD< 19 82000
[  751.192064] [abd5cfb8] CMDD 10 200
[  751.192081] [abd5cfb9] SDMA b8f0ab90 b95a3e40
[  751.192098] [abd5cfbd] FCM< b8f0aab0 b8f0ab28
[  751.192115] [abd5cfbe] RSP  900 0
[  751.192131] [abd5cfbf] FCM> b8f0aab0 0
[  751.192147] [abd5cfc0] CMD  19 200
[  751.192164] [abd5cfc1] REQ> b8f0aab0 0
[  751.192180] [2bd5d191] DMA< b8f0ab90 0
[  751.192197] [2bd5d192] DMA  99 10801
[  751.192213] [2bd5d195] FDA< b8f0aab0 0
[  751.192229] [2bd5d196] TCM< b8f0ab90 0
[  751.192246] [2bd5d198] CMD< c 0
[  751.192262] [2bd5d19a] TCM> b8f0ab90 0
[  751.192279] [2bd5d19b] FDA> b8f0aab0 b8f0ab5c
[  751.192294] [2bd5d19c] DMA> 0 0
[  751.192310] [2bd5d9c2] IRQ< 400 0
[  751.192327] [2bd5d9c4] IRQB b8f0ab5c 400
[  751.192344] [2bd5d9c5] FCM< b8f0aab0 b8f0ab5c
[  751.192360] [2bd5d9c7] RSP  c00 0
[  751.192377] [2bd5d9c9] FCM> b8f0aab0 b8f0ab5c
[  751.192392] [2bd5d9cb] IRQ> 0 0
[  751.192409] [2bd5d9d8] TSK< b8f0aab0 0
[  751.192425] [2bd5da0a] TSK> b8f0aab0 0
[  751.192442] [2bd5da4f] REQ< b8f4bdb8 10801
[  751.192458] [2bd5da50] CMD< d 70000
[  751.192475] [2bd5da57] FCM< b8f4bdb8 b8f4be20
[  751.192492] [2bd5da59] RSP  900 0
[  751.192508] [2bd5da6b] FCM> b8f4bdb8 0
[  751.192523] [2bd5da6d] CMD  d 0
[  751.192540] [2bd5da6e] REQ> b8f4bdb8 0
[  751.192556] [2bd5da9c] TSK< b8f4bdb8 0
[  751.192572] [2bd5dab7] TSK> b8f4bdb8 0
[  751.192589] [6bd5db4c] REQ< b8cf3dc8 10801
[  751.192605] [6bd5db50] CMD< d 70000
[  751.192622] [6bd5db58] FCM< b8cf3dc8 b8cf3e30
[  751.192638] [6bd5db5a] RSP  900 0
[  751.192655] [6bd5db6d] FCM> b8cf3dc8 0
[  751.192670] [6bd5db6f] CMD  d 0
[  751.192687] [6bd5db6f] REQ> b8cf3dc8 0
[  751.192703] [6bd5db9e] TSK< b8cf3dc8 0
[  751.192719] [6bd5dbb9] TSK> b8cf3dc8 0
[  751.192736] [6be59644] REQ< b8cf3dc8 10801
[  751.192752] [6be59648] CMD< d 70000
[  751.192769] [6be59651] FCM< b8cf3dc8 b8cf3e30
[  751.192785] [6be59655] RSP  900 0
[  751.192802] [6be59669] FCM> b8cf3dc8 0
[  751.192818] [6be5966a] CMD  d 0
[  751.192834] [6be5966b] REQ> b8cf3dc8 0
[  751.192850] [6be596a1] TSK< b8cf3dc8 0
[  751.192866] [6be596c2] TSK> b8cf3dc8 0
[  751.192883] [6bf574d1] REQ< b8cf3dc8 10801
[  751.192900] [6bf574d4] CMD< d 70000
[  751.192917] [6bf574df] FCM< b8cf3dc8 b8cf3e30
[... 461 lines cut ...]
[  761.435966] [2d00df7b] IRQD b8f05790 1
[  761.435983] [2d00df7d] XFP< b8f05790 1
[  761.435999] [2d00df99] XFP> b8f05790 1
[  761.436015] [2d00df9a] IRQ> 1 0
[  761.436030] [ed9d09d6] TIM< 0 0
[  761.436063] mmc0: cmd op 24 arg 0x2ff8 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[  761.436080] mmc0: data blocks 1 blksz 200 - err 0
[  761.436092] mmc0: =========== REGISTER DUMP ===========
[  761.436105] mmc0: SDCMD  0x00004098
[  761.436118] mmc0: SDARG  0x00002ff8
[  761.436130] mmc0: SDTOUT 0x017d7840
[  761.436142] mmc0: SDCDIV 0x00000003
[  761.436155] mmc0: SDRSP0 0x00000900
[  761.436167] mmc0: SDRSP1 0x0000182e
[  761.436179] mmc0: SDRSP2 0x7fffffff
[  761.436192] mmc0: SDRSP3 0x00024805
[  761.436204] mmc0: SDHSTS 0x00000081
[  761.436216] mmc0: SDVDD  0x00000001
[  761.436229] mmc0: SDEDM  0x00010801
[  761.436241] mmc0: SDHCFG 0x0000050e
[  761.436253] mmc0: SDHBCT 0x00000200
[  761.436265] mmc0: SDHBLC 0x00000001
[  761.436276] mmc0: ===========================================
```
</details>

MichaIng, 2020-10-14:
Got them again, after longer uptime:
<details>

```
[241550.240729] mmc0: timeout waiting for hardware interrupt.
[241550.240770] [3cb97fea] TCM> b8f03790 0
[241550.240787] [3cb97fea] FDA> b8f036b0 0
[241550.240803] [3cb97feb] DMA> 0 0
[241550.240820] [3cb97ffe] TSK< b8f036b0 0
[241550.240836] [3cb98018] TSK> b8f036b0 0
[241550.240852] [7cb9825b] PRD< b8f03990 0
[241550.240868] [7cb9825e] PRD1 b95a3e40 0
[241550.240884] [7cb98283] PRD2 1c 0
[241550.240901] [7cb982d4] PRD3 b7e9a404 0
[241550.240917] [7cb982d6] PDM> b8f03990 0
[241550.240934] [7cb982d9] REQ< b8f038b0 10801
[241550.240950] [7cb982da] CMD< 17 e0
[241550.240968] [7cb982dd] FCM< b8f038b0 b8f038f4
[241550.240984] [7cb982df] RSP  900 0
[241550.241001] [7cb982df] CMD< 12 31a020
[241550.241017] [7cb982e1] CMDD e0 200
[241550.241034] [7cb982e1] SDMA b8f03990 b95a3e40
[241550.241051] [7cb982e3] FCM< b8f038b0 b8f03928
[241550.241067] [7cb982e4] RSP  900 0
[241550.241083] [7cb982e5] FCM> b8f038b0 0
[241550.241100] [7cb982e5] FCM> b8f038b0 0
[241550.241116] [7cb982e6] CMD  12 200
[241550.241132] [7cb982e6] REQ> b8f038b0 0
[241550.241149] [3cb99017] DMA< b8f03990 1
[241550.241165] [3cb99018] DMA  52 10834
[241550.241181] [3cb9903f] FDA< b8f038b0 0
[241550.241198] [3cb99040] TCM< b8f03990 0
[241550.241214] [3cb99041] WTC< 10804 0
[241550.241230] [3cb99042] WTC> 10804 0
[241550.241247] [3cb99043] TCM> b8f03990 0
[241550.241263] [3cb99043] FDA> b8f038b0 0
[241550.241279] [3cb99044] DMA> 0 0
[241550.241295] [3cb99088] TSK< b8f038b0 0
[241550.241312] [3cb990a2] TSK> b8f038b0 0
[241550.241328] [7cb99236] PRD< b8f03b90 0
[241550.241344] [7cb99238] PRD1 b95a3e40 0
[241550.241360] [7cb99248] PRD2 a 0
[241550.241376] [7cb9926c] PRD3 b734c784 0
[241550.241392] [7cb9926e] PDM> b8f03b90 0
[241550.241409] [7cb99270] REQ< b8f03ab0 10801
[241550.241425] [7cb99271] CMD< 17 50
[241550.241442] [7cb99275] FCM< b8f03ab0 b8f03af4
[241550.241459] [7cb99276] RSP  900 0
[241550.241475] [7cb99277] CMD< 12 31a100
[241550.241491] [7cb99279] CMDD 50 200
[241550.241508] [7cb99279] SDMA b8f03b90 b95a3e40
[241550.241525] [7cb9927b] FCM< b8f03ab0 b8f03b28
[241550.241541] [7cb9927c] RSP  900 0
[241550.241557] [7cb9927d] FCM> b8f03ab0 0
[241550.241574] [7cb9927d] FCM> b8f03ab0 0
[241550.241590] [7cb9927e] CMD  12 200
[241550.241606] [7cb9927e] REQ> b8f03ab0 0
[241550.241623] [3cb998ca] DMA< b8f03b90 1
[241550.241639] [3cb998cc] DMA  52 10834
[241550.241655] [3cb998df] FDA< b8f03ab0 0
[241550.241672] [3cb998e0] TCM< b8f03b90 0
[241550.241688] [3cb998e1] WTC< 10804 0
[241550.241704] [3cb998e2] WTC> 10804 0
[241550.241720] [3cb998e4] TCM> b8f03b90 0
[241550.241737] [3cb998e4] FDA> b8f03ab0 0
[241550.241753] [3cb998e5] DMA> 0 0
[241550.241769] [3cb99906] TSK< b8f03ab0 0
[241550.241785] [3cb9992c] TSK> b8f03ab0 0
[241550.241801] [7cb99ec3] PRD< b8f03d90 0
[241550.241818] [7cb99ec5] PRD1 b95a3e40 0
[... 468 lines cut ...]
[241805.693611] mmc0: cmd op 24 arg 0x2001 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[241805.693628] mmc0: data blocks 1 blksz 200 - err 0
[241805.693641] mmc0: =========== REGISTER DUMP ===========
[241805.693653] mmc0: SDCMD  0x00004098
[241805.693666] mmc0: SDARG  0x00002001
[241805.693678] mmc0: SDTOUT 0x01dcd650
[241805.693690] mmc0: SDCDIV 0x00000002
[241805.693703] mmc0: SDRSP0 0x00000900
[241805.693715] mmc0: SDRSP1 0x0000182e
[241805.693727] mmc0: SDRSP2 0xffffffff
[241805.693740] mmc0: SDRSP3 0x0002400f
[241805.693752] mmc0: SDHSTS 0x00000081
[241805.693764] mmc0: SDVDD  0x00000001
[241805.693777] mmc0: SDEDM  0x00010801
[241805.693789] mmc0: SDHCFG 0x0000050e
[241805.693801] mmc0: SDHBCT 0x00000200
[241805.693813] mmc0: SDHBLC 0x00000001
[241805.693824] mmc0: ===========================================
[241805.693954] mmc0: reducing overclock due to errors
[241805.695009] mmc0: cancelling overclock
```
</details>
This time I had set `dtparam=sd_overclock=100`, which is why the firmware reduced it back to default in two steps, actually a nice feature. However in last post above this was `NOT` overclocked, so I don't expect it as the origin of the issue, as well since the overclocked SD frequency is something that worked nicely for the 5 years I run this RPi 24/7 😉.

I think the difficulty is that this error is very unspecific, is it? Related to the fact that I never saw this on any other firmware/kernel version (see above), probably someone has an idea where it might come from and how to debug? At least I would like to stay with overclocked mmc (which increases transfer rates significantly) and the SD card definitely supports it.

mennovf, 2020-10-29:
I have the same issue with the eMMC on multiple compute modules. There's no SD card. For some reason the timeout log and register dumps take more than 1s in a softIRQ, running with IRQs disabled. I have included the output from the sdhost driver from dmesg [here](https://gist.github.com/mennovf/dd2df1ba5e341ee95a7dd3b3e54b8520).

MichaIng, 2020-10-29:
In my case it seems to have been solved after enabling `uio_pdrv_genirq` kernel module. I was in attempt to find out which are required for what and without this it worked stable for a year until one of the Linux 5.4 releases, so I was not thinking about it anymore. But for some reason it seems to be required now. It "should" load by default, i.e. I blacklisted it before to prevent its load.

mennovf, 2020-11-04:
> I have the same issue with the eMMC on multiple compute modules. There's no SD card. For some reason the timeout log and register dumps take more than 1s in a softIRQ, running with IRQs disabled.
> I [...]

My issue seems to have been a hardware issue instead of a driver issue. Our controller board had the raspberry pi compute module IO board copied on it wholesale. Instead of connecting  a jumper to the connector for selecting the bootmode, we had a switch there, connected via a ~20cm wire. This wire probably picked up some noise during operation, switching the EMMC_DISABLE_N signal, causing issues with this driver.
My fix is outlined here: https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=289726

DG12, 2023-09-15:
Just reporting, after running for several YEARS.

2023-09-15 00:11:00 pi93graf kern.err kernel: [ 8518.855078] print_req_error: I/O error, dev mmcblk0, sector 42880405
2023-09-15 00:11:10 pi93graf kern.err kernel: [ 8529.092917] mmc0: timeout waiting for hardware interrupt.
2023-09-15 00:11:10 pi93graf kern.err kernel: [ 8529.095121] print_req_error: I/O error, dev mmcblk0, sector 42880406
...
2023-09-15 00:11:21 pi93graf kern.err kernel: [ 8539.335215] print_req_error: I/O error, dev mmcblk0, sector 42880407
2023-09-15 00:11:30 pi93graf daemon.err systemd[1]: systemd-journald.service: Watchdog timeout (limit 3min)!

Also from influxdb:

2023-09-15 08:09:23 pi93graf user.info influxd-systemd-start.sh[2501]: ts=2023-09-15T12:09:23.344486Z lvl=info msg="Compacting file" log_id=0kHjJEDG000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0kIBmEWl000 op_name=tsm1_compact_group db_shard_id=2066 tsm1_index=1 tsm1_file=/varnew/lib/influxdb/data/ruuvi/autogen/2066/000000002-000000001.tsm

2023-09-15 08:09:31 pi93graf user.info influxd-systemd-start.sh[2501]: ts=2023-09-15T12:09:31.579042Z lvl=info msg="Error replacing new TSM files" log_id=0kHjJEDG000 engine=tsm1 tsm1_strategy=full tsm1_optimize=false trace_id=0kIBmEWl000 op_name=tsm1_compact_group db_shard_id=2066 error="**cannot allocate memory**"

eastpole, 2023-11-10:
This is happening now on my up-to-date Raspberry Pi Zero 2. 

Because this has happened before and I could not see any reason, i left a couple logs open in screen.

from kernel.log: 
[seems normal up to here] 
Nov  9 20:09:19 troll systemd[1]: Removed slice User Slice of UID 0.
Nov  9 20:09:46 troll kernel: [213338.858612] mmc0: timeout waiting for hardware interrupt.
Nov  9 20:09:46 troll kernel: [213338.858635] [6b9d8a9c] PRD1 81785840 0
Nov  9 20:09:46 troll kernel: [213338.858653] [6b9d8a9f] PRD2 2 0

dnsmasq.log: 
[all normal up to here]
Nov 10 01:47:28 dnsmasq-dhcp[428]: failed to write /var/lib/misc/dnsmasq.leases: Read-only file system (retry in 60 s)
[repeated many times over subsequent hours; yes, it keeps serving DNS] 

It's hard to troubleshoot after the fact because no commands or files can be found on the filesystem, ls won't work, etc.
top continues to run and says that RAM is reasonably available, CPU mostly idle, etc. 

root@troll:/var/log# ls
bash: /bin/ls: Input/output error

Any suggestions welcome. I just wanted to confirm that this can still happen fairly frequently on a Pi with the current OS and updated firmware.  

I will add more info if I learn anything else.

DG12, 2023-11-18:
In my case I believe it to be a couple of bad locations on the SD !