raspberrypi/linux #2401: No result with poll_once on dtoverlay sdio-1bit

dalmago, 2018-02-26:
I connected a secondary SD card on my Raspberry Pi 3 B using the SDIO pins in "1 bit mode", but the system only finds the card when its already inserted on boot.

My **config.txt** has the following:
`dtoverlay=sdio-1bit,poll_once=off`

I also tried removing the `non-removable;` line from the **dts** file and tried compiling it again, but nothing changed.

dalmago, 2018-02-26:
I also tried using `poll_once=no` and adding `broken-cd` on the **dts** file.

pelwell, 2018-02-26:
This is an unfortunate problem with boolean Device Tree properties and overlays. Boolean properties are defined to be TRUE if the property is present, and FALSE if it isn't - the actual value assigned to the property is irrelevant. Overlays and add properties and modify them, but they can't delete them - it is a fundamental limitation of the overlay representation.

The MMC block on a Pi3 is used to control the WiFi interface via SDIO. We were advised by Broadcom to disable usual card status polling on the interface, hence the presence of the `non-removable` property in the base DTBs of the Pi 3 and the Pi Zero W.

There is an ugly workaround - it should be possible to modify the sdio overlays to disable the existing node (rather than patching it) and create a new one alongside it that only contains non-removable if poll_once is true. I'll post a DTBO here for testing.

pelwell, 2018-02-26:
Can you test [this .dtbo](https://drive.google.com/file/d/1V1AwYDyUcxyz70I8RMbfcopE2axSPcWq/view?usp=sharing) for me? Download it somewhere, then:
```
cp /boot/overlays/sdio-1bit.dtbo sdio-1bit.dtbo.bak
sudo cp sdio-1bit.dtbo /boot/overlays
````
and reboot.

dalmago, 2018-02-26:
I will test it as soon as I get my hands on the Raspberry again (probably tomorrow morning).
Thanks in advance.

pelwell, 2018-02-27:
Here you go: https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm/boot/dts/overlays/sdio-1bit-overlay.dts
It's in the rpi-4.14-y branch now, so it will be in future firmware+kernel releases.

closed 2018-02-27 by dalmago

DrW0if, 2018-03-16:
Using that dtbo, it's possible to use both the SD card and the wifi on a rpi 3? Thanks!

pelwell, 2018-03-16:
> both the SD card and the wifi

That's ambiguous, if one allows for less-than-perfect English. Out-of-the box the Pi 3 allows "both _the SD card_ and _the WiFi__" (i.e. the standard SD card interface and the WiFi) to be used, so I think you must mean "_both of the SD cards_ and _the WiFi_", in which case the answer is no. The WiFi chip occupies the SDIO interface, so it can't be used to drive a second SD card.

DrW0if, 2018-03-16:
Of course I mean secondary SD card and the built-in wifi chip. Sorry for bad explanation. So it's not possible, thanks anywhere!