raspberrypi/linux #2402: Stall while booting on new 4.16.y branch

Dark-Sky, 2018-02-27:
I have followed the latest dev branch for years and never had any issues until now. Can not get my pi 2 or 3 to boot. Been trying everything I can think of since the new 4.16 branch was created.

[image]

pelwell, 2018-02-27:
You are right on the bleeding edge with the 4.16 kernel - it only appeared in the repo a week ago and is still a work-in-progress; we find it easier to collaborate on GitHub, even right at the start of bringing up a new kernel version.

FIQ support is known to be somewhat broken, so add the following to `cmdline.txt`:
```
dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0
```

Dark-Sky, 2018-02-27:
Thanks that got it to boot. Going to test and see if anything else is affected. I never have had any real issues to speak of when using the latest in the past.

Thanks again.

[alarm@alarmpi ~]$ uname -a
Linux alarmpi 4.16.0-rc3-v7+ #1 SMP Tue Feb 27 10:12:42 CST 2018 armv7l GNU/Linux

closed 2018-02-27 by Dark-Sky