raspberrypi/linux #2403: Kernel crash with: Internal error: Oops: 37

mrRobot62, 2018-02-27:
Hi folks,

yesterday I made an apt-get update and apt-get upgrade on my Pi3 (running with STRETCH).
Over night  I got a kernel crash and looking inside kernel.log I found this:

Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.548170] **Unable to handle kernel paging request at virtual address ffffffe8**
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.552521] pgd = 80004000
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.556870] [ffffffe8] *pgd=377fa861, *pte=00000000, *ppte=00000000
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.561228] **Internal error: Oops: 37 [#1] SMP ARM**
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.565481] Modules linked in: cmac bnep hci_uart btbcm bluetooth evdev brcmfmac brcmutil cfg80211 rfkill snd_bcm2835 bcm2835_gpiomem snd_pcm snd_timer i2c_bcm2835 snd uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.575135] CPU: 0 PID: 17557 Comm: kworker/u8:0 Tainted: G        W       4.9.59-v7+ #1047

**Above I found this part**
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544315] **------------[ cut here ]------------**
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544340] WARNING: CPU: 0 PID: 16624 at mm/vmalloc.c:1494 __vunmap+0xb8/0xdc
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544346] Trying to vfree() bad address (ba000002)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544350] Modules linked in: cmac bnep hci_uart btbcm bluetooth evdev brcmfmac brcmutil cfg80211 rfkill snd_bcm2835 bcm2835_gpiomem snd_pcm snd_timer i2c_bcm2835 snd uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544446] CPU: 0 PID: 16624 Comm: kworker/0:0 Not tainted 4.9.59-v7+ #1047
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544451] Hardware name: BCM2835
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544624] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544653] [<8010fb3c>] (unwind_backtrace) from [<8010c058>] (show_stack+0x20/0x24)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544670] [<8010c058>] (show_stack) from [<80456764>] (dump_stack+0xd4/0x118)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544687] [<80456764>] (dump_stack) from [<8011d348>] (__warn+0xf8/0x110)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544700] [<8011d348>] (__warn) from [<8011d3a8>] (warn_slowpath_fmt+0x48/0x50)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544713] [<8011d3a8>] (warn_slowpath_fmt) from [<8024ee20>] (__vunmap+0xb8/0xdc)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544727] [<8024ee20>] (__vunmap) from [<8024eef4>] (vfree+0x58/0x9c)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.544887] [<8024eef4>] (vfree) from [<7f2d80e4>] (brcmf_sdio_bus_rxctl+0x184/0x250 [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.545200] [<7f2d80e4>] (brcmf_sdio_bus_rxctl [brcmfmac]) from [<7f2d4c44>] (brcmf_proto_bcdc_cmplt+0x64/0x7c [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.545505] [<7f2d4c44>] (brcmf_proto_bcdc_cmplt [brcmfmac]) from [<7f2d4e44>] (brcmf_proto_bcdc_set_dcmd+0x84/0xec [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.545810] [<7f2d4e44>] (brcmf_proto_bcdc_set_dcmd [brcmfmac]) from [<7f2c82b4>] (brcmf_fil_cmd_data+0x74/0xe8 [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.546117] [<7f2c82b4>] (brcmf_fil_cmd_data [brcmfmac]) from [<7f2c8688>] (brcmf_fil_cmd_int_set+0x70/0x90 [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.546423] [<7f2c8688>] (brcmf_fil_cmd_int_set [brcmfmac]) from [<7f2c0860>] (brcmf_do_escan+0x5c/0xdc [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.546728] [<7f2c0860>] (brcmf_do_escan [brcmfmac]) from [<7f2c1b10>] (brcmf_notify_sched_scan_results+0x25c/0x390 [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.547034] [<7f2c1b10>] (brcmf_notify_sched_scan_results [brcmfmac]) from [<7f2c8c9c>] (brcmf_fweh_call_event_handler+0x48/0x88 [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.547339] [<7f2c8c9c>] (brcmf_fweh_call_event_handler [brcmfmac]) from [<7f2c8ee0>] (brcmf_fweh_event_worker+0x204/0x444 [brcmfmac])
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.547502] [<7f2c8ee0>] (brcmf_fweh_event_worker [brcmfmac]) from [<80136720>] (process_one_work+0x158/0x454)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.547521] [<80136720>] (process_one_work) from [<80136a80>] (worker_thread+0x64/0x568)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.547538] [<80136a80>] (worker_thread) from [<8013c958>] (kthread+0x10c/0x124)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.547555] [<8013c958>] (kthread) from [<80108148>] (ret_from_fork+0x14/0x2c)
Feb 27 01:57:45 ioBroker-Pi3 kernel: [16254.547562] ---[ end trace 7d3f83ba85534680 ]---

Additional information

- installed on PI3 (test installation)
- installed software "ioBroker" (house automation)
- no external USB devices installed
- Connected via WLAN to my network
- 16GB SD-Card

Any ideas/help?

Thanks
Bernd

pelwell, 2018-02-27:
Is the SSID of your AP hidden? We have had reports of that causing problems with the latest firmware. The timing makes it unlikely to be the same problem, but it would be good to get a more complete picture of your environment.

mrRobot62, 2018-02-27:
no, SSID is not hidden

JamesH65, 2018-04-23:
@mrRobot62 Are you still seeing this issue?

JamesH65, 2018-05-11:
@mrRobot62 Will be closing due to lack of interaction in a few days unless asked to keep open.

closed 2018-06-27 by JamesH65