raspberrypi/linux #2404: dwc_otg_hcd_handle_hc_fsm:2459:Unexpected IRQ state on FSM transaction

jkuek, 2018-02-27:
I've just seen a similar issue to #2335 on one of my systems running Raspbian Wheezy (4.9.35). The unit stopped responding and was spamming this message to the console:

dwc_otg_hcd_handle_hc_fsm:2459:Unexpected IRQ state on FSM transaction:dev_addr=5 ep=0 fsm=6, hcint=0x00000002 

[image]

When I rebooted, there was no indication of a fault in the logs. The unit had previously been running continuously for 91 days without issue.

I've attached the output of lsusb and lsmod from after the unit was rebooted
[lsmod.txt](https://github.com/raspberrypi/linux/files/1765057/lsmod.txt)
[lsusb.txt](https://github.com/raspberrypi/linux/files/1765058/lsusb.txt)

JamesH65, 2018-04-23:
@P33M Any thoughts?
@jkuek was this a one off or is it still happening?

jkuek, 2018-05-07:
Sorry, for some reason I didn't get a notification on this.

I am definitely still seeing some hangs on 4.9.35 which I suspect are USB-driver related, but part of the problem is once the USB is broken, I can no longer access the console via USB, and the serial console is disabled on my unit, as the serial port is connected to another device.

ChrisEt, 2018-06-09:
I had a similar issue on my Pi 1 Model A (only it said `dev_addr=7` instead of `5`). It went away when using an newer power supply.

closed 2019-09-18 by JamesH65