raspberrypi/linux #2408: F/W 4.14.x causes dwc_otg_handle_mode_mismatch_intr:68: messages on boot with C-Media Electronics Inc. USB PnP Sound Device

hofnarr132, 2018-03-01:
Looks like an issue with dwc_otg was introduced with 4.14.x versions - 4.9.80 does not show issue.

When booting RPi 3, 2 or ZeroW with 4.14.x versions with a "C-Media Electronics Inc. USB PnP Sound Device" connected** get the following boot messages (extracted from dmesg):

```
[    3.292306] usb 1-1.3.4: new full-speed USB device number 7 using dwc_otg
[    3.320586] uart-pl011 3f201000.serial: no DMA platform data
[    3.450503] usb 1-1.3.4: New USB device found, idVendor=0d8c, idProduct=013c
[    3.467395] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    3.484570] usb 1-1.3.4: Product: USB PnP Sound Device          
[    3.500214] usb 1-1.3.4: Manufacturer: C-Media Electronics Inc.      
[    3.522058] input: C-Media Electronics Inc.       USB PnP Sound Device           as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.3/0003:0D8C:013C.0003/input/input2
[    3.622573] hid-generic 0003:0D8C:013C.0003: input,hidraw2: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device          ] on usb-3f980000.usb-1.3.4/input3
.
.
.
[    6.170183] bcm2835_alsa bcm2835_alsa: card created with 8 channels
[    6.374754] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[    6.374820] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[    6.376756] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[    6.406816] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[    6.410315] usbcore: registered new interface driver snd-usb-audio
[    7.518782] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[    7.520726] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[    7.712309] usb 1-1.5: reset high-speed USB device number 5 using dwc_otg
.
.
.
[    7.913820] usbcore: registered new interface driver rt2800usb
.
.
.
[   24.342952] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   24.343077] usb 1-1.3.4: 1:1: cannot set freq 44100 to ep 0x1
[   24.432846] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   24.512751] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   24.512823] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   24.526869] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   24.527122] usb 1-1.3.4: 2:1: cannot get freq at ep 0x82
[   24.528811] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   24.576791] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   24.576821] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
```

Tested and errors with 4.14.18 (and 4.14.22).
```
uname -a
Linux raspberrypi 4.14.18-v7+ #1093 SMP Fri Feb 9 15:33:07 GMT 2018 armv7l GNU/Linux
```

Back revved to 4.9.80 where the error messages do not show:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.80-v7+ #1092 SMP Fri Feb 9 13:56:00 GMT 2018 armv7l GNU/Linux
```

** note: the "C-Media Electronics Inc. USB PnP Sound Device" is actually embedded in a 4-Port HDMI & USB KVM Switch ([this one from ModMyPi](https://www.modmypi.com/raspberry-pi/accessories-198/video-cables-210/4-port-hdmi-and-usb-kvm-switch/)) - there is a push button switch to turn off audio which disables the Sound Device - the RPi will then boot without error (but then no sound..)

ajw99uk, 2018-03-10:
Experiencing similar symptoms, at first resolved by unplugging printer from USB hub to boot, but it has occurred without that connected (also unplugging the mouse and keyboard then enabled booting).
Sometimes the error appears a dozen or so times but booting proceeds as normal, sometimes it keeps appearing (requiring a power-off).
My current `lsusb` shows
`Bus 001 Device 009: ID 0d49:3200 Maxtor Personal Storage 3200
Bus 001 Device 008: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 007: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 006: ID 045e:0040 Microsoft Corp. Wheel Mouse Optical
Bus 001 Device 005: ID 04d9:1503 Holtek Semiconductor, Inc. Keyboard
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

Research suggested this may be a PSU issue, but (i) the USB hub has its own power, (ii) the error appears on a Pi2B with a 2.0A PSU, which should be adequate and (iii) I see no indications of under-powering during use - the system, once booted, is stable.

popcornmix, 2018-03-11:
@P33M any ideas on this one?

P33M, 2018-03-12:
Mode mismatch interrupts occur when a Device mode register is read/written while in host mode or vice versa - which is weird because the driver code hasn't changed between 4.9 and 4.14.

As the issue occurs across several Pi models, this is even more confusing.

You have several devices attached. Does the issue occur with just the C-media device attached?

Hellorheaven, 2018-03-13:
same problem here
Linux serverhome 4.14.24-v7+ #1097 SMP Mon Mar 5 16:42:05 GMT 2018 armv7l GNU/Linux

pelwell, 2018-03-13:
Please post the entire dmesg output.

Hellorheaven, 2018-03-13:
`Mar 11 06:27:03 smarthome liblogging-stdlog:  [origin software="rsyslogd" swVersion="8.24.0" x-pid="306" x-info="http://www.rsyslog.com"] rsyslogd was HUPed
Mar 11 14:00:06 smarthome kernel: [84656.159348] hrtimer: interrupt took 188854 ns
Mar 11 20:57:35 smarthome kernel: [    0.000000] Booting Linux on physical CPU 0x0
Mar 11 20:57:35 smarthome kernel: [    0.000000] Linux version 4.14.24-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1097 SMP Mon Mar 5 16:42:05 GMT 2018
Mar 11 20:57:35 smarthome kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Mar 11 20:57:35 smarthome kernel: [    0.000000] CPU: div instructions available: patching division code
Mar 11 20:57:35 smarthome kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Mar 11 20:57:35 smarthome kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
Mar 11 20:57:35 smarthome kernel: [    0.000000] Memory policy: Data cache writealloc
Mar 11 20:57:35 smarthome kernel: [    0.000000] cma: Reserved 8 MiB at 0x3dc00000
Mar 11 20:57:35 smarthome kernel: [    0.000000] random: fast init done
Mar 11 20:57:35 smarthome kernel: [    0.000000] percpu: Embedded 17 pages/cpu @be59f000 s38720 r8192 d22720 u69632
Mar 11 20:57:35 smarthome kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
Mar 11 20:57:35 smarthome kernel: [    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/sda2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Mar 11 20:57:35 smarthome kernel: [    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
Mar 11 20:57:35 smarthome kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Mar 11 20:57:35 smarthome kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Mar 11 20:57:35 smarthome kernel: [    0.000000] Memory: 990992K/1021952K available (7168K kernel code, 548K rwdata, 2028K rodata, 1024K init, 706K bss, 22768K reserved, 8192K cma-reserved)
Mar 11 20:57:35 smarthome kernel: [    0.000000] Virtual kernel memory layout:
Mar 11 20:57:35 smarthome kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Mar 11 20:57:35 smarthome kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Mar 11 20:57:35 smarthome kernel: [    0.000000]     vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
Mar 11 20:57:35 smarthome kernel: [    0.000000]     lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
Mar 11 20:57:35 smarthome kernel: [    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
Mar 11 20:57:35 smarthome kernel: [    0.000000]       .text : 0x80008000 - 0x80800000   (8160 kB)
Mar 11 20:57:35 smarthome kernel: [    0.000000]       .init : 0x80b00000 - 0x80c00000   (1024 kB)
Mar 11 20:57:35 smarthome kernel: [    0.000000]       .data : 0x80c00000 - 0x80c89244   ( 549 kB)
Mar 11 20:57:35 smarthome kernel: [    0.000000]        .bss : 0x80c90ed4 - 0x80d417b4   ( 707 kB)
Mar 11 20:57:35 smarthome kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mar 11 20:57:35 smarthome kernel: [    0.000000] ftrace: allocating 24646 entries in 73 pages
Mar 11 20:57:35 smarthome kernel: [    0.000000] Hierarchical RCU implementation.
Mar 11 20:57:35 smarthome kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Mar 11 20:57:35 smarthome kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Mar 11 20:57:35 smarthome kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Mar 11 20:57:35 smarthome kernel: [    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Mar 11 20:57:35 smarthome kernel: [    0.000022] Switching to timer-based delay loop, resolution 52ns
Mar 11 20:57:35 smarthome kernel: [    0.000273] Console: colour dummy device 80x30
Mar 11 20:57:35 smarthome kernel: [    0.000817] console [tty1] enabled
Mar 11 20:57:35 smarthome kernel: [    0.000857] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Mar 11 20:57:35 smarthome kernel: [    0.000897] pid_max: default: 32768 minimum: 301
Mar 11 20:57:35 smarthome kernel: [    0.001221] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
Mar 11 20:57:35 smarthome kernel: [    0.001254] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
Mar 11 20:57:35 smarthome kernel: [    0.002203] Disabling memory control group subsystem
Mar 11 20:57:35 smarthome kernel: [    0.002300] CPU: Testing write buffer coherency: ok
Mar 11 20:57:35 smarthome kernel: [    0.002710] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Mar 11 20:57:35 smarthome kernel: [    0.003104] Setting up static identity map for 0x100000 - 0x10003c
Mar 11 20:57:35 smarthome kernel: [    0.003238] Hierarchical SRCU implementation.
Mar 11 20:57:35 smarthome kernel: [    0.003887] smp: Bringing up secondary CPUs ...
Mar 11 20:57:35 smarthome kernel: [    0.004539] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
Mar 11 20:57:35 smarthome kernel: [    0.005253] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[... cut ...]
Mar 13 14:13:10 smarthome kernel: [148421.497255] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
Mar 13 14:13:10 smarthome kernel: [148421.497255]
Mar 13 14:56:37 smarthome kernel: [151029.288323] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
Mar 13 14:56:37 smarthome kernel: [151029.288323]
Mar 13 16:22:16 smarthome kernel: [156168.375240] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
Mar 13 16:22:16 smarthome kernel: [156168.375240]
`

I boot on a SSD

pelwell, 2018-03-13:
Thanks - there's nothing obviously wrong there, but it's still useful to have a more complete picture.

6by9, 2018-03-13:
I've seen the same error message. I currently have a local patch to nuke the error message, and everything still works OK.
The only USB devices I have connected are a mouse, keyboard and network on a Pi2. It does have an NFS file system.
I'll see if I still get the error if I revert the patch.

hofnarr132, 2018-03-28:
Please don't close an issue that is not resolved. Saying "there's nothing obviously wrong" does not mean it's fixed!

Have now experienced the problem on a RPi 3B+ after the latest standard update pushed 4.14.30-v7+.

The new RPi 3B+ had been fresh installed with from NOOBS 2.7.0 and worked fine with 4.9.80.

The updated firmware caused the RPi to fail to complete the init startup processes with 100s of errors:
```
WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
```

Only resolved by disabling the C-Media USB sound device in the USB HDMI KVM Switch and rebooting - RPi then proceeded to do a clean boot without problem.

Re-enabling the C-Media USB Sound device produces this from 'dmesg' (note that [ 18.476520] is the last entry from boot up, 133 onwards relates to turning on the USB device):
```
[   16.871637] Bluetooth: RFCOMM socket layer initialized
[   16.871662] Bluetooth: RFCOMM ver 1.11
[   18.175922] EXT4-fs (sda8): mounted filesystem with ordered data mode. Opts: (null)
[   18.303320] EXT4-fs (sda5): mounted filesystem with ordered data mode. Opts: (null)
[   18.476520] EXT4-fs (sda10): mounted filesystem with ordered data mode. Opts: (null)
[  133.775588] usb 1-1.2.5.4: new full-speed USB device number 9 using dwc_otg
[  133.913936] usb 1-1.2.5.4: New USB device found, idVendor=0d8c, idProduct=013c
[  133.913951] usb 1-1.2.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  133.913961] usb 1-1.2.5.4: Product: USB PnP Sound Device          
[  133.913970] usb 1-1.2.5.4: Manufacturer: C-Media Electronics Inc.      
[  133.919250] input: C-Media Electronics Inc.       USB PnP Sound Device           as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.5/1-1.2.5.4/1-1.2.5.4:1.3/0003:0D8C:013C.0003/input/input2
[  133.986156] hid-generic 0003:0D8C:013C.0003: input,hidraw2: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device          ] on usb-3f980000.usb-1.2.5.4/input3
[  134.096558] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.096585] usb 1-1.2.5.4: 2:1: cannot get freq at ep 0x82
[  134.096604] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.096612] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.104537] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.112536] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.112641] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.120551] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.120585] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.120594] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.120623] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.120631] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.122656] usbcore: registered new interface driver snd-usb-audio
[  134.144539] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  134.144644] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
```
lsusb output:
```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 009: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller
Bus 001 Device 008: ID 040b:2013 Weltrend Semiconductor 
Bus 001 Device 007: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 005: ID 05e3:0718 Genesys Logic, Inc. IDE/SATA Adapter
Bus 001 Device 004: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
```

pelwell, 2018-03-28:
Easy there - the issue is still open.

hofnarr132, 2018-03-28:
Apologies - saw the 'Closed' state against the cross referenced issue and jumped to wrong conclusion!

hofnarr132, 2018-03-28:
I have also noticed that dmesg continues to log the ``WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt:`` message at a rate of about 150 lines per second! This did not happen with the previous version.

macmpi, 2018-03-29:
@popcornmix has just pushed a corruption fix: does it help?

hofnarr132, 2018-03-31:
Tested on 1 x RPi 3B and found much improved. Will test more..

Prep for f/w 4.14.31 2148253d340d7c646583cb8f0be6403511522cc4
=============================================================

Tested on RPi 3B with 16GB SD-Card (boot device) and directly connected 4GB USB memory stick as storage device (log capture etc).
Also a 4x1 USB HDMI KVN Switch connected for HDMI display and provides a USB2 hub in which a C-Media USB sound device is embedded (this can be turned On/Off with a push button)
and a wireless mouse/keyboard is connected.

1). Started with NOOBS 2.7.0 2018-03-14 / Raspian 2018-03-13 Kernel 4.9 (firmware 4.9.80)
2). Applied all updates EXCEPT firmware and kernel related packages shown below:
```
pi@raspberrypi:~ $ apt list --upgradable
Listing... Done
base-files/stable 9.9+rpi1+deb9u4 armhf [upgradable from: 9.9+rpi1+deb9u3]
libraspberrypi-bin/stable 1.20180328-1 armhf [upgradable from: 1.20180313-1]
libraspberrypi-dev/stable 1.20180328-1 armhf [upgradable from: 1.20180313-1]
libraspberrypi-doc/stable 1.20180328-1 armhf [upgradable from: 1.20180313-1]
libraspberrypi0/stable 1.20180328-1 armhf [upgradable from: 1.20180313-1]
raspberrypi-bootloader/stable 1.20180328-1 armhf [upgradable from: 1.20180313-1]
raspberrypi-kernel/stable 1.20180328-1 armhf [upgradable from: 1.20180313-1]
raspberrypi-sys-mods/stable 20180328+1 armhf [upgradable from: 20180312]
rc-gui/stable 1.16 armhf [upgradable from: 1.15]
rpd-icons/stable 0.15 all [upgradable from: 0.14]
```
3). reboot and test USB sound device (OK)
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
```
4). rpi-update to 4.14.31 firmware
5). reboot and test USB sound device (OK)
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.31-v7+ #1104 SMP Thu Mar 29 16:52:18 BST 2018 armv7l GNU/Linux
```
6). 'sudo apt-get upgrade' to apply latest firmware and kernel
7). reboot and test USB sound device (OK)
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux
```
**Hmm. not sure how to check that latest firmware is actually being used - any pointers?**
 Tried re-applying firmware update - it reports it is the latest...
```
pi@raspberrypi:~ $ time sudo rpi-update 2148253d340d7c646583cb8f0be6403511522cc4
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date
```
In all cases I get between 12 and 25 WARN messages and then no more (looks like the 'usb_core' clears or resolves the 'dwc_otg_handle_mode_mismatch_intr:68'
```
usbcore: registered new interface driver snd-usb-audio
```

dmesg logs available if required.

kauailabs, 2018-04-04:
I'm still seeing this issue w/4.14.31, which I installed via rpi-update.  I have a USB headset, mouse and keyboard plugged into an IOgear USB switch, connected to the RPI.  I can't get it to boot due to the continual messages during boot - unless I disconnect the Logitech USB headset.  And then even after I do disconnect the Logitech USB headset and reboot, I'm still having significant performance problems.  Here's the last few seconds of entry from dmesg, followed by uname -a to display kernel version number.

This problem is occurring both on a Rpi3B and Rpi3B+.

Last few lines of output from dmesg:

[  422.847026] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  422.847142] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  422.862955] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[  422.863034] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.31-v7+ #1104 SMP Thu Mar 29 16:52:18 BST 2018 armv7l GNU/Linux

hofnarr132, 2018-04-07:
I've carried out further testing with 4.14.32 and also still see problems, but hopefully help narrow this down a bit to 'daisy chained' USB hubs..

Tested system: RPi 3B+ booting from Corsair SSD  which USB connected - using a PlusBerryPi USB device (SATA + USB Hub) and connected to a USB/HDMI KVM Switch with embedded USB hub, emdedded C-Media USB Sound Device and USB Wireless Keyboard/Mouse.

When the USB/HDMI KVM switch is plugged into to the PlusBerryPi board with the C-Media USB device active/connected (ie. 'daisy chained' RPi USB Hub --> PlusBerryPi Hub --> USB/HDMI KVM hub), I get *many* more WARN::dwc_otg_handle_mode_mismatch_intr:68: messages - this often seem to overwhelm the system on a startup and cause fail or time out (unless USB Sound device disconnected.

If the USB/HDMI KVM switch is connected directly to the RPi, then I get a few WARN messages as per my previous post on testing with a RPi 3B.

usb-device output:
```

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 1
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.14
S:  Manufacturer=Linux 4.14.32-v7+ dwc_otg_hcd
S:  Product=DWC OTG Controller
S:  SerialNumber=3f980000.usb
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480 MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1
P:  Vendor=0424 ProdID=2514 Rev=0b.b3
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=2mA
I:  If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=480 MxCh= 3
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1
P:  Vendor=0424 ProdID=2514 Rev=0b.b3
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=2mA
I:  If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub

T:  Bus=01 Lev=03 Prnt=03 Port=00 Cnt=01 Dev#=  6 Spd=480 MxCh= 0
D:  Ver= 2.10 Cls=ff(vend.) Sub=00 Prot=ff MxPS=64 #Cfgs=  1
P:  Vendor=0424 ProdID=7800 Rev=03.00
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=2mA
I:  If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=ff Driver=lan78xx

T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#=  4 Spd=480 MxCh= 6
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1
P:  Vendor=1a40 ProdID=0201 Rev=01.00
S:  Product=USB 2.0 Hub [MTT]
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I:  If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub

T:  Bus=01 Lev=03 Prnt=04 Port=00 Cnt=01 Dev#= 14 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=090c ProdID=1000 Rev=11.00
S:  Manufacturer=SMI Corporation
S:  Product=USB DISK
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage

T:  Bus=01 Lev=03 Prnt=04 Port=01 Cnt=02 Dev#=  5 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=05e3 ProdID=0718 Rev=00.09
S:  Product=USB Storage
S:  SerialNumber=000000000033
C:  #Ifs= 1 Cfg#= 1 Atr=c0 MxPwr=96mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage

T:  Bus=01 Lev=03 Prnt=04 Port=04 Cnt=03 Dev#=  7 Spd=480 MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=05e3 ProdID=0608 Rev=77.63
S:  Product=USB2.0 Hub
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=100mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub

T:  Bus=01 Lev=04 Prnt=07 Port=01 Cnt=01 Dev#=  8 Spd=1.5 MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
[... 82 lines cut ...]
[    4.384921] usb 1-1.2.5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    4.395137] usb 1-1.2.5.4: Product: USB PnP Sound Device          
[    4.405308] usb 1-1.2.5.4: Manufacturer: C-Media Electronics Inc.      
[    4.421768] input: C-Media Electronics Inc.       USB PnP Sound Device           as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.5/1-1.2.5.4/1-1.2.5.4:1.3/0003:0D8C:013C.0003/input/input2
[    4.451079] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
.
.  (Lots more WARN messages!)
.
[   14.258226] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   14.258323] usb 1-1.2.5.4: 1:1: cannot get freq at ep 0x1
[   14.266069] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
.
.  (More WARN messages)
.
[   14.314102] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

[   14.321911] usbcore: registered new interface driver snd-usb-audio
[   16.723415] uart-pl011 3f201000.serial: no DMA platform data
```

11. Subsequent reboots and WARN messages displaying for more than 600 seconds with may being dropped.

12. Connect KVM USB directly to RPi (this is requires the case lid to be removed).
Successfully boots C-Media USB device detected (tested with a mutiple reboots - WARN messages appear for between 4 and 10 seconds).

P33M, 2018-04-09:
Can you try booting with the following parameter settings in /boot/cmdline.txt?
```
dwc_otg.fiq_fsm_mask=0x0
dwc_otg.fiq_fsm_mask=0x1
dwc_otg.fiq_fsm_mask=0x3
dwc_otg.fiq_fsm_mask=0x7
```
Try each one in turn and see if it generates the errors in question. Note that the audio device may become unreliable (garbled audio) when testing these options.

citlee, 2018-04-10:
I am also seeing this issue (syslog flooded with ```WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode```), on a Pi 3B+ freshly imaged with 2018-03-13-raspbian-stretch-lite and then ```apt update/upgrade```'d and rebooted. The only attached USB devices are a keyboard, which is also a one port USB hub, which has a mouse attached. Unplugging the mouse from the keyboard hub stops these messages.

```uname -a``` reports:
```Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv71 GNU/Linux```

pelwell, 2018-04-10:
Please work through the four cmdline.txt settings (one at a time) as described by @P33M above.

dcs3spp, 2018-04-10:
Hi,

I have originally logged a separate issue [#2494](https://github.com/raspberrypi/linux/issues/2494) related to a usb bluetooth device not working in an externally powered hub. However, I am also receiving the *dwc_otg* warning messages and have been directed to this issue. 

In short, using *dwc_otg.fiq_fsm_mask=0x0* in cmdline.txt cleared the *dwc_otg* warning messages, however bluetooth controller remains unavailable.

I have an Asus BT-400 usb bluetooth device that is plugged into a self powered hub, Logik L4THUB10.

Details for the usb bluetooth device are:
> Bus 001 Device 006: ID 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth

Other usb devices running on the system are:
```
Bus 001 Device 007: ID 1005:b113 Apacer Technology, Inc. Handy Steno/AH123 / Handy Steno 2.0/HT203
Bus 001 Device 006: ID 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth
Bus 001 Device 005: ID 7392:7711 Edimax Technology Co., Ltd EW-7711UTn nLite Wireless Adapter [Ralink RT2870]
Bus 001 Device 004: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. SMC9512/9514 USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The usb tree is reported as follows:
```
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/3p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=rt2800usb, 480M
            |__ Port 3: Dev 6, If 3, Class=Application Specific Interface, Driver=, 12M
            |__ Port 3: Dev 6, If 1, Class=Vendor Specific Class, Driver=btusb, 12M
            |__ Port 3: Dev 6, If 2, Class=Vendor Specific Class, Driver=btusb, 12M
            |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=btusb, 12M
            |__ Port 4: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M
```

I have performed the following actions.

1. Downloaded and installed latest updates
`sudo pacman -Syu`
2. This downloaded a later kernel version
`uname -a`
> Linux raspberrypi.ubuntu.private 4.14.33-1-ARCH #1 SMP Tue Apr 10 00:59:56 UTC 2018 armv6l GNU/Linux
3. Powered off hub and Pi, removed bluetooth from external usb hub and then
rebooted Pi.
No *dwc_otg* mode mismatch interrupt warning messages are logged when the
bluetooth device is removed from the external usb hub.
4. Followed dmesg:
`dmesg --follow`
5. Inserted bluetooth device into external usb hub. The *dwc_otg* warning messages are now logged. The bluetooth device patch command fails,
command 0x0c03 tx timeout occurs and reset fails.
```
[ 1115.741505] usb 1-1.3.3: new full-speed USB device number 7 using dwc_otg
[ 1115.880547] usb 1-1.3.3: New USB device found, idVendor=0b05, idProduct=17cb
[ 1115.888158] usb 1-1.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1115.896100] usb 1-1.3.3: Product: BCM20702A0
[ 1115.900556] usb 1-1.3.3: Manufacturer: Broadcom Corp
[ 1115.905791] usb 1-1.3.3: SerialNumber: 5CF370607CCB
[ 1116.500200] Bluetooth: Core ver 2.22
[ 1116.509339] NET: Registered protocol family 31
[ 1116.514202] Bluetooth: HCI device and connection manager initialized
[ 1116.524615] Bluetooth: HCI socket layer initialized
[ 1116.529716] Bluetooth: L2CAP socket layer initialized
[ 1116.552891] Bluetooth: SCO socket layer initialized
.
. WARN messages
.
[ 1116.701903] usbcore: registered new interface driver btusb
.
. WARN messages
[... 428 lines cut ...]
[   25.351568] Bluetooth: SCO socket layer initialized
[   25.570883] usbcore: registered new interface driver btusb
[   26.671208] random: crng init done
[   27.119880] usb 1-1.3.1: reset high-speed USB device number 6 using dwc_otg
[   27.271010] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
[   27.322489] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 0005 detected
[   27.471682] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   27.500641] usbcore: registered new interface driver rt2800usb
[   27.610078] Bluetooth: hci0 command 0x0c03 tx timeout
[   28.197981] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   32.970289] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   32.983712] Bluetooth: BNEP filters: protocol multicast
[   33.136518] Bluetooth: BNEP socket layer initialized
[   35.097168] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[   35.230395] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[   35.689887] Bluetooth: hci0: BCM: Reset failed (-110)
[   36.451045] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   45.260754] wlan0: authenticate with 9c:c1:72:09:d2:50
[   45.340450] wlan0: send auth to 9c:c1:72:09:d2:50 (try 1/3)
[   45.357121] wlan0: authenticated
[   45.390040] wlan0: associate with 9c:c1:72:09:d2:50 (try 1/3)
[   45.424099] wlan0: RX AssocResp from 9c:c1:72:09:d2:50 (capab=0x411 status=0 aid=5)
[   45.472479] wlan0: associated
[   50.763835] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

hofnarr132, 2018-04-15:
@P33M and @pelwell  - apologies not been able to run thru /boot/cmdline.txt switches yet.

**Update on daisy chained hubs... Power related?** The setup that is giving me most grief is as follows:

RPi 3B+ (with USB Hub) --> PlusBerryPi *Powered Hub (with SSD drive) --> USB/HDMI KVM hub

The PlusBerryPi is powered by a 3.5 AMP 5 Volt supply with powers the RPi3B+ and provides a 7 port USB hub (1 port connected to RPi 3B+, a USB to SATA interface with a SSD drive attached, 1 port to the KVM switch). I'm using USB booting from the SSD drive (no SD Card). Additionally I'm using the onboard Wifi and HDMI.

As I was getting the occasional low power warning, I considered that the PlusBerryPi was not getting enough power through to the RPi - sadly I am not equipped to measure USB power draw, so I've replaced the 3B+ with a 3B (as it should draw less power) which is configured for USB boot (no change to the OS on the SSD) and now seem to have a 'power' stable system - however I still get ``WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode`` messages on boot or when activating (on/off button on KVM) the C-Media sound device on the KVM.

As to the RPi 3B+, I have put that in the 3B case with a 2.5 AMP 5.25 Volt supply directly to the RPi 3B, connected directly to the KVM switch (RPi 3B USB Hub -> KVM Switch hub) and booting from SD Card. This system is now 'power' stable so far and almost never showing the ``WARN`` message.
 
I will continue to test when able and report back - I hope to be able to run the cmdline.txt options on both RPis this week.

pelwell, 2018-04-16:
Thanks for the feedback. As you may have guessed, we still don't know why those warnings have started appearing with the 4.14 kernel. I seem the occasionally at boot time, but I usually only have a mouse and keyboard attached.

P33M, 2018-04-26:
Ah. That's a clue. Can you post a full `sudo lsusb -v` with the dongle plugged into the USB hub?

TheNetStriker, 2018-04-26:
Here is the output of the command:

```
pi@retropie:~ $ sudo lsusb -v

Bus 001 Device 009: ID 045e:0291 Microsoft Corp. Xbox 360 Wireless Receiver for Windows
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0         8
  idVendor           0x045e Microsoft Corp.
  idProduct          0x0291 Xbox 360 Wireless Receiver for Windows
  bcdDevice            1.07
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          321
    bNumInterfaces          8
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              260mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     93 
      bInterfaceProtocol    129 
      iInterface              0 
      ** UNRECOGNIZED:  14 22 00 01 13 81 1d 00 17 01 02 08 13 01 0c 00 0c 01 02 08
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
[... 714 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
```

This time the dongle also worked when plugging in directly to the Pi. Maybe this is only a problem when the device is running for a while or when the temperature is higher.

TheNetStriker, 2018-05-14:
What's the status regarding this problem? I now also have this problem on one of my Rasberry Pi 3's, not just on the plus.

JamesH65, 2018-05-14:
ping @P33M

xKaiser, 2018-05-14:
I'm coming here from https://github.com/raspberrypi/linux/issues/2438 - similar issues but different device.

I'm getting a lot of "WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode" messages, and device issues attaching a ThingMagic USB RFID reader to the raspberry pi via a [D-Link USB hub](https://www.amazon.com/D-Link-including-Charging-Adapter-DUB-H4/dp/B00006B7DA) which is only present when attaching from the USB hub. Occasionally the device will disappear from the system with the following in the dmesg log:

[17019.648594] cdc_acm 1-1.2.1:1.0: failed to set dtr/rts
[17464.841078] cdc_acm 1-1.2.1:1.0: failed to set dtr/rts
[17702.508226] usb 1-1.2-port1: disabled by hub (EMI?), re-enabling...
[17702.508551] usb 1-1.2.1: USB disconnect, device number 6
[17702.508694] cdc_acm 1-1.2.1:1.0: failed to set dtr/rts
[17703.620472] usb 1-1.2-port1: Cannot enable. Maybe the USB cable is bad?
[17704.530443] usb 1-1.2-port1: Cannot enable. Maybe the USB cable is bad?
[17704.530778] usb 1-1.2-port1: attempt power cycle
[17705.780540] usb 1-1.2-port1: Cannot enable. Maybe the USB cable is bad?
[17706.690496] usb 1-1.2-port1: Cannot enable. Maybe the USB cable is bad?
[17706.690791] usb 1-1.2-port1: unable to enumerate USB device

lsusb -v log is here: https://pastebin.com/psWPp6Ya

dzerrenner, 2018-05-15:
Just as a FYI: the same issue occurs on my Zero W and 4.14 with this KVM-switch: [Lindy 42340](https://www.lindy.co.uk/kvm-c6/kvm-switches-c295/2-port-hdmi-usb-2-0-audio-kvm-switch-compact-p6899). I connect HDMI and the integrated USB hub of the switch, which is itself connected to a mouse and a keyboard. No audio jack connected.

I can't help much on debugging, but i can provide some logs / tests if needed.

balintkozma, 2018-05-28:
Same issue here [Irdroid USB](https://www.irdroid.com/irdroid-usb-ir-transceiver/) in a self-powered USB 3.0 hub with 5V 4A PSU.
Boots up without any problem, but after doing some IR stuff (receiving ~20 IR commands, then sending one) the indicator led hangs, and the warning messages come.

xKaiser, 2018-06-12:
I modified /boot/cmdline.txt to contain:
dwc_otg.fiq_enable=1 
dwc_otg.fiq_fsm_enable=1 
dwc_otg.fiq_fsm_mask=0x3

Other than one 

> [   49.911033] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

 at boot, I stopped receiving timeouts with the RFID reader attached to the powered hub. I see other oddities (no RFID tags read, when there are tags in range) about once every couple hours, but this is an improvement from my last post.

amfasis, 2018-06-15:
I also had trouble after updating the kernel a few months back and have been keeping an eye on this thread.

I can confirm the three additions to /boot/cmdline.txt stated by xKaiser work on my RaspberryPi 2B+ with a C-Media Electronics device connected through a Genesys Logic, Inc. Hub (which indeed has Single TT).

I was on 4.9.80 through rpi-update, but just did a `apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel` to 4.14.34 and some reboots with and without the /boot/cmdline.txt additions: without I get the endless stream of warnings.
Also note that inserting the faulting usb device while shutting down the RP will sort of halt the shutdown process

tolland, 2018-06-21:
I am seeing the same issue on a Pi Model B Plus Rev 1.2 which has a startech KVM plugged in to one of the sockets;
```
root@raspberrypi-00007:~# lsusb
Bus 001 Device 011: ID 1e7d:2e23 ROCCAT
Bus 001 Device 009: ID 413c:2005 Dell Computer Corp. RT7D50 Keyboard
Bus 001 Device 008: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
[dmesg is here](https://pastebin.com/4nAqbMKw )

dracenmarx, 2018-06-21:
I have an update. I can reproduce the issue with the 7-port USB Hub (D-Link), mouse and keyboard. The weird thing is: When mouse and keyboard are plugged into the left ports of the Hub, everything works. As soon as I plug mouse or keyboard into the ports at the right (port 6, 7), then the syslog gets filled with those error messages. As soon as I plug then into the other ports, the messages stop.

pelwell, 2018-06-21:
What does `lsusb -t` report in the working and non-working positions?  If it's difficult to capture the output in the non-working state you could try running `watch lsusb -t` and taking a photo of the output.

JamesH65, 2018-06-21:
@dracenmarx Ah, that is a very interesting datapoint...

tolland, 2018-06-22:
Apologies, I am catching up here, so I might have missed something earlier in the thread. 

I have a couple of pis setup as intercoms and they were both crashed at the error message after a reboot after being updated with `raspi-update`. I flashed them back to `2018-04-18-raspbian-stretch-lite` and they still had the same problem on reboot, but going back to `2018-03-13` they are both fine again, and no error messages.

P33M, 2018-06-22:
My current working theory is that the driver is doing something bad when multiple periodic transfers are queued to start in the same microframe to a single-TT hub. This would explain why people see the issue all the time (with unhappy combinations causing many collisions) and others virtually never see it, or see it once on boot.

What should happen is that the first transfer should execute exclusively, with the second transfer held off until we can guarantee that the hub won't be asked to perform transfers out-of-order (and drop data). 

Why this didn't happen in 4.9, but happens in 4.14 - I don't know.

As a workaround, by only attaching a single Full- or Low-speed device to a single-TT hub will mitigate the issue. This probably isn't going to work with the Xbox wireless adapter though - there are endpoints there that are both input and output direction with an interval of 1 frame...

xKaiser, 2018-06-22:
I currently have about 20 devices always running that exhibit this issue randomly. When it occurs, I can give access to a RPF engineer SSH access to a malfunctioning device if it would be of any help for debugging purposes (if we connect privately). I do have two high speed devices (Hologram Nova modem, and ThingMagic M6e RFID reader) connected to the hub, which is consistent with the working theory. I can spend some time testing any potential fixes as well.

xKaiser, 2018-06-26:
@TheNetStriker I've seen the same USB connection issues on 4.9, but the message didn't appear the spam the logs as much, so I'm not sure that there was a time in which this worked.

To load a specific kernel version from [here](https://github.com/Hexxeh/rpi-firmware/commits/master),

`sudo rpi-update 2663a143525df3c8305253c6f93f1707156e6a9d`

where the hex string after is the commit identifier.

Low87, 2018-06-28:
Same problem here with a Deltaco 7 port USB 3.0 hub. Connecting the hub is fine, but as soon as I connect a device to it, I get a series of errors:

`$ uname -a
Linux mhzpi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux`

`Jun 28 11:20:12 mhzpi kernel: [ 1488.621595] usb 1-1.1.2: new high-speed USB device number 61 using dwc_otg
Jun 28 11:20:12 mhzpi kernel: [ 1488.754874] usb 1-1.1.2: New USB device found, idVendor=2109, idProduct=2812
Jun 28 11:20:12 mhzpi kernel: [ 1488.754889] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 28 11:20:12 mhzpi kernel: [ 1488.754899] usb 1-1.1.2: Product: USB2.0 Hub
Jun 28 11:20:12 mhzpi kernel: [ 1488.754907] usb 1-1.1.2: Manufacturer: VIA Labs, Inc.
Jun 28 11:20:12 mhzpi kernel: [ 1488.755698] hub 1-1.1.2:1.0: USB hub found
Jun 28 11:20:12 mhzpi kernel: [ 1488.755980] hub 1-1.1.2:1.0: 4 ports detected
Jun 28 11:20:13 mhzpi kernel: [ 1489.091588] usb 1-1.1.2.1: new high-speed USB device number 62 using dwc_otg
Jun 28 11:20:13 mhzpi kernel: [ 1489.234599] usb 1-1.1.2.1: New USB device found, idVendor=2109, idProduct=2812
Jun 28 11:20:13 mhzpi kernel: [ 1489.234613] usb 1-1.1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 28 11:20:13 mhzpi kernel: [ 1489.234622] usb 1-1.1.2.1: Product: USB2.0 Hub
Jun 28 11:20:13 mhzpi kernel: [ 1489.234630] usb 1-1.1.2.1: Manufacturer: VIA Labs, Inc.
Jun 28 11:20:13 mhzpi kernel: [ 1489.235460] hub 1-1.1.2.1:1.0: USB hub found
Jun 28 11:20:13 mhzpi kernel: [ 1489.235710] hub 1-1.1.2.1:1.0: 4 ports detected
Jun 28 11:20:21 mhzpi kernel: [ 1497.801629] usb 1-1.1.2.4: new full-speed USB device number 63 using dwc_otg
Jun 28 11:20:21 mhzpi kernel: [ 1497.801694] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
Jun 28 11:20:21 mhzpi kernel: [ 1497.801694]
Jun 28 11:20:21 mhzpi kernel: [ 1497.801719] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
Jun 28 11:20:21 mhzpi kernel: [ 1497.801719]
Jun 28 11:20:21 mhzpi kernel: [ 1497.801809] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
Jun 28 11:20:21 mhzpi kernel: [ 1497.801809]
Jun 28 11:20:21 mhzpi kernel: [ 1497.801839] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode`

gbfromhb, 2018-06-28:
Having same issues. 
Raspberry pi 3
Google Voice Kit version 1
Amazon basic 4 port USB 3 powered hub

**Without the USB HUB works fine**.

**If I have the HUB connected during boot up I get the following:**

`[    3.009856] snd-googlevoicehat-soundcard soc:sound: snd_soc_register_card() failed: -517`
(later in log)
`[    9.481546] usb 1-1.3.4: new full-speed USB device number 5 using dwc_otg
[    9.481601] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
`
(later in log)
`[   89.174694] usb 1-1.3.4: device descriptor read/64, error -71
`

**Boot up with out the HUB connected, and connect it after boot, and get the following.**
I get the same as above, but without the sound card plus the following
`[   91.534772] usb 1-1.3.4: device not accepting address 8, error -71
[   91.535047] usb 1-1.3-port4: unable to enumerate USB device
`
Any advice would be greatly appreciated.

Glenn

P33M, 2018-06-28:
Please post full logs of the failing case(s).

gbfromhb, 2018-06-28:
First time here, sorry about that. At the moment I'm having to move forward with some time constants. Right now I will will not be able to regenerate the log. If you feel the post is not helpful I will be glad to take it down, and by the way thanks for all the help.

mypiandrew, 2018-06-29:
Hi

Just to chip in

I get the same issue on a custom CM3 board when trying to access a FTDI 230XS USB serial device connected via STT Genesys GL850G hub

Didn't see this on 4.9.80 

Problem current as of

```
root@raspberrypi:~# uname -a
Linux raspberrypi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux
```

Here's the USB config :

```
root@raspberrypi:~# lsusb
Bus 001 Device 008: ID 1618:9113
Bus 001 Device 007: ID 0424:2240 Standard Microsystems Corp.
Bus 001 Device 006: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 005: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 003: ID 0424:9512 Standard Microsystems Corp. SMC9512/9514 USB Hub
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@raspberrypi:~# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 6, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 3: Dev 7, If 0, Class=Mass Storage, Driver=usb-storage, 480M

```
dmesg :
https://gist.github.com/mypiandrew/331cf98189973884872b8e3c77086b01

If I try and access the FTDI USB device it spams the dmesg output, here's a sample of the output 

```
[13582.090686] ftdi_sio ttyUSB0: failed to get modem status: -32
[13582.138524] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13582.138802] ftdi_sio ttyUSB0: failed to get modem status: -32
[13582.266620] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13582.266719] ftdi_sio ttyUSB0: failed to get modem status: -71
[13582.442622] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13582.442712] ftdi_sio ttyUSB0: failed to get modem status: -71
[13582.602613] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13582.602715] ftdi_sio ttyUSB0: failed to get modem status: -71
[13582.747325] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13583.748824] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13584.750451] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13584.750540] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13584.874473] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13584.874505] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13584.874597] ftdi_sio ttyUSB0: failed to get modem status: -71
[13584.970347] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
[13584.970503] ftdi_sio ttyUSB0: failed to get modem status: -71
[13585.082402] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode
```

How can I assist with debugging?

kaltinril, 2018-07-14:
I had no space left on my 32GB drive and noticed the /var/log file was taking up 12+ GB of space.
- /var/log/syslog
- /var/log/messages
- /var/log/kern.log

all three are over 4 gigs

**When checking the /var/log/kern.log i noticed the same error**
WARN::dwc_otg_handle_mode_mismatch_intr:68 Mode Mismatch Interrupt: currently in Host mode

if I tail the log file, the screen just scrolls with new entries over and over.

I have connected:
1. logitec wireless K520 keyboard and mouse
2. XBOX360 dongle
3. HDMI cord
4. CANA Kit power cord

Raspberry pi 3b

Upgraded recently from Jessie to Stretch  (Unsure if it was happening before).

currently on 4.14.52-v7  1123

hippocritical, 2018-07-14:
Hi to all, I had the same problem and thanks to kaltinril I solved the problem for me.

Setup:Raspberry pi Stretch with post - install installed retropie package

Hardware: raspberry pi "raspberry usb hub" with connected wifi dongle for the xbox controller 360 from china

solved with: switched chinese xbox360 wifi dongle from usb hub to native raspberry usb connection.

kaltinril, 2018-07-15:
@P33M Would you like a log from the RetroFlag case which has the same error?

mypiandrew, 2018-07-15:
Hi, this looks to me like a problem between STT USB Hub Chips and the RPI USB interface. 

The RetroPi also uses the same GL850 hub chip featured repeatedly in the above chain as best I can make out.

This effect is a regression from 4.9.80 which does not have this problem

jfarid27, 2018-07-18:
I can also confirm this issue on a reflash with Raspbian 4.14. In light of the "USB hub", my error came up when I was actually using a Rosewill keyboard with a built in USB hub and I received the same error.

I did see a mention of `low voltage` before the long string of repeated `Mode Mismatch Interrupt` errors, so I guess using any sort of usb hub is the root issue. I also just used a different keyboard and the error went away.

mypiandrew, 2018-07-18:
@dracenmarx 

I don't think so, this problem is occurring only on kernel 4.14.x   It does not appear on 4.9.80

The common thread is appears to be usage of FS/LS devices in conjunction with the STT GL850 hub and the RPi USB interface.

It's a common hub chip, see the RetroPi comment above

We have a custom Compute Module based setup, using the same hub chip, and see the same issue when we switch to the  4.14 branch.

I suspect it's more likely just a USB bug that's most likely been fixed once already, gotten lost in the migration, and needs adding in again to the 4.14 branch.

xKaiser, 2018-07-18:
@mypiandrew 
I have observed the issue in the 4.9.80 with my device, but the log messages were much less common. I don't think we can say 4.9.80 works with certainty.

cicci44, 2018-07-21:
can also confirm issue with Retroflag Nespi case.

Joennuh, 2018-08-05:
I got the same issue while trying to get a noname wireless USB mouse to a Vivanco USB 2.0 hub. The hub is attached to a USB OTG cable. The USB OTG cable is attached to a Raspberry Pi Zero WH.

More technical I have the following devices attached:
- a Vivanco USB 2.0 hub (which shows up at lsusb as a Genesys Logic, Inc. Hub and at lsusb -v as a Single-TT hub).
- a noname wireless keyboard/ballmouse (which shows up at lsusb as Holtek Semiconductor, Inc.), connected to above mentioned hub.
- a noname wireless mouse (which shows up at lsusb as Creative Labs Wireless Keyboard/Mouse), connected to above mentioned hub and causes the error messages.

At first the mouse seemed to be slow and I discovered at https://www.raspberrypi.org/forums/viewtopic.php?p=600742#p600742 that I needed to add "usbhid.mousepoll=0" to /boot/cmdline.txt. It seems to be that the issue occurs since that moment. At boot I see lots of "[  628.185745] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode" messages. Only when I disconnect the noname wireless mouse from the USB hub the Raspberri Pi Zero WH continues to boot normally. After reattaching the noname wireless mouse after a few moments the messages came much less to zero at dmesg.

I have quite a beginner level at Raspbian debugging and I can't get the messages from boot time in dmesg.

Output of lsusb -v and dmesg:
[https://joen.stackstorage.com/s/DQZhNd9JIJpUgR1](https://joen.stackstorage.com/s/DQZhNd9JIJpUgR1)

Output of lsusb and  lsusb -t:
```
pi@energizer-v2:~ $ lsusb
Bus 001 Device 011: ID 04d9:a01c Holtek Semiconductor, Inc.
Bus 001 Device 010: ID 062a:4101 Creative Labs Wireless Keyboard/Mouse
Bus 001 Device 009: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@energizer-v2:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 9, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 10, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 10, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 11, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 11, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M

```		
Output of uname -a:
```
pi@energizer-v2:~ $ uname -a
Linux energizer-v2 4.14.52+ #1123 Wed Jun 27 17:05:32 BST 2018 armv6l GNU/Linux
```

I hope this is enough information for the developers to solve this frustrating issue.

jakraska, 2018-08-06:
I am also having problems with Retroflag Nespi case.

AnthraxLeprosyPi, 2018-08-11:
Hey everybody, 

Exact same thing came from Jessie and upgraded to stretch wondered where the 12gig I had left in a 128GB SD card went... 

Found them in the logs folder... 😪

Also got nespi plus case and a Xbox wireless receiver.

Is there any bleeding edge kernel stuff somewhere I can try? 😜

Regards,
Anthrax

TheNetStriker, 2018-08-18:
I've just tested the with latest kernel and firmware, but the problem is still there. I am booting my RPI from an external USB SSD. Could this be part of the problem?

hofnarr132, 2018-08-20:
I confirm my HDMI KVM switch has a `ID 05e3:0608 Genesys Logic, Inc. Hub` and `lsusb -v` reports `bDeviceProtocol 1 Single TT` for it.

TheNetStriker, 2018-08-23:
@Joennuh I bought myself an Elektron Overhub to test this. This devices is built to attach multiple realtime music devices to a pc, so it should be very good. And the situation has actually improved with this hub. I still got a few of the error messages on startup or e.g. when starting the Openhab service, but then the error doesn't seam to appear anymore.

The main problem that I had was a massive delay in my ZWave network. Switching a light on with the ZWave USB stick on the usb hub did take 10 times as long as without the hub. Now with the Overhub it is back to normal.

kennymingt, 2018-08-30:
Hi,
I also have this issue with a USBHub and wireless mouse.
Here is my dmesg output right when I plug the mouse into the USBhub.
`
[    6.796033] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    6.796064] brcmfmac: power management disabled
[    7.186102] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   11.061879] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   12.228961] fuse init (API version 7.26)
[   12.498541] Bluetooth: Core ver 2.22
[   12.498607] NET: Registered protocol family 31
[   12.498610] Bluetooth: HCI device and connection manager initialized
[   12.498628] Bluetooth: HCI socket layer initialized
[   12.498637] Bluetooth: L2CAP socket layer initialized
[   12.498662] Bluetooth: SCO socket layer initialized
[   12.512373] Bluetooth: HCI UART driver ver 2.3
[   12.512382] Bluetooth: HCI UART protocol H4 registered
[   12.512385] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.512531] Bluetooth: HCI UART protocol Broadcom registered
[   12.845481] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.845498] Bluetooth: BNEP filters: protocol multicast
[   12.845522] Bluetooth: BNEP socket layer initialized
[   12.913183] Bluetooth: RFCOMM TTY layer initialized
[   12.913229] Bluetooth: RFCOMM socket layer initialized
[   12.913256] Bluetooth: RFCOMM ver 1.11
[   21.083497] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
[  308.177996] usb 1-1.2: reset full-speed USB device number 4 using dwc_otg
[  308.828055] usb 1-1.1.3: reset low-speed USB device number 5 using dwc_otg
[  309.248040] usb 1-1.2: reset full-speed USB device number 4 using dwc_otg
[  309.478038] usb 1-1.3: new high-speed USB device number 7 using dwc_otg
[  309.610528] usb 1-1.3: New USB device found, idVendor=8564, idProduct=4100
[  309.610544] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  309.610554] usb 1-1.3: Product: USB2.0 Hub
[  309.610563] usb 1-1.3: Manufacturer: Transcend Info
[  309.612146] hub 1-1.3:1.0: USB hub found
[  309.612569] hub 1-1.3:1.0: 4 ports detected

[  344.038484] usb 1-1.3.1: new low-speed USB device number 8 using dwc_otg
[  344.176528] usb 1-1.3.1: New USB device found, idVendor=046d, idProduct=c31c
[  344.176545] usb 1-1.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  344.176555] usb 1-1.3.1: Product: USB Keyboard
[  344.176563] usb 1-1.3.1: Manufacturer: Logitech
[  344.193438] input: Logitech USB Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.1/1-1.3.1:1.0/0003:046D:C31C.0007/input/input6
[  344.262887] hid-generic 0003:046D:C31C.0007: input,hidraw0: USB HID v1.10 Keyboard [Logitech USB Keyboard] on usb-3f980000.usb-1.3.1/input0
[  344.286001] input: Logitech USB Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.1/1-1.3.1:1.1/0003:046D:C31C.0008/input/input7
[  344.349063] hid-generic 0003:046D:C31C.0008: input,hidraw1: USB HID v1.10 Device [Logitech USB Keyboard] on usb-3f980000.usb-1.3.1/input1
^[[B^[[A^[[1;2D^[[1;2A[  388.039165] usb 1-1.1.2: new full-speed USB device number 9 using dwc_otg
[  388.180284] usb 1-1.1.2: New USB device found, idVendor=1d57, idProduct=fa60
[  388.180304] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  388.180313] usb 1-1.1.2: Product: 2.4G Wireless Mouse
[  388.180323] usb 1-1.1.2: Manufacturer: LXDDZ
[  388.186831] input: LXDDZ 2.4G Wireless Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2/1-1.1.2:1.0/0003:1D57:FA60.0009/input/input8
[  388.250262] hid-generic 0003:1D57:FA60.0009: input,hidraw2: USB HID v11.01 Keyboard [LXDDZ 2.4G Wireless Mouse] on usb-3f980000.usb-1.1.2/input0
[  388.255950] input: LXDDZ 2.4G Wireless Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2/1-1.1.2:1.1/0003:1D57:FA60.000A/input/input9
[  388.256571] hid-generic 0003:1D57:FA60.000A: input,hidraw3: USB HID v11.01 Mouse [LXDDZ 2.4G Wireless Mouse] on usb-3f980000.usb-1.1.2/input1
[  388.266860] input: LXDDZ 2.4G Wireless Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2/1-1.1.2:1.2/0003:1D57:FA60.000B/input/input10
[  388.329863] hid-generic 0003:1D57:FA60.000B: input,hiddev96,hidraw4: USB HID v11.01 Device [LXDDZ 2.4G Wireless Mouse] on usb-3f980000.usb-1.1.2/input2
[  388.334193] input: LXDDZ 2.4G Wireless Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2/1-1.1.2:1.3/0003:1D57:FA60.000C/input/input11
[  388.399803] hid-generic 0003:1D57:FA60.000C: input,hidraw5: USB HID v11.01 Keyboard [LXDDZ 2.4G Wireless Mouse] on usb-3f980000.usb-1.1.2/input3

[  429.030329] usb 1-1.1.2: USB disconnect, device number 9
[  439.039819] usb 1-1.3.4: new full-speed USB device number 10 using dwc_otg
[  439.179802] usb 1-1.3.4: New USB device found, idVendor=1d57, idProduct=fa60
[  439.179818] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  439.179829] usb 1-1.3.4: Product: 2.4G Wireless Mouse
[  439.179839] usb 1-1.3.4: Manufacturer: LXDDZ
[  439.188726] input: LXDDZ 2.4G Wireless Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.0/0003:1D57:FA60.000D/input/input12
[  439.250917] hid-generic 0003:1D57:FA60.000D: input,hidraw2: USB HID v11.01 Keyboard [LXDDZ 2.4G Wireless Mouse] on usb-3f980000.usb-1.3.4/input0
[  439.256562] input: LXDDZ 2.4G Wireless Mouse as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.4/1-1.3.4:1.1/0003:1D57:FA60.000E/input/input13
[... cut ...]
[  439.342016] WARN::dwc_otg_handle_mode_mismatch_intr:68: Mode Mismatch Interrupt: currently in Host mode

`

Then is goes forever
`
pi@raspberrypi:~ $ sudo du /var/log -d 1 -h
144K	/var/log/apt
36K	/var/log/lightdm
4.0K	/var/log/samba
94M	/var/log
pi@raspberrypi:~ $ sudo du /var/log -d 1 -h
144K	/var/log/apt
36K	/var/log/lightdm
4.0K	/var/log/samba
102M	/var/log
pi@raspberrypi:~ $ sudo du /var/log -d 1 -h
144K	/var/log/apt
36K	/var/log/lightdm
4.0K	/var/log/samba
107M	/var/log
`

I hope it is useful

WaeCo, 2018-08-31:
As a workaround for me adding `dwc_otg.speed=1` to cmdline.txt worked.
Note that this will slow down all USB devices to USB 1.1 speeds (full-speed).

[37 more replies]

closed 2018-09-24 by P33M