raspberrypi/linux #2409: audioinjector-octo-soundcard.c - Clocks turn off

Dloranger, 2018-03-02:
The code turns off the clocks entirely when not playing audio out which is breaking an application that depends on these audio clocks being present.  I have spoken to the original developer about this and he provided the changes needed (moving 3 case statements) to resolve.

I would like to see these changes made mainstream, but I am not sure if there might be other factors involved that would warrant not changing them that I am not aware of. 

The changes are to move lines 180-182 to line 140 so the clocks stay on during all cases.

pelwell, 2018-03-03:
I don't like the idea of changing the current behaviour for all users, so instead I've add a DT property the prevents the clocks from being switched off. The property is enabled with a new DT parameter of the `audioinjector-addons` overlay - `non-stop-clocks`:
```
dtoverlay=audioinjector-addons,non-stop-clocks
```

Dloranger, 2018-03-03:
Makes total sense for me, appreciate the help.

closed 2018-03-05 by Dloranger