raspberrypi/linux #2411: audioinjector-octo-soundcard - Random channel assignment in speaker-test

Dloranger, 2018-03-03:
When I run the speaker-test utility and probe audio output channel 0 (note card has 8 channels out, 6 input) with an oscilloscope, the channel reported where I get sound from appears to be randomly assigned

I have detailed the testing and my setup here:
http://forum.audioinjector.net/viewtopic.php?f=9&t=4501#p7264

~/.asoundrc and /etc/asound.conf are configured using the manual method described here
http://forum.audioinjector.net/viewtopic.php?f=9&t=1222#p2684

Pillar1989, 2018-03-09:
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/dma/bcm2835-dma.c has some bugs. But no people have time to fixed it.

Dloranger, 2018-03-11:
Can you expand on what the bugs are and how they would relate to this issue?  I am willing to take a look and tinker to see if I can figure out what the issues are

JamesH65, 2018-06-29:
@popcornmix @Pillar1989 Do we have any idea of what the issues are in this driver?

pelwell, 2018-06-29:
@flatmax might be a better person to ask.

Dloranger, 2018-06-29:
the main issue i see comes from a lack of synchronization to channels,
especially between threads. likely a bi-product of the time division
multiplexing being used.

the card generally works as expected for single threads, but there seems to
be a need for a channel sync signal or etc.

when sound is playing in thread 1, and another thread plays sounds, the
channel assignments get messed up so channel zero is now 3/5/6 etc so what
you expect to come out on channel 0 is not on zero.

i currently have a board setup for testing should someone be willing to
spend some time on it in the software space.  I would be happy to be the
hardware tester for potential fixes

flatmax, 2018-07-01:
Hi there,

There are no random channel assignments when your application opens the card and accesses the device. The links at the top of this post are about card detection at boot time.

I have not run speaker test, however hundreds of users run all sorts of applications without random channel swapping.

When you talk about multiple thread access, which devices are you using ? How is your software accessing the audio device ?

thanks
Mat

Dloranger, 2018-07-03:
I am building an image that I will share that demonstrates the problem, should have it posted later today

Dloranger, 2018-07-03:
Here is a link to the image for download, note this is a lite image, run headless

http://www.ics-ctrl.com/content/pi_images/octobase-sound-issues.zip

Test the sound output on the channel 0 & 1 load (headphones or etc) as follows

`speaker-test -D pasym0 -t sine -f 500 -c 8`

you should hear a pure tone on channels 0 and 1, or perhaps 2 other channels.

Next reload the svxlink software in a second console session

`sudo service svxlink restart & tail -f /var/log/svxlink`

if you toggle between the two sessions killing and reloading the commands you will see that the tone being emitted by speaker test changes channels

flatmax, 2018-07-03:
What is the device pasym0 ?

Dloranger, 2018-07-03:
its a shared sound device, see asound.conf as the code there has been
update per alsa instructions for sharing the sound card between applications

Dloranger, 2018-07-04:
> If you are experiencing channel swaps it is most likely due to the way
> the ALSA plugin you are using operates.
>
> The Octo requires either continuous half duplex operation or continuous
> full duplex operation. In other words, only one input or output device
> is opened OR both input and output devices must be opened at the same
> time. You can't have operations jumping in mid stream with arbitrary
> channel counts.
>
> I notice that pasym is a non standard user space plugin where dmix and
> dsnoop are used to create a single virtual device.
>
> People have all sorts of solutions in the case that they want to open
> the device in asymmetric full duplex mode (e.g. open the output then
> later open the input) or in any other strange manner. I suggest you
> start there and experiment with how you are operating dsnoop and dmix.
>
> thanks
>
> Matt

I wouldnt call this non standard, as its prescribed by alsa team directly
as the solution here for sharing a device
https://alsa.opensrc.org/AlsaSharing
(its offline at the moment)

which is duplicated here for reference since its offline

https://github.com/opensrc/alsa/blob/master/lib/md/AlsaSharing.md

If you can point me to a solution that demonstratably fits the requirements
of handling the asynchronous needs, I am happy to use it as this has been
frustrating to no end for me.

appreciate any ideas for making this all work together